espressif/esp-mesh-lite #1: Is it able to make the mesh lite AP hide it's SSID?

novflying, 2023-01-29:
Is it able to make the mesh lite AP hide it's SSID?

novflying, 2023-01-29:
wifi_cfg.ap.ssid_hidden = 1;

xcguang, 2023-01-31:
If the number of nodes is too large, it is not recommended to hide SSID.

novflying, 2023-01-31:
> If the number of nodes is too large, it is not recommended to hide SSID.

Why is that? I don't think it's a good idea to make the space full of SSIDs which will make the end users confused.

xcguang, 2023-01-31:
It needs to test if there is any issue, we have not tested this.

tswen, 2023-01-31:
After a preliminary test, hiding the ssid has no effect on the mesh networking functionality.

yel-best, 2024-01-24:
> ```
>     wifi_cfg.ap.ssid_hidden = 1;
> ```
Hi @novflying 
If you use this parameter, it seems that all nodes will connect to the route instead of forming a mesh, and it may also lead to the inability to connect to the network. I wonder if you have encountered this?

yel-best, 2024-01-24:
> After a preliminary test, hiding the ssid has no effect on the mesh networking functionality.

Hi @tswen 

Does mesh-lite definitely need to display all esp32 WiFi SSIDs? If hiding it will cause all nodes to be directly connected to the route, I can see all the esp32 direct connections through my route background, and they do not form a mesh. And if the esp32 is restarted, it will cause the network to be unable to connect. Can't we hide the SSID?

[image]

yel-best, 2024-01-25:
Can I make all esp32 using mesh-lite use the same wifi ssid for mesh networking? For example: My route SSID: iot-wifi, mesh AP SSID: All nodes use esp-mesh.

yel-best, 2024-01-25:
> > Can I make all esp32 using mesh-lite use the same wifi ssid for mesh networking? For example: My route SSID: iot-wifi, mesh AP SSID: All nodes use esp-mesh.
> 
> Yes, you can do like this

Is it necessary to modify the logic of mesh scan here? I modify **CONFIG_BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC = 0** in the following code so that all softAPs created by nodes use **ESP_Bridge**. Is this okay?
I didn't find a place where the parameters of **CONFIG_BRIDGE_SOFTAP_SSID** could be modified. I used idf.py menuconfig and found no config that could be modified.

[image]