espressif/esp-mesh-lite #2: Getting Error: fatal error: mbedtls/config.h: No such file or directory when compiling example mesh-local-control

Teesmo, 2023-02-08:
OS - Ubuntu
IDF - release/v4.4
Development board - ESP32-S2-DevKit-M-1 v1.0
IDE - no (running IDF in command line)

__Problem description__

Hi,
I am trying out esp-mesh-lite, and have run into problems striaght away. I tried running the mesh-lcoal-control example, having made no changes to the code (I ran it as is), and encountered this error during the build process:
**_Error: fatal error: mbedtls/config.h: No such file or directory_**

This is a slightly more verbose version of the error:

**_In file included from /home/onepower/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/build_info.h:65,
                 from /home/onepower/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/sha256.h:29,
                 from /home/onepower/esp/esp-idf/components/bootloader_support/src/idf/bootloader_sha.c:12:
/home/onepower/esp/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:31:10: fatal error: mbedtls/config.h: No such file or directory
 #include "mbedtls/config.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated._**

What may be causing this problem?

PS: The example runs in the latest version (master branch) of IDF, but then the board reboots due to a Guru Meditation error, or sometimes and assert error. 
Please help!

wujiangang, 2023-02-09:
It's not related to esp-mesh-lite self, please update the submodules of esp-idf by `git submodule update --init --recursive --progress`.

Teesmo, 2023-02-09:
> It's not related to esp-mesh-lite self, please update the submodules of esp-idf by `git submodule update --init --recursive --progress`.

I tried it and it worked. Thank you.

wujiangang, 2023-02-10:
Great, if the issue is solved, please close it. If you have any other issues, feel free to raise new ones.

closed 2023-02-10 by Teesmo