espressif/esp-mesh-lite #3: Bump esp-iot-bridge version to 0.3 to get USB bridge working

redfast00, 2023-03-13:
We'd like to use the new USB bridge feature with esp-mesh-lite. I've set `espressif/iot_bridge: ">=0.3"` in `components/mesh_lite/idf_component.yml`, but this results in the following error:

```
/home/user/Projects/mesh/esp-mesh-lite/examples/mesh_local_control/managed_components/espressif__iot_bridge/src/bridge_usb.c:30:10: fatal error: tinyusb.h: No such file or directory
   30 | #include "tinyusb.h"
      |          ^~~~~~~~~~~
compilation terminated.
```

redfast00, 2023-03-13:
This is very likely caused by https://github.com/espressif/esp-iot-bridge/issues/39

tswen, 2023-03-14:
Thank you for your feedback, you can add the following patch to the code for now, we will update the example for the NIC solution later.

You can refer to https://github.com/espressif/esp-iot-bridge/issues/39 and we will update iot-bridge later to better support usb_device.

For now, you can temporarily modify the esp-iot-bridge repository and push a branch to your repository, then you can temporarily modify the **yml file** under the **mesh_lite** component by changing `espressif/iot_bridge: 0.*` to 
```
iot_bridge:
    path: components/iot_bridge
    git: Your github repository address
    version: xxxxxx
```

redfast00, 2023-05-29:
@tswen Thank you for the temporary workaround!

I'd like to avoid having to document this temporary workaround, would it be okay if I made a PR with the patches you suggested?