espressif/esp-mesh-lite #4: Connect from outside of the mesh to internal nodes

redfast00, 2023-03-30:
We have a (non-espressif) device connected to the SoftAP network that is set up by the ESP32 mesh nodes. This device can access the internet, but we can't connect to an open port on the device from outside of the mesh. We've tried to add an IP route that goes via the root node, but this doesn't seem to work.

tswen, 2023-04-03:
Because the data flow transmission and forwarding between devices are based on NAPT, if external access to the internal is required, portmap needs to be manually added. However, for ESP-Mesh-Lite, it is not suitable for external direct access to internal devices because it involves too many levels and portmap will become more complex.
You can use MQTT to send and receive data via publish and subscribe.

redfast00, 2023-04-03:
Thank you for the suggestion. MQTT would not work for us, since we need generic TCP/IP networking. We'd like to use the ESP32 as a sort of mesh networking wireless card for Linux single-board-computers. Do you think that would be possible with the ESP32?

tswen, 2023-04-04:
MQTT is also based on TCP/IP networking, it's just one of the applications at the application layer, but this is just one suggestion.

If I understand correctly, you want a Linux computer to connect to the Mesh-Lite network wirelessly and be able to access the external network, and this is possible.

tswen, 2023-04-06:
As mentioned above, accessing devices inside the internal mesh network from outside is quite difficult, but not impossible, although it is not recommended.
This is because port mapping needs to be added at each level on the mesh devices, which is difficult to maintain for dynamic DHCP. If you still want to implement this method, you can first connect the Linux computer to the root node and try adding a simple portmap on the root node.

```c
#define IP_PROTO_TCP     6
#define IP_PROTO_UDP     17
static void ip_event_sta_got_ip_handler(void *arg, esp_event_base_t event_base,                   
                                        int32_t event_id, void *event_data)
{     
    ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;
    ESP_LOGI(TAG, "Connected with IP Address:" IPSTR, IP2STR(&event->ip_info.ip));

    ip_portmap_add(IP_PROTO_TCP, event->ip_info.ip.addr, 7788, ESP_IP4TOADDR(192, 168, 4, 2), 7080);
    ip_portmap_add(IP_PROTO_UDP, event->ip_info.ip.addr, 7788, ESP_IP4TOADDR(192, 168, 4, 2), 7080);
}
```
Replace `192.168.4.2` with the IP address obtained by the Linux computer. Externally, only access the 7788 port corresponding to the IP address of the root node device Station, and you can access the 7080 port corresponding to `192.168.4.2`.

redfast00, 2023-04-06:
Thank you, this looks more like what I'm trying to accomplish :)

Are there plans to develop a layer 2 mesh network (instead of the current layer 3 mesh network) where you're directly connected to the host network? That way, you'd get your IP address from the DHCP server of the host network, and networking is easier.

tswen, 2023-04-13:
For Mesh-Lite, it is not supported. What you mentioned may be similar to [ip_internal_network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network).

tswen, 2025-06-10:
In point-to-point communication, only child nodes can actively initiate communication with their parent nodes. Additionally, for point-to-point communication, we highly recommend using ESP-NOW, as it is more efficient.

mjgci, 2025-06-10:
As suggested in the User Guide to use network interfaces [https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md#data-transmission](url), it would have been useful to reach the IP addresses beyond the parent nodes within the mesh network.

ESP-NOW is handy however, the throughput is limited.