espressif/esp-mesh-lite #5: CMake Error at managed_components/espressif__cmake_utilities/package_manager.cmake:43 (target_compile_options):   target_compile_options may only set INTERFACE properties on INTERFACE   targets Call Stack (most recent call first):   managed_components/espressif__mesh_lite/CMakeLists.txt:32 (cu_pkg_define_version)

287662354, 2023-04-20:
i  used this components in former code（use  esp-mesh,sub-devices in the network to independently  ）   my  idf_component.yml  such as:
IDFused 5.0.1

## IDF Component Manager Manifest File
dependencies:
  espressif/iot_bridge: "^0.4.0"

  espressif/mesh_lite: "^0.1.1"
  espressif/esp_modem:
    version: 0.*
  espressif/esp_modem_usb_dte:
    public: true
    rules:
    - if: target in [esp32s2, esp32s3]
    version: 1.*
    
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true

tswen, 2023-04-20:
Are you using the latest version of esp-mesh-lite? Is the mesh-lite component version confirmed to be 0.1.1?

tswen, 2023-04-20:
Please modify the title of the issue and use a sentence to summarize the problem. Please provide the specific reason for the problem in the comment section of the issue. Also, it is recommended to pay attention to the formatting.

287662354, 2023-06-20:
不知道为什么用vscode组件包管理器中默认的install添加格式上不知道是否有问题   目前手动修改为如下格式成功解决,另外如果有遇到bool关键字未定义的情况 ,  可以把头文件包含 从.c文件中移动到.h文件中解决
`
## IDF Component Manager Manifest File
dependencies:
  espressif/mesh_lite: "^0.2.0"
  joltwallet/littlefs: "^1.5.4"
  espressif/esp_modem:
    version: 0.*
  espressif/esp_modem_usb_dte:

    public: true

    rules:

    - if: target in [esp32s2, esp32s3]

    - if: idf_version >= 4.4

    version: 1.*
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true

依赖中 包含LittleFS不需要用的话可以移除

`

closed 2024-03-04 by xcguang