espressif/esp-mesh-lite #6: 引用 espressif/mesh_lite: "^0.1.2"後報錯

8888alex, 2023-05-21:
IDF v5.0.2
blink 範例專案
idf_component.yml
dependencies:
  espressif/esp_modem_usb_dte: "^0.0.3"
  espressif/mesh_lite: "^0.1.2"
  espressif/led_strip: "^2.0.0"

錯誤一
-- Project sdkconfig file C:/Users/alex/esp/idf_Examples/v5.0/blink/sdkconfig
warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)
錯誤二
CMake Error at managed_components/espressif__cmake_utilities/package_manager.cmake:43 (target_compile_options):
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
Call Stack (most recent call first):
  managed_components/espressif__mesh_lite/CMakeLists.txt:25 (cu_pkg_define_version)

報錯日誌
-- Found Git: C:/Users/alex/esp/idf_tools/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
Processing 8 dependencies:
[1/8] espressif/cmake_utilities (0.4.7)
[2/8] espressif/esp_modem (0.1.28)     
[3/8] espressif/esp_modem_usb_dte (0.0.3)
[4/8] espressif/iot_bridge (0.5.0)
[5/8] espressif/led_strip (2.3.1)
[6/8] espressif/mesh_lite (0.1.2)
[7/8] espressif/usb_host_cdc_acm (1.0.4)
[8/8] idf (5.0.2)
-- Project sdkconfig file C:/Users/alex/esp/idf_Examples/v5.0/blink/sdkconfig
warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/alex/esp/idf_tools/python_env/idf5.0_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "blink" version: 1
-- Adding linker script C:/Users/alex/esp/idf_Examples/v5.0/blink/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/soc/esp32s2/ld/esp32s2.peripherals.ld
error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply
fatal: not a git repository (or any of the parent directories): .git
-- IOT_BRIDGE: 0.5.0
-- MESH_LITE: 0.1.2
CMake Error at managed_components/espressif__cmake_utilities/package_manager.cmake:43 (target_compile_options):
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
Call Stack (most recent call first):
  managed_components/espressif__mesh_lite/CMakeLists.txt:25 (cu_pkg_define_version)

-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cmake_utilities espressif__esp_modem espressif__esp_modem_usb_dte espressif__iot_bridge espressif__led_strip espressif__mesh_lite espressif__usb_host_cdc_acm esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/alex/esp/esp-idf-v5.0.2/components/app_trace C:/Users/alex/esp/esp-idf-v5.0.2/components/app_update C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader_support C:/Users/alex/esp/esp-idf-v5.0.2/components/bt C:/Users/alex/esp/esp-idf-v5.0.2/components/cmock C:/Users/alex/esp/esp-idf-v5.0.2/components/console C:/Users/alex/esp/esp-idf-v5.0.2/components/cxx C:/Users/alex/esp/esp-idf-v5.0.2/components/driver C:/Users/alex/esp/esp-idf-v5.0.2/components/efuse C:/Users/alex/esp/esp-idf-v5.0.2/components/esp-tls C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_adc C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_app_format C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_common C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_eth C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_event C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_gdbstub C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hid C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_client C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_ota C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hw_support C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_lcd C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_local_ctrl C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_netif C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_partition C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_phy C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_pm C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_psram C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_ringbuf C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_timer C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_wifi C:/Users/alex/esp/esp-idf-v5.0.2/components/espcoredump C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__cmake_utilities C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__esp_modem C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__esp_modem_usb_dte C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__iot_bridge C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__led_strip C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__mesh_lite C:/Users/alex/esp/idf_Examples/v5.0/blink/managed_components/espressif__usb_host_cdc_acm C:/Users/alex/esp/esp-idf-v5.0.2/components/esptool_py C:/Users/alex/esp/esp-idf-v5.0.2/components/fatfs C:/Users/alex/esp/esp-idf-v5.0.2/components/freertos C:/Users/alex/esp/esp-idf-v5.0.2/components/hal C:/Users/alex/esp/esp-idf-v5.0.2/components/heap C:/Users/alex/esp/esp-idf-v5.0.2/components/http_parser C:/Users/alex/esp/esp-idf-v5.0.2/components/idf_test C:/Users/alex/esp/esp-idf-v5.0.2/components/ieee802154  [... cut ...] /esp/esp-idf-v5.0.2/components/xtensa
-- Configuring incomplete, errors occurred!
See also "C:/Users/alex/esp/idf_Examples/v5.0/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/alex/esp/idf_Examples/v5.0/blink/build/CMakeFiles/CMakeError.log".

 *  終端機處理序 "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\alex\esp\idf_Examples\v5.0\blink\build -S c:\Users\alex\esp\idf_Examples\v5.0\blink" 已終止。結束代碼: 1。

tswen, 2023-05-23:
是不是 idf 的 submodule 没有更新导致 patch 没有正常打入。

真正的错误原因
error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply

8888alex, 2023-05-24:
> 是不是 idf 的 submodule 没有更新导致 patch 没有正常打入。
> 
> 真正的错误原因 error: patch failed: src/core/ipv4/ip4.c:332 error: src/core/ipv4/ip4.c: patch does not apply
請問我該如何確認?
又該如何修正?

tswen, 2023-05-24:
你先确认下  [esp-iot-bridge/components/iot_bridge/patch/ip4_forward.patch](https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_forward.patch) 能不能正常打入 $IDF_PATH/components/lwip/lwip

Even-yuwei, 2023-05-24:
../managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)  这个可能是什么导致的，需要怎么修改

8888alex, 2023-05-24:
> 你先确认下 [esp-iot-bridge/components/iot_bridge/patch/ip4_forward.patch](https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_forward.patch) 能不能正常打入 $IDF_PATH/components/lwip/lwip。

不好意思我不知道要如何"打入 $IDF_PATH/components/lwip/lwip"

8888alex, 2023-05-29:
不好意思，請問下我還能做甚麼檢查?

8888alex, 2023-05-31:
新的發現。
經測試在對照[ip4_forward.patch](https://github.com/espressif/esp-iot-bridge/blob/master/components/iot_bridge/patch/ip4_forward.patch)
復原ipv4後，第一次build的錯誤其中並沒有
error: patch failed: src/core/ipv4/ip4.c:332
error: src/core/ipv4/ip4.c: patch does not apply
且查看ip4.c內容已經被套用ip4_forward.patch但build失敗
第二是以後的build因為ip4_forward.patch已經被套用所以上面的錯誤
以下為第一次build的日誌

-- Found Git: C:/Users/alex/esp/idf_tools/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/alex/esp/idf_tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
Processing 8 dependencies:
[1/8] espressif/cmake_utilities (0.4.8)
[2/8] espressif/esp_modem (0.1.28)
[3/8] espressif/esp_modem_usb_dte (0.0.3)
[4/8] espressif/iot_bridge (0.5.0)
[5/8] espressif/led_strip (2.3.2)
[6/8] espressif/mesh_lite (0.1.2)
[7/8] espressif/usb_host_cdc_acm (1.0.4)
[8/8] idf (5.0.2)
-- Project sdkconfig file C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/sdkconfig
warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/alex/esp/idf_tools/python_env/idf5.0_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "blink" version: 1
-- Adding linker script C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/soc/esp32s2/ld/esp32s2.peripherals.ld
fatal: not a git repository (or any of the parent directories): .git
-- IOT_BRIDGE: 0.5.0
-- MESH_LITE: 0.1.2
CMake Error at managed_components/espressif__cmake_utilities/package_manager.cmake:43 (target_compile_options):
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
Call Stack (most recent call first):
  managed_components/espressif__mesh_lite/CMakeLists.txt:25 (cu_pkg_define_version)

-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cmake_utilities espressif__esp_modem espressif__esp_modem_usb_dte espressif__iot_bridge espressif__led_strip espressif__mesh_lite espressif__usb_host_cdc_acm esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/alex/esp/esp-idf-v5.0.2/components/app_trace C:/Users/alex/esp/esp-idf-v5.0.2/components/app_update C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader_support C:/Users/alex/esp/esp-idf-v5.0.2/components/bt C:/Users/alex/esp/esp-idf-v5.0.2/components/cmock C:/Users/alex/esp/esp-idf-v5.0.2/components/console C:/Users/alex/esp/esp-idf-v5.0.2/components/cxx C:/Users/alex/esp/esp-idf-v5.0.2/components/driver C:/Users/alex/esp/esp-idf [... cut ...] sers/alex/esp/esp-idf-v5.0.2/components/wpa_supplicant C:/Users/alex/esp/esp-idf-v5.0.2/components/xtensa
-- Configuring incomplete, errors occurred!
See also "C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/build/CMakeFiles/CMakeOutput.log".

 *  終端機處理序 "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build -S c:\Users\alex\esp\idf_Examples\v5.0\blink\blink" 已終止。結束代碼: 1。

tswen, 2023-05-31:
检查一下 CONFIG_MESH_LITE_ENABLE 配置项是不是没开。

8888alex, 2023-05-31:
確實，CONFIG_MESH_LITE_ENABLE 配置项没开(如圖一)
但因引用後無法正常build，進而無法開啟設定介面
移除引用後可以開啟設定介面卻又沒有MESH相關選項可以設定(如圖二)
請問該如何完成設置?
圖一
[image]
圖二
[image]
介面開啟日誌如下
[SDK Configuration Editor]
Executing action: confserver
Running cmake in directory C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\alex\esp\idf_Examples\v5.0\blink\blink"...
-- ccache will be used for faster recompilation

[SDK Configuration Editor]
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2

[SDK Configuration Editor]
Processing 8 dependencies:

[1/8] espressif/cmake_utilities (0.4.8)

[SDK Configuration Editor]
[2/8] espressif/esp_modem (0.1.28)

[SDK Configuration Editor]
[3/8] espressif/esp_modem_usb_dte (0.0.3)

[SDK Configuration Editor]
[4/8] espressif/iot_bridge (0.5.0)

[SDK Configuration Editor]
[5/8] espressif/led_strip (2.3.2)

[SDK Configuration Editor]
[6/8] espressif/mesh_lite (0.1.2)

[SDK Configuration Editor]
[7/8] espressif/usb_host_cdc_acm (1.0.4)

[SDK Configuration Editor]
[8/8] idf (5.0.2)

[SDK Configuration Editor]
-- Project sdkconfig file C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/sdkconfig

[SDK Configuration Editor]
---------------------------ERROR--------------------------

warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- App "blink" version: 1

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/build/esp-idf/esp_system/ld/memory.ld

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system/ld/esp32s2/sections.ld.in

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.ld

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld

[SDK Configuration Editor]
[... 37 lines cut ...]
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
Call Stack (most recent call first):
  managed_components/espressif__mesh_lite/CMakeLists.txt:25 (cu_pkg_define_version)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cmake_utilities espressif__esp_modem espressif__esp_modem_usb_dte espressif__iot_bridge espressif__led_strip espressif__mesh_lite espressif__usb_host_cdc_acm esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/alex/esp/esp-idf-v5.0.2/components/app_trace C:/Users/alex/esp/esp-idf-v5.0.2/components/app_update C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader_support C:/Users/alex/esp/esp-idf-v5.0.2/components/bt C:/Users/alex/esp/esp-idf-v5.0.2/components/cmock C:/Users/alex/esp/esp-idf-v5.0.2/components/console C:/Users/alex/esp/esp-idf-v5.0.2/components/cxx C:/Users/alex/esp/esp-idf-v5.0.2/components/driver C:/Users/alex/esp/esp-idf-v5.0.2/components/efuse C:/Users/alex/esp/esp-idf-v5.0.2/components/esp-tls C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_adc C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_app_format C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_common C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_eth C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_event C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_gdbstub C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hid C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_client C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_ota C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hw_support C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_lcd C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_local_ctrl C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_netif C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_partition C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_phy C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_pm C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_psram C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_ringbuf C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_timer C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_wifi C:/Users/alex/esp/esp-idf-v5.0.2/components/espcoredump C:/Users/alex/esp/idf [... cut ...]
cmake failed with exit code 1, output of the command is in the C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build\log\idf_py_stderr_output_25776 and C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build\log\idf_py_stdout_output_25776

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

8888alex, 2023-05-31:
> 检查一下 CONFIG_MESH_LITE_ENABLE 配置项是不是没开。

確實，CONFIG_MESH_LITE_ENABLE 配置项没开(如圖一)
但因引用後無法正常build，進而無法開啟設定介面(介面開啟日誌如下)
移除引用後可以開啟設定介面卻又沒有MESH相關選項可以設定(如圖二)
請問該如何完成設置?
圖一
[image]
圖二
[image]
介面開啟日誌
[SDK Configuration Editor]
Executing action: confserver
Running cmake in directory C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Users\alex\esp\idf_Examples\v5.0\blink\blink"...
-- ccache will be used for faster recompilation

[SDK Configuration Editor]
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2

[SDK Configuration Editor]
Processing 8 dependencies:

[1/8] espressif/cmake_utilities (0.4.8)

[SDK Configuration Editor]
[2/8] espressif/esp_modem (0.1.28)

[SDK Configuration Editor]
[3/8] espressif/esp_modem_usb_dte (0.0.3)

[SDK Configuration Editor]
[4/8] espressif/iot_bridge (0.5.0)

[SDK Configuration Editor]
[5/8] espressif/led_strip (2.3.2)

[SDK Configuration Editor]
[6/8] espressif/mesh_lite (0.1.2)

[SDK Configuration Editor]
[7/8] espressif/usb_host_cdc_acm (1.0.4)

[SDK Configuration Editor]
[8/8] idf (5.0.2)

[SDK Configuration Editor]
-- Project sdkconfig file C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/sdkconfig

[SDK Configuration Editor]
---------------------------ERROR--------------------------

warning: the int symbol BRIDGE_SOFTAP_MAX_CONNECT_NUMBER (defined at C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/managed_components/espressif__iot_bridge/Kconfig:60) has a non-int default BRIDGE_CUSTOM_MAX_CONNECT_NUMBER (undefined)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- App "blink" version: 1

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/idf_Examples/v5.0/blink/blink/build/esp-idf/esp_system/ld/memory.ld

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system/ld/esp32s2/sections.ld.in

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.ld

[SDK Configuration Editor]
-- Adding linker script C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
[... 39 lines cut ...]
  target_compile_options may only set INTERFACE properties on INTERFACE
  targets
Call Stack (most recent call first):
  managed_components/espressif__mesh_lite/CMakeLists.txt:25 (cu_pkg_define_version)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__cmake_utilities espressif__esp_modem espressif__esp_modem_usb_dte espressif__iot_bridge espressif__led_strip espressif__mesh_lite espressif__usb_host_cdc_acm esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/alex/esp/esp-idf-v5.0.2/components/app_trace C:/Users/alex/esp/esp-idf-v5.0.2/components/app_update C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader C:/Users/alex/esp/esp-idf-v5.0.2/components/bootloader_support C:/Users/alex/esp/esp-idf-v5.0.2/components/bt C:/Users/alex/esp/esp-idf-v5.0.2/components/cmock C:/Users/alex/esp/esp-idf-v5.0.2/components/console C:/Users/alex/esp/esp-idf-v5.0.2/components/cxx C:/Users/alex/esp/esp-idf-v5.0.2/components/driver C:/Users/alex/esp/esp-idf-v5.0.2/components/efuse C:/Users/alex/esp/esp-idf-v5.0.2/components/esp-tls C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_adc C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_app_format C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_common C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_eth C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_event C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_gdbstub C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hid C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_client C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_http_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_ota C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_https_server C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_hw_support C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_lcd C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_local_ctrl C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_netif C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_partition C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_phy C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_pm C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_psram C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_ringbuf C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_rom C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_system C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_timer C:/Users/alex/esp/esp-idf-v5.0.2/components/esp_wifi C:/Users/alex/esp/esp-idf-v5.0.2/components/espcoredump [... cut ...]
cmake failed with exit code 1, output of the command is in the C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build\log\idf_py_stderr_output_25776 and C:\Users\alex\esp\idf_Examples\v5.0\blink\blink\build\log\idf_py_stdout_output_25776

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

wujiangang, 2023-05-31:
你直接把你的工程打包提供一下。

8888alex, 2023-05-31:
> 你直接把你的工程打包提供一下。
謝謝 麻煩了
[blink.zip](https://github.com/espressif/esp-mesh-lite/files/11612017/blink.zip)

wujiangang, 2023-06-01:
有几个问题：
1、main/Kconfig.projbuild 里的 MESH_LITE_ENABLE 选项可以去除，mesh_lite 组件内有同名的选项
2、把 mesh_local_control 下的 sdkconfig.defaults 拿到 blink 下，把你希望要对 mesh_lite 和 iot_bridge 要做的配置放到 defaults 里，当前你在 main/Kconfig.projbuild 里 select 了选项，sdkconfig.defaults 拿过来后，你需要删除 sdkconfig，重新做下 idf.py set-target esp32-s2 build
3、mesh_lite 内部要做些处理，解决 MESH_LITE_ENABLE 未打开时的编译问题，这个需要更新一个 mesh_lite 版本，我们会尽快更新

8888alex, 2023-06-01:
謝謝您，已先按您說的先將 mesh_local_control 下的 sdkconfig.defaults 拿到 blink 下做臨時地處裡。
處理過程如下
在複製sdkconfig.defaults後還需要(光是只刪除 managed_components 無效)
1. 先移除idf_component.yml中的
espressif/esp_modem_usb_dte: "^0.0.3"
espressif/mesh_lite: "^0.1.2"
2. build以清除錯誤的設定內容. 
3. 加回idf_component.yml中的
espressif/esp_modem_usb_dte: "^0.0.3"
espressif/mesh_lite: "^0.1.2"

要有這三個步驟才能成功完成建構

期待更新的版本。
辛苦了。

wujiangang, 2023-06-07:
@8888alex 最新的 mesh_lite 组件，v0.2.0，应该解决了这个问题，参见：https://components.espressif.com/components/espressif/mesh_lite，这里先关闭 issue，如仍有问题，可以再打开。

closed 2023-06-07 by wujiangang