espressif/esp-mesh-lite #7: Get topology of network

redfast00, 2023-05-29:
For debugging purposes, I'd like to visualize the topology of the network, in order to be able to draw a tree like below (but with MAC-addresses for every node):
[image]

Is there an API I can use to retrieve this information?

xcguang, 2023-05-30:
We‘re going to do this in the future, if you are interested in this, it's welcome to push a PR. you can get children's MAC in each node, and report the list to the root node, then you can get all nodes topology in the mesh-network.

redfast00, 2023-06-01:
Alright, I opened a draft PR in https://github.com/espressif/esp-mesh-lite/pull/8 (the code is not ready yet, but this shows the approach I'm going to take).