espressif/esp-mesh-lite #10: 在build範例中改用local_control.c出現堆棧溢位

8888alex, 2023-06-06:
IDF v5.0.2
blink 範例專案
重現步驟清空blink 範例專案
將mesh_local_control\main中的所有檔案複製到blink\main(如圖)
[image]
將mesh_local_control下的partitions.csv及sdkconfig.defaults也比照複製
順利編譯燒錄後監看
得到錯誤後重啟
***ERROR*** A stack overflow in task Tmr Svc has been detected.

錯誤日誌如下
PS C:\Users\alex\esp\idf_Examples\v5.0\blink> set IDF_PATH=C:/Users/alex/esp/esp-idf-v5.0.2/
PS C:\Users\alex\esp\idf_Examples\v5.0\blink> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\esp\idf_Examples\v5.0\blink\build\blink.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (21) boot: compile time 08:52:47
I (21) boot: chip revision: v1.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=220ach (139436) map
I (118) esp_image: segment 1: paddr=000320d4 vaddr=3ffc6210 size=02db0h ( 11696) load
I (121) esp_image: segment 2: paddr=00034e8c vaddr=40022000 size=0b18ch ( 45452) load
I (135) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=86008h (548872) map
I (245) esp_image: segment 4: paddr=000c6030 vaddr=4002d18c size=09078h ( 36984) load
I (264) boot: Loaded app from partition at offset 0x10000
I (264) boot: Disabling RNG early entropy source...
I (276) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (276) cpu_start: Pro cpu up.
I (295) cpu_start: Pro cpu start user code
I (295) cpu_start: cpu freq: 160000000 Hz
I (295) cpu_start: Application information:
I (298) cpu_start: Project name:     blink
I (302) cpu_start: App version:      1
I (307) cpu_start: Compile time:     Jun  6 2023 08:52:28
I (313) cpu_start: ELF file SHA256:  c4d1ed6687f8a237...
I (319) cpu_start: ESP-IDF:          v5.0.2-dirty
I (324) cpu_start: Min chip rev:     v0.0
I (329) cpu_start: Max chip rev:     v1.99 
I (334) cpu_start: Chip rev:         v1.0
I (339) heap_init: Initializing. RAM available for dynamic allocation:
I (346) heap_init: At 3FFCCF20 len 0002F0E0 (188 KiB): DRAM
I (352) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (358) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (365) spi_flash: detected chip: generic
I (369) spi_flash: flash io: dio
W (373) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (387) cpu_start: Starting scheduler on PRO CPU.
I (399) bridge_common: esp-iot-bridge version: 0.5.0
I (409) wifi:wifi driver task: 3ffd5bec, prio:23, stack:6656, core=0
I (409) system_api: Base MAC address is not set
[... 200 lines cut ...]
I (1596) ESP_Mesh_Lite_Comm: Bind Socket 54, port 6364
I (1596) ESP_Mesh_Lite_Comm: Bind Socket 55, port 6363
I (1606) ESP_Mesh_Lite_Comm: Bind Socket 56, port 6366
I (1606) ESP_Mesh_Lite_Comm: Bind Socket 57, port 6365
W (1616) vendor_ie: [Unexpected restart]: the rtc data: level:1, ssid:MikroTik-8A6FB4, bssid b8:69:f4:8a:6f:b4
W (1626) vendor_ie: Try to connect the original parent node
I (2396) wifi:ap channel adjust o:1,1 n:6,1
I (2396) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1
I (3336) wifi:state: init -> auth (b0)
I (3346) wifi:state: auth -> assoc (0)
I (3346) wifi:state: assoc -> run (10)
I (3366) wifi:connected with MikroTik-8A6FB4, aid = 3, channel 6, 40U, bssid = b8:69:f4:8a:6f:b4
I (3366) wifi:security: WPA2-PSK, phy: bgn, rssi: -63
I (3366) wifi:pm start, type: 1

I (3376) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3386) wifi:<ba-add>idx:0 (ifx:0, b8:69:f4:8a:6f:b4), tid:0, ssn:3, winSize:64
I (4366) esp_netif_handlers: sta ip: 192.168.88.54, mask: 255.255.255.0, gw: 192.168.88.1
I (4366) bridge_wifi: Connected with IP Address:192.168.88.54
I (4366) vendor_ie: RTC store: ssid:MikroTik-8A6FB4; bssid:b8:69:f4:8a:6f:b4 crc:2024743569
I (4376) local_control: Create a tcp client, ip: 192.168.0.1, port: 8070
重複發生堆棧溢位重啟，以下省略。

隨附工程檔案
[blink.zip](https://github.com/espressif/esp-mesh-lite/files/11658809/blink.zip)

wujiangang, 2023-06-07:
你把 sdkconfig.defaults 里的 CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=3072 改成 4096，或者通过 menuconfig 把这个配置改大一些试一下。

8888alex, 2023-06-07:
> CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH=4096
在設置中將CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH改成4096有效
但將設置放在sdkconfig.defaults中無效
謝謝回覆

xcguang, 2023-06-08:
放到 sdkconfig.defaults，需要删除 sdkconfig 后编译，才有效