espressif/esp-mesh-lite #11: confusing mesh lite node ip

novflying, 2023-06-07:
I have 2 of EPS32S3 boards running the mesh_local_control, the ip addresses are confusing seems like they are connected to each other resulting in a loop?

The first node's log:
I (96195) bridge_wifi: STA Connecting to the AP again...
I (96228) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.7.2
I (104343) ESP_Mesh_Lite_Comm: approved: 1
I (114345) ESP_Mesh_Lite_Comm: approved: 1
I (121550) local_control: System information, channel: 2, layer: 5, self mac: 60:55:f9:f5:99:4c, parent bssid: 60:55:f9:f5:9d:0d, parent rssi: -31, free heap: 261956
I (121555) local_control: WIFI_AP_IP : 192.168.7.1
I (121560) local_control: WIFI_STA_IP: 192.168.6.2
I (121566) local_control: Child mac: 60:55:f9:f5:9d:0c
I (124344) ESP_Mesh_Lite_Comm: approved: 1
I (134347) ESP_Mesh_Lite_Comm: approved: 1
I (144345) ESP_Mesh_Lite_Comm: approved: 1
I (151550) local_control: System information, channel: 2, layer: 5, self mac: 60:55:f9:f5:99:4c, parent bssid: 60:55:f9:f5:9d:0d, parent rssi: -29, free heap: 261744
I (151555) local_control: WIFI_AP_IP : 192.168.7.1
I (151560) local_control: WIFI_STA_IP: 192.168.6.2
I (151566) local_control: Child mac: 60:55:f9:f5:9d:0c
I (154347) ESP_Mesh_Lite_Comm: approved: 1
I (164346) ESP_Mesh_Lite_Comm: approved: 1
I

The second node's log:
I (36870) wifi:station: 60:55:f9:f5:99:4c join, AID=1, bgn, 40U
I (36911) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.6.2
W (44507) vendor_ie: Scanning in progress, please try again later
E (50805) local_control: socket connect, ret: -1, ip: 192.168.3.18, port: 8070
E (50806) bridge_wifi: STA Disconnect to the AP
I (50808) Mesh-Lite: netif network segment conflict check
I (50814) local_control: WIFI_AP_IP : 192.168.6.1
I (50819) local_control: WIFI_STA_IP: 192.168.5.2
I (50824) bridge_wifi: STA Connecting to the AP again...
I (50830) Mesh-Lite: netif network segment conflict check
I (50836) local_control: WIFI_AP_IP : 192.168.6.1
I (50842) local_control: WIFI_STA_IP: 192.168.5.2
I (50848) ESP_Mesh_Lite_Comm: approved: 1
I (54345) ESP_Mesh_Lite_Comm: approved: 1
I (57654) wifi:state: run -> init (4a0)
I (57655) wifi:pm stop, total sleep time: 9111841 us / 26111001 us

I (57655) wifi:new:<2,1>, old:<2,1>, ap:<2,1>, sta:<2,1>, prof:1
I (61507) ESP_Mesh_Lite_Comm: Retry to connect to the AP
I (61511) wifi:primary chan differ, old=2, new=2, start CSA timer
I (61512) wifi:new:<2,1>, old:<2,1>, ap:<2,1>, sta:<2,1>, prof:1
I (61515) wifi:state: init -> auth (b0)
I (61520) wifi:state: auth -> assoc (0)
I (61528) wifi:state: assoc -> run (10)
I (61540) wifi:connected with ESP_Bridge_f5994d, aid = 1, channel 2, 40U, bssid = 60:55:f9:f5:99:4d
I (61541) wifi:security: WPA2-PSK, phy: bgn, rssi: -22
I (61544) wifi:pm start, type: 1

I (61547) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (61555) wifi:BcnInt:102400, DTIM:2
I (61564) local_control: System information, channel: 2, layer: 4, self mac: 60:55:f9:f5:9d:0c, parent bssid: 60:55:f9:f5:99:4d, parent rssi: -23, free heap: 261740
I (61574) local_control: WIFI_AP_IP : 192.168.6.1
I (61579) local_control: WIFI_STA_IP: 0.0.0.0
I (61584) local_control: Child mac: 60:55:f9:f5:99:4c
I (62229) esp_netif_handlers: sta ip: 192.168.7.2, mask: 255.255.255.0, gw: 192.168.7.1
I (62230) bridge_wifi: Connected with IP Address:192.168.7.2
I (62234) vendor_ie: RTC store: ssid:ESP_Bridge_f5994d; bssid:60:55:f9:f5:99:4d crc:1257172232
I (62243) local_control: WIFI_AP_IP : 192.168.6.1
I (62248) local_control: WIFI_STA_IP: 192.168.7.2
I (62254) local_control: Create a tcp client, ip: 192.168.3.18, port: 8070

tswen, 2023-06-15:
Hi, I have recently fixed some issues related to circular connections. Would you like to test them using this temporary [branch](https://github.com/tswen/esp-mesh-lite/tree/bugfix/fix_circular_recovery_issue)?

novflying, 2023-06-15:
Sure

closed 2023-08-01 by wujiangang