espressif/esp-mesh-lite #12: 自結點不斷連入斷開

8888alex, 2023-06-08:
我有三塊 ESP32-S2-DevKitM-1 v1.0以相同的程序刷入。
但僅其中特定一塊為ROOT時，其他子節點會不斷的join,leave，
想請問有沒有辦法排除這個問題。
-------------------------------------------
IDF v5.0.2
idf_component.yml
dependencies:
  idf: ">=5.0"
  mesh_lite:
    version: "*"
  espressif/esp_modem:
    version: 0.*
  espressif/esp_modem_usb_dte:
    version: 1.*
    public: true
    rules:
      - if: "target in [esp32s2, esp32s3]"

錯誤日誌
-------------------------------------------
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> set IDF_PATH=C:/Users/alex/esp/esp-idf-v5.0.2/
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2\build\aIoTSys.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (21) boot: compile time 19:35:18
I (21) boot: chip revision: v1.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (62) boot:  1 ota_0            OTA app          00 10 00010000 00100000
I (70) boot:  2 storage          Unknown data     01 22 00110000 000a0000
I (77) boot:  3 nvs              WiFi data        01 02 001b0000 00250000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=21fa4h (139172) map
I (125) esp_image: segment 1: paddr=00031fcc vaddr=3ffc6090 size=02db0h ( 11696) load
I (128) esp_image: segment 2: paddr=00034d84 vaddr=40022000 size=0b294h ( 45716) load
I (142) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=861ech (549356) map
I (252) esp_image: segment 4: paddr=000c6214 vaddr=4002d294 size=08df4h ( 36340) load
I (271) boot: Loaded app from partition at offset 0x10000
I (272) boot: Disabling RNG early entropy source...
I (283) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (283) cpu_start: Pro cpu up.
I (303) cpu_start: Pro cpu start user code
I (303) cpu_start: cpu freq: 160000000 Hz
I (303) cpu_start: Application information:
I (306) cpu_start: Project name:     aIoTSys
I (311) cpu_start: App version:      1
I (315) cpu_start: Compile time:     Jun  8 2023 19:35:08
I (321) cpu_start: ELF file SHA256:  f609da062eb1136a...
I (327) cpu_start: ESP-IDF:          v5.0.2-dirty
I (333) cpu_start: Min chip rev:     v0.0
I (337) cpu_start: Max chip rev:     v1.99 
I (342) cpu_start: Chip rev:         v1.0
I (347) heap_init: Initializing. RAM available for dynamic allocation:
[... 99 lines cut ...]
I (29533) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (30913) mesh_lite_main: System information, channel: 6, layer: 1, self mac: 70:04:1d:f7:cb:a6, parent bssid: b8:69:f4:8a:6f:b4, parent rssi: -73, free heap: 79316
I (32012) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (32013) wifi:station: 70:04:1d:f7:b4:78 join, AID=1, bgn, 40U
I (35021) wifi:station: 70:04:1d:f7:b4:78 leave, AID = 1, bss_flags is 131170, bss:0x3ffef4f4
I (35022) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (40913) mesh_lite_main: System information, channel: 6, layer: 1, self mac: 70:04:1d:f7:cb:a6, parent bssid: b8:69:f4:8a:6f:b4, parent rssi: -73, free heap: 79316
I (45292) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (45293) wifi:station: 70:04:1d:f7:b4:84 join, AID=1, bgn, 40U
I (48301) wifi:station: 70:04:1d:f7:b4:84 leave, AID = 1, bss_flags is 131170, bss:0x3ffef4f4
I (48302) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (50788) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (50789) wifi:station: 70:04:1d:f7:b4:78 join, AID=1, bgn, 40U
I (50913) mesh_lite_main: System information, channel: 6, layer: 1, self mac: 70:04:1d:f7:cb:a6, parent bssid: b8:69:f4:8a:6f:b4, parent rssi: -73, free heap: 77612
I (50918) mesh_lite_main: Child mac: 70:04:1d:f7:b4:78
I (53798) wifi:station: 70:04:1d:f7:b4:78 leave, AID = 1, bss_flags is 131170, bss:0x3ffef4f4
I (53798) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (55322) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (55323) wifi:station: 70:04:1d:f7:b4:84 join, AID=1, bgn, 40U
I (58330) wifi:station: 70:04:1d:f7:b4:84 leave, AID = 1, bss_flags is 131170, bss:0x3ffef4f4
I (58331) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (60831) wifi:new:<6,1>, old:<6,1>, ap:<6,1>, sta:<6,1>, prof:1
I (60832) wifi:station: 70:04:1d:f7:b4:78 join, AID=1, bgn, 40U
I (60913) mesh_lite_main: System information, channel: 6, layer: 1, self mac: 70:04:1d:f7:cb:a6, parent bssid: b8:69:f4:8a:6f:b4, parent rssi: -70, free heap: 77612
I (60918) mesh_lite_main: Child mac: 70:04:1d:f7:b4:78

tswen, 2023-06-09:
能否将三个设备的日志分别保存成三个文件上传到这里。

tswen, 2023-06-13:
fc0 含义为 STA 收到了 deauth 帧，reason 为密码错误

8888alex, 2023-06-14:
這三塊版子都是相同的燒錄
ELF file SHA256:  763c4052091687f3...
[image]
密碼設置也都相同
[image]

測試步驟為
1.先啟動b1待網路就緒後才啟動b2，待b2就緒後再啟動b3
此時網路建構正常
2.b2保持，b1及b3按下boot+reset進入 rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0/1/SPI))
脫離網路
3.等待b2重新連上AP
E (76443) mqtt_client: Poll read error: 119, aborting connection
I (77013) ESP_Mesh_Lite_Comm: Retry to connect to the AP
4.b1及b3按下reset重啟，
這時b1及b3開始不段的反覆加入離開b2

tswen, 2023-06-16:
Hi, I have recently fixed some issues related to circular connections. Would you like to test them using this temporary [branch](https://github.com/tswen/esp-mesh-lite/tree/bugfix/fix_circular_recovery_issue)?

8888alex, 2023-06-29:
之前的版本也會發生突然脫離mesh，但一直沒法抓到LOG。
因為當開啟Monitor的時候等同重啟，而每次重起便恢復聯網，且無法重現錯誤。
這次的情況也是重啟後便恢復聯網，只是剛好正在Monitor所以有抓到LOG。
錯誤日誌如下，
[buglog.txt](https://github.com/espressif/esp-mesh-lite/files/11908084/buglog.txt)

通常上殿後的待機狀態會停在
I (22997) ESP_Mesh_Lite_Comm: approved: 1

同時間有兩片離線但另外一片有自動恢復聯網
下面是另外一片的日誌
[另外一片.txt](https://github.com/espressif/esp-mesh-lite/files/11908189/default.txt)

我的這個工程對網路的穩定有一定程度的要求，請問我要如何確保發生離線後能自動恢復連接?
謝謝您的時間。

tswen, 2023-06-30:
我注意到错误的日志里面，Wi-FI 成功连接上了，但是没有获取到 IP 地址。
最近我们也有别的客户遇到类似的问题，通过以下办法都解决了：
1. 更新 idf 到最新的 release/v5.0，最近 idf 修复了一些 Wi-Fi 的 bug，解决了某些情况下获取不到 IP 的问题。
2. 因为目前 PMF 还不太完善，使用 Mesh 方案需要将 PMF 禁用掉，在 esp_bridge_create_all_netfi 之后调用以下 API
esp_wifi_disable_pmf_config(WIFI_IF_STA);
esp_wifi_disable_pmf_config(WIFI_IF_AP);

8888alex, 2023-06-30:
> * 更新 idf 到最新的 release/v5.0，最近 idf 修复了一些 Wi-Fi 的 bug，解决了某些情况下获取不到 IP 的问题。
> * 因为目前 PMF 还不太完善，使用 Mesh 方案需要将 PMF 禁用掉，在 esp_bridge_create_all_netfi 之后调用以下 API
>   esp_wifi_disable_pmf_config(WIFI_IF_STA);
>   esp_wifi_disable_pmf_config(WIFI_IF_AP);

已按方服用，有問題再回報，謝謝您。

closed 2023-06-30 by 8888alex