espressif/esp-mesh-lite #15: **ERROR*** A stack overflow in task Tmr Svc has been detected.

287662354, 2023-06-19:
<= [2023-06-19 16:42:02.766]: I (20500) wifi:<ba-add>idx:1 (ifx:0, f4:12:fa:f8:d2:d5), tid:2, ssn:0, winSize:64

<= [2023-06-19 16:42:02.990]: 
***ERROR*** A stack overflow in task Tmr Svc has been detected.

I (10686) vendor_ie: Mesh ID: 77
W (10686) vendor_ie: Error Get[4354]
W (10696) vendor_ie: Error Get[4354]
I (10696) ESP_Mesh_Lite_Comm: msg action add success
I (00:00:09.682) Mesh-Lite: Mesh-Lite connecting
I (10706) ESP_Mesh_Lite_Comm: Bind Socket 54, port 6364
wifi_lite

  when  second  device  connect  fist one  then    ***ERROR*** A stack overflow in task Tmr Svc has been detected.
   erverytime

287662354, 2023-06-20:
问题已解决  默认的tmr 任务堆栈默认只有2K左右  手动调节大一点已解决

closed 2024-02-06 by xcguang

iwonako, 2024-12-18:
用 `idf.py create-project-from-example "espressif/mesh_lite=*:mesh_local_control"` 直接创建 mesh lite 工程，能编译且成功组网。

但是在已有工程用 `idf.py add-dependency "espressif/mesh_lite=*"` 添加组件，相同的程序复制过去，编译报错 `CONFIG_MESH_CHANNEL`未定义。

打开两个工程的 sdk 比较，发现没有根节点定义 ‘CONFIG_MESH_ROOT’， `configTIMER_TASK_STACK_DEPTH` 的预设值也不同，使用的时候需要额外注意。