espressif/esp-mesh-lite #17: 當ROOT失敗後網路重建的時間會隨網路規模變大而增加

8888alex, 2023-07-08:
說明:
--------------------------------------------------------------------
當我使用三塊版子組網，ROOT失敗網路完成重建時間約需要50秒
但是當我用九塊板子組網時，網路基本無法恢復。(過了10分鐘還沒完成)
我的項目節點規模預計是200+，若按目前重建的算法顯然會有問題。
不知能否有多個ROOT或者備援ROOT之類的解決方案，以便在ROOT失敗時能快速恢復網路。
謝謝。

環境:
--------------------------------------------------------------------
IDF v5.0.2
blink 範例專案
idf_component.yml
dependencies:
    idf: ">=5.0"
    mesh_lite:
        version: "*"
    espressif/esp_modem:
        version: 0.*
    espressif/esp_modem_usb_dte:
        version: 1.*
        public: true
        rules:
        - if: "target in [esp32s2, esp32s3]"
    espressif/led_strip: "^2.0.0"

工程內容使用 [branch](https://github.com/tswen/esp-mesh-lite/tree/bugfix/fix_circular_recovery_issue)
並按 [禁用PMF](https://github.com/espressif/esp-mesh-lite/issues/12#issuecomment-1614039968) 調整

紀錄:
--------------------------------------------------------------------
三塊板子組網紀錄如下 
[b3.txt](https://github.com/espressif/esp-mesh-lite/files/11987346/b3.txt)(ROOT)
[b1.txt](https://github.com/espressif/esp-mesh-lite/files/11987340/b1.txt)
[b2.txt](https://github.com/espressif/esp-mesh-lite/files/11987343/b2.txt)

擴充到9塊板子組網的紀錄如下(僅保留增加的紀錄)
[b3-1.txt](https://github.com/espressif/esp-mesh-lite/files/11987354/b3-1.txt)(ROOT)
[b2-1.txt](https://github.com/espressif/esp-mesh-lite/files/11987353/b2-1.txt)

tswen, 2023-07-11:
Hi，可以使用这个[分支](https://github.com/tswen/esp-mesh-lite/tree/bugfix/fix_circular_recovery_issue)新的提交再试下吗，后面做了一些优化。
因为当根节点意外重启时，其有可能连接到别的节点形成一个环形连接，打破环形连接然后重新恢复新的组网拓扑需要是的时间，我这边测试 12个设备大概在六十多秒左右，不会出现一直无法恢复的情况。

tswen, 2023-07-11:
另外对于你们的场景，不是很建议在一个 mesh 组网中布置 200+ 个设备，建议为分为多个 mesh，仅需要通过不同的 Mesh-ID 来进行区分，比如分为 4 组 Mesh-ID 不同的四个 Mesh 组网拓扑，每组 Mesh 网络内 50 个设备，这样的稳定性会比一个组网内 200+ 设备好很多。

8888alex, 2023-07-12:
改用这个[分支](https://github.com/tswen/esp-mesh-lite/tree/bugfix/fix_circular_recovery_issue)後
初步測試大致可以恢復連接，還是會有個別的情況需要費時超過五分鐘才完成連接。
如圖中的96399c
[image]
96399c的紀錄
[96399c.txt](https://github.com/espressif/esp-mesh-lite/files/12023983/96399c.txt)
更甚至f7b484費時約13分鐘才恢復連接，但因僅連接電源上電故沒有日誌。
後續再做更多測試後再行回報。
另，首次測試ROOT失敗時，一度似乎所有板子都在競爭ROOT，所有板子頻繁反覆重啟，不知是否因為同步上電的關係。
(因USB口不夠用，其中六片我使用麵包板同步上電。)

8888alex, 2023-07-12:
> 另外对于你们的场景，不是很建议在一个 mesh 组网中布置 200+ 个设备，建议为分为多个 mesh，仅需要通过不同的 Mesh-ID 来进行区分，比如分为 4 组 Mesh-ID 不同的四个 Mesh 组网拓扑，每组 Mesh 网络内 50 个设备，这样的稳定性会比一个组网内 200+ 设备好很多。

在實際的應用場景中可能因為小卡宿主設備的尺寸、場所動線及AP分布等因素導致，導致設備的分布密度不均。
例如在兩立方米的空間中，可能存在1~24個以上的設備，
這個部分能否自動(或有條件的自動或半自動，按信號強度節點深度等或者其他條件)分組，
手動分配的話，目前暫時沒找到在哪查詢信號強度的數據還請撥冗告知，謝謝。

tswen, 2023-07-26:
> > 另外对于你们的场景，不是很建议在一个 mesh 组网中布置 200+ 个设备，建议为分为多个 mesh，仅需要通过不同的 Mesh-ID 来进行区分，比如分为 4 组 Mesh-ID 不同的四个 Mesh 组网拓扑，每组 Mesh 网络内 50 个设备，这样的稳定性会比一个组网内 200+ 设备好很多。
> 
> 在實際的應用場景中可能因為小卡宿主設備的尺寸、場所動線及AP分布等因素導致，導致設備的分布密度不均。 例如在兩立方米的空間中，可能存在1~24個以上的設備， 這個部分能否自動(或有條件的自動或半自動，按信號強度節點深度等或者其他條件)分組?
> 
> 謝謝您的時間

目前分组主要依靠的是 Mesh-ID 和路由器信息，只要这两个参数相同，便可以组成同一个 Mesh-Lite 网络，任一参数不同就会分成不同的 Mesh-Lite 网络，对于不同的 Mesh 组的划分，建议您在配置设备时指定设备的 Mesh-ID 以及路由器信息来为设备划分不同的 Mesh-Lite 网络。

tswen, 2023-07-26:
> 這個問題依然存在。 就我的觀察，ROOT穩定存在便相安無事，但只要ROOT一失敗，所有的節點便不斷的競爭試圖成為ROOT，如圖。 [image] 具體的投票機制我不清楚，但目前的算法會不斷重組，導致資料無法順利傳遞，還不如犧牲一點信號強度，保持整體的資料傳遞穩定。

感谢您的反馈。有一点我想确认的是，ROOT 失败是指 “ROOT 节点下电” 还是 “ROOT 节点软复位重启” 还是 “ROOT 节点硬重启”。因为这三种对应三种不同的情况。
- “ROOT 节点下电”：二级节点中相对路由器信号强度最好的设备在重连 ROOT 失败后会先去连接路由器，其余设备后面在重连ROOT 失败多次后也会重新去寻找新的可选父节点，最终恢复网络。
- “ROOT 节点软复位重启”：ROOT 节点会根据 RTC_Mem 内部存储的信息重连路由器，二级节点也会在重连 ROOT 节点的过程中恢复网络。
- “ROOT 节点硬重启”：硬重启大概率会导致最初形成一个环形连接，不过会在内部的一些环形连接检测机制的作用下破除该环形连接，随后再恢复成正常的 Mesh 网络。

其中严重情况以及恢复时间长度：“ROOT 节点硬重启” >  “ROOT 节点下电” > “ROOT 节点软复位重启”
因为这个涉及到比较严重的网络拓扑破坏，Mesh 内部是需要花一定时间去修复网络结构的。而且理论上这种结构破化不会是频繁出现的。不过我们也会致力于优化网络修复的时间。

8888alex, 2023-07-26:
> 目前分组主要依靠的是 Mesh-ID 和路由器信息，只要这两个参数相同，便可以组成同一个 Mesh-Lite 网络，任一参数不同就会分成不同的 Mesh-Lite 网络，对于不同的 Mesh 组的划分，建议您在配置设备时指定设备的 Mesh-ID 以及路由器信息来为设备划分不同的 Mesh-Lite 网络。

好的，這個部份我再另行評估合適方案。謝謝。
眼前因競爭ROOT導致不斷重啟，網絡重組期間資料無法順利傳遞進而導致資料延遲。
這個問題比較大還請設法排除，謝謝。

8888alex, 2023-07-26:
> 感谢您的反馈。有一点我想确认的是，ROOT 失败是指 “ROOT 节点下电” 还是 “ROOT 节点软复位重启” 还是 “ROOT 节点硬重启”。因为这三种对应三种不同的情况。
>
> * “ROOT 节点下电”：二级节点中相对路由器信号强度最好的设备在重连 ROOT 失败后会先去连接路由器，其余设备后面在重连ROOT 失败多次后也会重新去寻找新的可选父节点，最终恢复网络。
> * “ROOT [...]

我目前試過三種
1.root直接斷電停機
2.程序中調用esp_restart()
3.按下實體reset重啟root
這三種皆會引發進入競爭

目前我測試最快恢復組網的方法是所有的都斷電或者boot+reset進入等待，
只先上電一個，等他接上網路後其他的一起上電。

8888alex, 2023-08-30:
這個問題依然存在
下面是最新的LOG
這次僅有三塊模組卻依然無窮等待
[LOG.txt](https://github.com/espressif/esp-mesh-lite/files/12471650/LOG.txt)

8888alex, 2023-09-01:
今天的情況是原本好好的網路突然一次上電四片(不知道數量是不是原因)
然後就導致整個網路崩潰。
LOG裡面可以看到重啟的這片一開始有加入且成功發送MQTT資訊然後就進入不斷重新連接AP
幾次重啟都一樣所以可以抓到這個LOG
[LOG.txt](https://github.com/espressif/esp-mesh-lite/files/12500097/LOG.txt)
您在研究看看

8888alex, 2023-09-03:
關於ROOT推選的算法(因為不知道具體的過程所以純屬瞎猜亂說，如有謬誤還請指正)
目前我看到的過程(純粹視覺上看到的)是每個節點都嘗試成為ROOT(返回Level 1)
[image]
[image]
然後又有另外一個NODE上來回報為LEVEL1，每次迭代的過程約5-10秒不等，我說我的項目預設規模為200-300個單位。
所以即便是偶發性的，因為設備斷電或升級等原因導致崩潰需要重組網絡，那也是很可怕的事情。
這也是我的項目一直沒趕上線的主因(其他的問題陸續排除得差不多了)
在我有限的知識範圍內，我有個思路不知可不可行，
首先,mesh對我而言最大的優勢就是不需要另外架設中繼AP，因為每個NODE都是中繼AP。
其次您之前有提過Mesh-ID，我覺得Mesh-ID應該是可以自動生成的。
既然可以有多個Mesh-ID表示可以有多個ROOT去連接AP。
基於不要吊死在同一棵樹上，下面是我的一點想法。
首先組網不是基於固定的Mesh-ID，而是先把節點當是一般的AP連接。
(好吧先忘了NAT，也先忘了這樣連接能不能跟父節點對話...)
每個節點先利用 esp_wifi_scan_get_ap_records() 一次性的各自取得自己與AP或其他節點的Rssi(可以用約定的SSID前綴或其他方法識別)，這裡還需要一份nodeTable(含Rssi)用來記錄自己的子孫節點(用來避免循環連接)。
信號最好且不在nodeTable中的就直接當是父節點進行連接。
連接父節點後便向父節點遞交自己的nodeTable，
這時可能發現出現循環連接，在循環連接中尋找信號最好的外部節點。通告脫離然後改換連接的父節點。
連接父節點繼續重複以上的檢查動作。
過程中如果節點對AP的RSSI優於對其他節點便是候選ROOT，先直連AP以減少資料傳遞的中斷時間。
最後就是ROOT，我一直覺得同時存在多個ROOT並沒有問題，該廣播的用廣播的也是全網都收得到。
關鍵是不會因為一個ROOT崩潰就導致全網崩潰無法傳送訊息。
只要能連AP就是傳送資料就是王道。要是限定只能有一個ROOT反而限制了mesh的優勢。
所以最後就是ROOT，
我不清楚mesh是否一定要基於Mesh-ID才能正常運作，若需要則可由ROOT廣播，現在設立了一個新的Mesh-ID，然後基於路由表，這個Mesh-ID底下有哪些節點以及節點間的從屬關係。然後就地解散按照路由表重新用這個Mesh-ID重新組網。
以上不知可不可行?

tswen, 2023-09-05:
> 今天的情況是原本好好的網路突然一次上電四片(不知道數量是不是原因) 然後就導致整個網路崩潰。 LOG裡面可以看到重啟的這片一開始有加入且成功發送MQTT資訊然後就進入不斷重新連接AP 幾次重啟都一樣所以可以抓到這個LOG 直到我把後上電的四片關閉，原有的網路就恢復了。 然後我再重新上電那四片，就可以正常加入網路，也沒有造成崩潰。 下面是重啟的那片LOG，您再研究看看 [LOG.txt](https://github.com/espressif/esp-mesh-lite/files/12500097/LOG.txt)

您能使用最新的 https://github.com/espressif/esp-mesh-lite 再试下吗，因为最新的 lib 修复了一些已知的问题，感觉您上面提到的都经过修复了。另外新的 lib 有个地方应用层需要更新，外部需要在 esp_mesh_lite_init 后调用 esp_mesh_lite_start。

tswen, 2023-09-05:
决定一个 mesh-lite 网络的两个因素是 `路由器` 和 `Mesh-ID`，若这两个因素都匹配，设备就会处于同一个 Mesh-lite 组网里。另外 ESP-Mesh-Lite 跟之前的 ESP-WIFI-Mesh 不同，每个设备都是通过 napt 为其下层节点提供向上访问网络的途径，但是并不维护其下挂节点的一个路由表。
Mesh-Lite 内部已经做了一些有关避免形成环形链接以及环形链接解除的一些策略，一般需要等待几十秒或者几分钟的时间来恢复成正常的网络拓扑（这取决于 Mesh 网络内设备数量），所以不建议在一个 Mesh 网络内布置很多设备，可以将 200 个设备分为 4-5 个不同的 Mesh 网络。通过 esp_mesh_lite_config_t 中的 mesh_lite 参数或者 esp_mesh_lite_set_mesh_id() 来设置。

8888alex, 2023-09-07:
> 您能使用最新的 https://github.com/espressif/esp-mesh-lite 再试下吗
(您這連結是master的版本(我之前用的tswen的分支))

好的，我改用master版再試看看。

8888alex, 2023-09-08:
> 决定一个 mesh-lite 网络的两个因素是 `路由器` 和 `Mesh-ID`，若这两个因素都匹配，设备就会处于同一个 Mesh-lite 组网里。另外 ESP-Mesh-Lite 跟之前的 ESP-WIFI-Mesh 不同，每个设备都是通过 napt 为其下层节点提供向上访问网络的途径，但是并不维护其下挂节点的一个路由表。 Mesh-Lite [...]

好的。
我的想法是要可以透過介面去調整Mesh-ID，調整後退出當前的樹然後去加入另一組Mesh-ID的樹，
剛發現我無法這樣變更 Mesh-ID ...... 這要另開一個Issues嗎?

[image]

tswen, 2023-09-08:
你只需要在 esp_mesh_lite_init 和 esp_mesh_lite_start 之间调用 esp_mesh_lite_set_mesh_id(x, true) 就好了

8888alex, 2023-09-08:
> 你只需要在 esp_mesh_lite_init 和 esp_mesh_lite_start 之间调用 esp_mesh_lite_set_mesh_id(x, true) 就好了

這個有效。謝謝。

8888alex, 2023-09-09:
> 你只需要在 esp_mesh_lite_init 和 esp_mesh_lite_start 之间调用 esp_mesh_lite_set_mesh_id(x, true) 就好了

剛發現變更設定後，ID是變了。
但是使用
esp_mesh_lite_start()
esp_mesh_lite_connect();
esp_restart();
都無法取的新的LEVEL。
例如某個LEVEL3的節點從mesh_id 77換到88
LEVEL還是保持LEVEL3(只有單一片應該要取得LEVEL1加入已經存在的TREE也應該取得相應的LEVEL)
但試了上面這三個方法也不會改變，只有按下硬體的RESET才會重新取得正確的LEVEL。

tswen, 2023-09-11:
如果是在初始化时设定，最好是在 esp_mesh_lite_init 之后和 esp_mesh_lite_start 之前设置 mesh id。
如果是在设备运行中途设定，设定完之后想要其转移到另外一个网络。需要先执行 esp_wifi_disconnected() 先将其与当前父节点断开连接。然后执行 esp_mesh_lite_connect();

8888alex, 2023-09-11:
> 如果是在初始化时设定，最好是在 esp_mesh_lite_init 之后和 esp_mesh_lite_start 之前设置 mesh id。 如果是在设备运行中途设定，设定完之后想要其转移到另外一个网络。需要先执行 esp_wifi_disconnected() 先将其与当前父节点断开连接。然后执行 esp_mesh_lite_connect();

            esp_mesh_lite_set_mesh_id(atoi(data), true);
            esp_wifi_disconnect();
            esp_mesh_lite_connect();
            
          沒有作用 :(

tswen, 2023-09-14:
不好意思，没有及时回复。我这边测试了下，设备断开后确实仍连接了源父节点，主要是因为断开事件触发了内部重连父节点机制。我后面会进行优化这部分逻辑。
针对你们这种场景，您目前可以尝试以下方式：
设置完 mesh id 之后，设置一下 wifi 的路由器配置，可以是另外一个路由器的也可以是当前路由器的，之后再进行 wifi 的断开和连接操作。
```c
esp_mesh_lite_set_mesh_id()
wifi_config_t wifi_config = {
        .sta = {
            .ssid = CONFIG_ROUTER_SSID,
            .password = CONFIG_ROUTER_PASSWORD,
        },
    };
esp_bridge_wifi_set(WIFI_MODE_STA, (char *)wifi_config.sta.ssid, (char *)wifi_config.sta.password, NULL);
esp_wifi_disconnect()
esp_wifi_connect()
```

closed 2023-09-21 by 8888alex