espressif/esp-mesh-lite #18: ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space

xiao-orien, 2023-08-03:
(16716) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
E (16726) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
E (16736) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
E (16746) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
I (16756) ESP_Mesh_Lite_Comm: Retry to connect to the AP
E (16756) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
I (16766) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,2>, prof:1
I (16766) wifi:state: init -> auth (b0)
E (16766) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
I (16766) wifi:state: auth -> assoc (0)
E (16776) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
E (16786) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
E (16796) ESP_Mesh_Lite_Comm: Error occurred during sending: errno               <=====return error
I (16796) wifi:state: assoc -> run (10)
E (16806) ESP_Mesh_Lite_Comm: Error occurred during sending: errno 
I (16816) wifi:connected with ESP_Bridge, aid = 1, channel 6, 40D, bssid = f4:12:fa:e2:0b:b9
I (16816) wifi:security: WPA2-PSK, phy: bgn, rssi: -17
I (16816) wifi:pm start, type: 1

I (16826) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (16906) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (17176) wifi:<ba-add>idx:0 (ifx:0, f4:12:fa:e2:0b:b9), tid:0, ssn:0, winSize:64
I (17836) esp_netif_handlers: sta ip: 192.168.4.4, mask: 255.255.255.0, gw: 192.168.4.1
I (17836) bridge_wifi: Connected with IP Address:192.168.4.4
I (17836) ip select: IP Address:192.168.5.1
I (17836) ip select: GW Address:192.168.5.1
I (17846) ip select: NM Address:255.255.255.0
I (17856) bridge_common: ip reallocate new:192.168.5.1
W (17856) bridge_wifi: SoftAP IP network segment has changed, deauth all station
I (17866) vendor_ie: RTC store: ssid:ESP_Bridge; bssid:f4:12:fa:e2:0b:b9 crc:1787568907
I (17876) esp_mqtt_glue: AWS PPI: ?Platform=APN3|A0|RM|EX00|RMDev|1x0|F412FAE6CCD8
I (17886) esp_mqtt_glue: Initialising MQTT
E (17886) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space              <=====return error
I (17886) esp_rmaker_mqtt_budget: MQTT Budgeting initialised. Default: 100, Max: 1024, Revive count: 16, Revive period: 5
I (17906) esp_mqtt_glue: Connecting to mqtt://iot.qylian.cn
I (17916) esp_rmaker_core: Waiting for MQTT connection
I (18066) esp_mqtt_glue: MQTT Connected
W (18066) app_rainmaker: Unhandled RainMaker Common Event: 3
I (18136) Mesh-Lite: netif network segment conflict check
E (497056) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space              <=====return error
E (497066) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space
E (497076) ESP_Mesh_Lite_Comm: Error occurred during sending: errno Not enough space

root nod side is :
E (11511981) bridge_wifi: STA Disconnect to the AP
I (11511991) wifi:<ba-del>idx
I (11511991) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (11511991) wifi:station: f4:12:fa:e6:cc:d8 join, AID=1, bgn, 40D
I (11516001) wifi:station: f4:12:fa:e6:cc:d8 leave, AID = 1, bss_flags is 658530, bss:0x3fcae0a4
I (11516001) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (11523171) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1
I (11523171) wifi:station: f4:12:fa:e6:cc:d8 join, AID=1, bgn, 40D
I (11523191) bridge_wifi: STA Connecting to the AP again...
I (11523251) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.4
I (11524271) wifi:<ba-add>idx:2 (ifx:1, f4:12:fa:e6:cc:d8), tid:0, ssn:0, winSize:64
I (11524271) app_bridge: [Child Info Join]  Level:2  Mac:f4:12:fa:e6:cc:d8 IP 192.168.4.4

I (12004301) ESP_Mesh_Lite_Comm: approved: 1

tswen, 2023-08-04:
Hello, because Mesh nodes internally have some communication to synchronize information and maintain the Mesh network state, communication failures may occur when the network structure changes. However, there is no need to worry, once the network structure stabilizes, communication will resume normally.

xiao-orien, 2023-08-04:
OK.Thankyou.

116426319wzk, 2023-10-25:
E (08:01:10.125) mesh: connect -1
E (08:01:10.126) mesh: Socket unable to connect: errno 113

当我的节点在 layter  2 时无法访问 tcp server 。说是没有路由，同时也出现上面的错误信息。
这个问题你们有遇到过吗？

tswen, 2023-10-26:
你是直接使用 mesh_local_control demo 跑的吗，外部 tcp server 的 ip 和 port 配置正确吗？如果是自己移植的 mesh_lite 组件到自己的工程，需要注意是否开启以下配置项：
# LWIP
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

CONFIG_MESH_LITE_ENABLE=y