espressif/esp-mesh-lite #20: Failed to resolve component 'usb_device'

xiao-orien, 2023-08-14:
I have tried configuring this new version many times. Always report this error ‘Failed to resolve component 'usb_device'’.

D:\Espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory 'd:\espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.

D:\Espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'd:\espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory d:\espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 d:\espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light"...
-- Found Git: D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Skipping optional dependency: espressif/cbor
Processing 20 dependencies:
[1/20] app_insights (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[2/20] app_reset (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[3/20] esp_diagnostics (d21bfb3854e057adc65bf4403d2da3f0f2231020)
[4/20] esp_insights (d21bfb3854e057adc65bf4403d2da3f0f2231020)
[5/20] esp_rainmaker (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[6/20] esp_schedule (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[7/20] espressif/cmake_utilities (0.5.0)
[8/20] espressif/esp_modem (0.1.28)
[9/20] espressif/esp_modem_usb_dte (1.1.0~1)
[10/20] espressif/iot_bridge (0.7.1)
[11/20] espressif/lightbulb_driver (0.2.2)
[12/20] espressif/usb_host_cdc_acm (2.0.0)
[13/20] gpio_button (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[14/20] idf (4.4.4)
[15/20] json_generator (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[16/20] json_parser (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[17/20] mesh_lite (0.4.0)
[18/20] qrcode (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[19/20] rmaker_common (4f5a8860dc4043e005fefbb2f323355a435cf47c)
[20/20] rtc_store (d21bfb3854e057adc65bf4403d2da3f0f2231020)
-- DEBUG: Use esp-modbus component folder: D:/Espressif/frameworks/esp-idf-v4.4.4/components/freemodbus.
Skipping optional dependency: espressif/cbor
Skipping optional dependency: espressif/mdns
CMake Error at D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'usb_device'.
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:233 (__build_expand_requirements)
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:228 (__build_expand_requirements)
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:507 (__build_expand_requirements)
  D:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:19 (project)

-- Configuring incomplete, errors occurred!
See also "D:/Espressif/frameworks/mesh-lite/esp-mesh-lite/examples/rainmaker_led_light/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

D:\Espressif\frameworks\mesh-lite\esp-mesh-lite\examples\rainmaker_led_light>

tswen, 2023-08-14:
You can add this to the yml file in the example main. Just wait for the iot-bridge component to be updated to another version. MR has been raised, and you can temporarily use the above method to change it.

 # Wait until the `usb_device` component is listed in the package manager
  # before adding the dependency to the `yml` file of the `iot_bridge` component.
  #
  usb_device:
    path: components/usb/usb_device
    git: https://github.com/espressif/esp-iot-bridge.git
    rules:
      - if: "target in [esp32s2, esp32s3]"

xiao-orien, 2023-08-14:
ok,thankyou.

287662354, 2023-08-22:
感觉这个组件用起来好麻烦  有错误半天都找不到问题  之前 esp_modem_usb_dte 还一直报HTTP错误