espressif/esp-mesh-lite #22: 如何修改为自建服务器OTA升级

287662354, 2023-08-23:
源码里面提供了三个有关于OTA升级的接口但是没有描述如何使用 ,文档和源码也是基于rainmaker,请问有办法修改为自建服务器吗,或者有详细的API文档

tswen, 2023-08-23:
可以参考如下文档章节：
https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota

8888alex, 2023-08-29:
> 可以参考如下文档章节： https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota

您好
我迷失在 esp_mesh_lite_ota_from_extern_url
我原本的架構試 服務器用MQTT通知下載檔案的URL
e.q. url=http://192.168.0.100/uploads/be9f08bb2ec50b8d.bin
然後使用esp_http_client_init進行下載

請問這裡我該如何使用esp_mesh_lite_ota_from_extern_url替換我原本的URL?

tswen, 2023-08-30:
其实 esp_mesh_lite_ota_from_extern_url 是一个你需要自己实现的函数，函数的用处是设备独立地从外部 URL 直接获取新版本固件并进行 OTA 的流程。
在执行 esp_mesh_lite_ota_start() 时，将该函数指针传递进去，设备会有线执行局域网 OTA（即向其上级节点请求固件），如果请求不到则会通过您传递的函数指针执行原有的 OTA 流程。
参考链接：https://github.com/espressif/esp-mesh-lite/blob/master/examples/rainmaker_led_light/components/app_rainmaker/app_rainmaker_ota_topic.c#L34

8888alex, 2023-08-31:
謝謝回覆，我再試試。

8888alex, 2023-09-01:
您好，
我可以在console看到下面的打印，但我還想要攔截這個下載通知以便可以發送MQTT回報進度，不知可以怎麼做?
I (8640) mesh_lite_ota: OTA Image version for Project: v0.1.0.11. Expected: v0.1.0.12
I (11123) ESP_Mesh_Lite_OTA: Image bytes read: 102712
I (13447) ESP_Mesh_Lite_OTA: Image bytes read: 205424
I (15736) ESP_Mesh_Lite_OTA: Image bytes read: 308136
I (17918) ESP_Mesh_Lite_OTA: Image bytes read: 410848
I (19983) ESP_Mesh_Lite_OTA: Image bytes read: 512172
...

tswen, 2023-09-04:
根据日志看出来是，这个根节点在 OTA 更新完成之后重启且连接了路由器，但是由于路由器下同时仍存在相同 mesh id 的其他根节点设备（有可能是有别的设备意外连接到了路由器），经过 “竞争融合” 机制，该 ROOT 相对于路由器信号强度弱于另一个根节点，所以其与路由器断开，重新去寻找新的父节点了。
如果你一定要求其作为根节点，可以在固件中使用 esp_mesh_lite_set_allowed_level(1)，将其固定为根节点，它就不会因为某些机制而动态变化了。

8888alex, 2023-09-05:
請問，我是否可以在更新(或重啟)前先 esp_mesh_lite_set_allowed_level(1) 將其固定，以避免這種意外。
等重啟後再下 esp_mesh_lite_set_allowed_level(0) 恢復不固定?

tswen, 2023-09-05:
这样还是会有可能出现根节点切换的问题。因为 esp_restart 属于软重启，软重启节点会先去连接源父节点（也就是路由器），所以重启之前不管是否设置 esp_mesh_lite_set_allowed_level(1) 根节点都还是会连接路由器。之后将再恢复不固定，还是有可能因为别的信号强度更好的设备连接到路由器触发融合，让这个根节点设备变成子节点。不过这种机制也是为了让拓扑更加稳定，毕竟根节点相对于路由器的信号强度越强越好嘛！

8888alex, 2023-09-07:
信號好的當ROOT當然是最好的，其實關鍵是網路重組的速度。