espressif/esp-mesh-lite #26: Getting Error: The component usb_device could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component. Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command. Refer to the migration guide for more details about moved components. Refer to the build-system guide for more details about how components are found and included in the build.

Teesmo, 2023-09-06:
I have been exploring esp-mesh-lite for some months now, and had it working well with some modifications I made in the local_control.c file. I made some changes now (IP HOST address), and when I try to set-target to my board (esp32s2) using idf.py, I get this error

```
CMake Error at /home/onepower/esp/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'usb_device'.
Call Stack (most recent call first):
  /home/onepower/esp/esp-idf/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  /home/onepower/esp/esp-idf/tools/cmake/build.cmake:288 (__build_expand_requirements)
  /home/onepower/esp/esp-idf/tools/cmake/build.cmake:288 (__build_expand_requirements)
  /home/onepower/esp/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  /home/onepower/esp/esp-idf/tools/cmake/project.cmake:530 (idf_build_process)
  CMakeLists.txt:7 (project)

```

```
-- Configuring incomplete, errors occurred!
See also "/home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/CMakeFiles/CMakeOutput.log".
```
HINT: The component usb_device could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the /home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/log/idf_py_stderr_output_2625994 and /home/onepower/projects/esp/esp-mesh-lite/examples/mesh_local_control/build/log/idf_py_stdout_output_2625994

I never had issues before with setting the target until this happened, today. I tried finding the missing component (usb_device) in the registry, but it does not exist.

tswen, 2023-09-07:
Hello, please update esp-mesh-lite to the latest, the latest master branch has fixed this problem.
issue link: https://github.com/espressif/esp-mesh-lite/issues/20

Teesmo, 2023-09-10:
I tried it and it worked, thanks

closed 2023-09-11 by Teesmo