espressif/esp-mesh-lite #30: No-router example Core 0 panic

Mbadra96, 2023-09-16:
I've uploaded the no-router example as a child node and it logged everything was okay until `I (2097) Mesh-Lite: Mesh-Lite connecting
I (4947) vendor_ie: Mesh-Lite Scan done` after that Core 0 panicked (Double exception). with backtrace `  #0  0x4008c797:0x3ffbe260 in _xt_context_save at C:\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/xtensa_context.S:202
  #1  0x4015ed03:0x400803c0 in esp_event_loop_run at C:\.platformio\packages\framework-espidf\components\esp_event/esp_event.c:601 (discriminator 3)`

tswen, 2023-09-18:
Can you provide the following information to help us better analyze:
idf version and commit id:
Chip used:

Mbadra96, 2023-09-18:
idf version: I'm using platformIO which uses espressif32@6.3.2, so idf version should be ESP-IDF v5.1.1
commit id: d34d8c33d02b5292b678e14ef51631e10dfd3e5a
chip used: esp32

tswen, 2023-09-19:
Currently, the esp-mesh-lite solution does not support idf5.1. You can compile with idf5.0 first. We are already working on supporting idf5.1 and will release it soon.