espressif/esp-mesh-lite #31: no Internet access from devices connected to the router

hard-cat, 2023-09-22:
the root device is connected to a Wi-Fi router with the Internet, the second board is connected to the root device, this can be seen in the logs.
- The root device successfully completes the http request
- The connected device to the root device cannot fulfill the HTTP request
- local access via UDP between devices is available
- The firmware is the same in both boards with ESP32
- ESP-IDF v4.4.1-dirty
- esp-mesh-lite 0.7

tswen, 2023-09-22:
Hi, could you please confirm if the following configuration items are enabled:
```
# LWIP
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

CONFIG_MESH_LITE_ENABLE=y`
```

Also, it is recommended to use esp-idf/release5.0 or esp-idf/v4.4.5, as the latest tag versions have fixed some wifi bugs.

hard-cat, 2023-09-22:
yes, these options have been disabled:
CONFIG_LWIP_IP_FORWARD=n
CONFIG_LWIP_IPV4_NAPT=n

enabling these options solved the problem, thanks :-) I will test further

xcguang, 2023-09-22:
Could you please close this issue? If you have any other question, please raise a new one.

closed 2023-09-24 by hard-cat