espressif/esp-mesh-lite #33: 671702a 版本進行meshOTA

8888alex, 2023-09-28:
### 說明
基於 esp_mesh_lite_start 使用 671702a 版本進行 OTA
在三塊開發版穩定取得mesh情況下，發布新版的OTA檔案。
過程描述
收到OTA通知位於level1的ROOT開始向服務器進行自定義的下載，完成後自我重啟。(到這裡沒問題)
因為ROOT重啟引發mesh解散重組，但原本b1的沒有保持root的位置。
其他node取得ROOT，同樣以ROOT的角色開始向服務器進行自定義的下載，完成後自我重啟。

問題一
在於最一開始的b1的沒有保持root的位置。
或者說每個ROOT重啟後沒有保持到root的位置。
使得原本設計的使用樹狀結構傳播的更新無法實現，每個子節點輪流上去當一回ROOT向服務器要OTA檔案。
我不確定我是否遺漏了些甚麼，我只知道原本可以正常進行mesh架構的OTA改用671702a後動作異常。

問題二
ota_url在進到http_resumeable_download_task之前沒有亂碼
進到http_resumeable_download_task之後最前面的http://跟結尾都變成亂碼
ota_url 是全域變數，原本使用static私有因為有這個問題所以改成全域，但問題依然存在。

### 環境
IDF v5.0.2
blink 範例專案
idf_component.yml
dependencies:
idf: ">=5.0"
mesh_lite:
version: ""
espressif/esp_modem:
version: 0.
espressif/esp_modem_usb_dte:
version: 1.*
public: true
rules:
- if: "target in [esp32s2, esp32s3]"
espressif/led_strip: "^2.0.0"

### 調用的代碼
```
static void http_resumeable_download_task(void *args)
{
    // printf("http_resumeable_download_task ota_url: %s\n", ota_url);
    // esp_http_client_config_t config = {
    //     .url = ota_url,
    //     .event_handler = http_event_handler,
    // };
    // client = esp_http_client_init(&config);
    // printf("client:%p\n", client);
    // char range_header[50] = {0};
    // snprintf(range_header, sizeof(range_header), "bytes=%d-", LAST_DOWNLOADED_OFFSET);
    char *range_header;
    asprintf(&range_header, "bytes=%d-", LAST_DOWNLOADED_OFFSET);
    // printf(range_header);
    esp_err_t err = esp_http_client_set_header(client, "Range", range_header);
    if (err == ESP_OK)
    {
        printf("開始下載....\n");
        InOTAProcess = true;
    }

    esp_err_t http_err = esp_http_client_perform(client);
    if (http_err == ESP_OK)
    {
        if (LAST_DOWNLOADED_OFFSET == iOtaFileSize)
        {
            ota_end(true);
            printf("HTTP下載成功，重啟設備\n");
            // nvs_writeUint32(NVS_NS_SETTINGS, "StartCount", 0);
            //  重啟設備
            Reboot();
        }
    }
    else
    {
        printf("HTTP下載失敗，錯誤碼：%d\n", http_err);
    }

    esp_http_client_cleanup(client);
    // int i = 0;
    // while (1)
    // {
    //     printf("http_resumeable_download_task %d\n", i);
    //     i++;
    //     vTaskDelay(1000);
    // }
    vTaskDelete(NULL);
}
bool reOtaTaskStarted;
/// @brief 繼續OTA下載任務
void reOtaTaskStart()
{
#ifdef CONFIG_ESP_MESH_LITE_OTA_ENABLE
    /* Check if the OTA process is allowed. If there are higher-level nodes also requesting the firmware
     * from the external URL, the OTA process of this node will be stopped, and esp_mesh_lite_wait_ota_allow()
     * will return ESP_FAIL. It will restart the OTA process after its parent or higher-level nodes finish
     * requesting the firmware.
     */
    if (esp_mesh_lite_wait_ota_allow() != ESP_OK)
    {
        printf("!esp_mesh_lite_wait_ota_allow");
        return;
    }
    mesh_lite_ota_main();

#endif
    if (iOtaFileSize > 0 && !reOtaTaskStarted)
    {
        reOtaTaskStarted = true;

        // ESP_LOGI(TAG, "reOtaTaskStart(), url=%s, iOtaFileSize=%d\n", ota_url, iOtaFileSize);
        /* 不知為何 ota_url 到了 http_resumeable_download_task 裡面會出現亂碼 */
        esp_http_client_config_t config = {
            .url = ota_url,
            .event_handler = http_event_handler,
        };
        client = esp_http_client_init(&config);
        // printf("client:%p\n", client);
        xTaskCreate(&http_resumeable_download_task, "http_resumeable_download_task", 2048, NULL, 5, NULL); // 8192
    }
}

esp_err_t ota_task_start()
{
    // printf("ota_task_start ota_handle: %lu\n", ota_handle);
    if (ota_handle == 0)
    {
        // printf("url: %s, size: %d", url, size);
        // asprintf(&ota_url, "%s", url);
        // iOtaFileSize = size;
        LAST_DOWNLOADED_OFFSET = 0;
    }
    if (ota_handle == 0 || isDISCONNECTED)
        reOtaTaskStart();

    return ESP_OK;
}

esp_err_t ota_process(esp_mqtt_event_handle_t *event)
{
    // 拆解 topic
    char topic_buffer[(*event)->topic_len + 1];
    memcpy(topic_buffer, (*event)->topic, (*event)->topic_len);
    topic_buffer[(*event)->topic_len] = '\0';

    char *last_slash = strrchr(topic_buffer, '/');
    if (last_slash != NULL)
    {
        char *last_token = last_slash + 1;
        // printf("last_token: %s\n", last_token);
        //   int last_token_int = atoi(last_token);
        if (strcmp(last_token, "fileInfo") == 0)
        {
            char *data = calloc(1, (*event)->data_len + 1);
            strncpy(data, (*event)->data, (*event)->data_len);

            cJSON *root = cJSON_Parse(data);
            if (root == NULL)
            {
                printf("Error parsing JSON: %s\n", cJSON_GetErrorPtr());
                free(data);
                return ESP_FAIL;
            }

            // cJSON *hash = cJSON_GetObjectItem(root, "hash");
            cJSON *fw_version = cJSON_GetObjectItem(root, "version");
            cJSON *file_size = cJSON_GetObjectItem(root, "fileSize");
            if (fw_version != NULL && file_size != NULL)
            {
                // iOtaFileSize = 0; // 清零所要OTA的目標大小
                // printf("getCurrent_elf_sha256: %s; hash: %s\n", getCurrent_elf_sha256(), cJSON_GetStringValue(hash));
                // 如果OTA的HASH與目前的相同則不動作
                // char *sha256 = getCurrent_elf_sha256();
                // if (strcmp(sha256, cJSON_GetStringValue(hash))) // 不同為1
                // {
                iOtaFileSize = cJSON_GetNumberValue(file_size);
                char *version = cJSON_GetStringValue(fw_version);
                const char *currVersion = getCurrentVersion();
                // printf("currVersion: %s, version: %s, filesize: %d\n", currVersion, version, filesize);
                // printf("%d", strncmp(currVersion, version, strlen(version)));
                if (strncmp(currVersion, version, strlen(version)) == 0) // 慣例是使用 0 來表示相等或成功
                {
                    // ESP_LOGW(TAG, "Current running version is same as the new. We will not continue the update.");
                }
                else
                {
                    InOTAProcess = true;

                    cJSON *url = cJSON_GetObjectItem(root, "url");
                    if (url != NULL)
                        ota_url = cJSON_GetStringValue(url);

#ifdef CONFIG_ESP_MESH_LITE_OTA_ENABLE
                    if (esp_mesh_lite_get_level() > 1)
                    {
                        // 通知子節點暫停OTA進程
                        esp_mesh_lite_ota_notify_child_node_pause();
                        ESP_LOGI(TAG, "OTA Image version for Project: %s. Expected: %s", getCurrentVersion(), version);
                        int i = 0;
                        while (i < 6)
                        {
                            OTAWriting();
                            vTaskDelay(500);
                            i++;
                        }
                        esp_mesh_lite_ota_start(iOtaFileSize, version, esp_mesh_lite_ota_from_extern_url);
                    }
                    else
                    {
                        // esp_mesh_lite_set_allowed_level(1);
                        // nvs_writeUint32(NVS_NS_SETTINGS, "IsRoot", 1);
                        // 通知子節點暫停OTA進程
                        esp_mesh_lite_ota_notify_child_node_pause();
#endif
                        ESP_LOGI(TAG, "ota_task_start version: %s, url:%s", version, ota_url);
                        ota_task_start();

#ifdef CONFIG_ESP_MESH_LITE_OTA_ENABLE
                    }
#endif
                }
                // free(sha256);
            }
            else
            {
                printf("Missing JSON field(s)\n");
            }

            // printf("cJSON_Delete\n");
            cJSON_Delete(root);
            free(data);
        }
    }
    return ESP_OK;
}
```

### 附件
三塊版子的過程記錄

[b1.txt](https://github.com/espressif/esp-mesh-lite/files/12745310/b1.txt)
[b2.txt](https://github.com/espressif/esp-mesh-lite/files/12745312/b2.txt)
[b3.txt](https://github.com/espressif/esp-mesh-lite/files/12745309/b3.txt)

tswen, 2023-10-07:
[artifacts.zip](https://github.com/espressif/esp-mesh-lite/files/12837560/artifacts.zip)
请您使用该库文件替换原有的库问题，这个修复将会在后面同步到 github 上去。

8888alex, 2023-10-08:
> [artifacts.zip](https://github.com/espressif/esp-mesh-lite/files/12837560/artifacts.zip) 请您使用该库文件替换原有的库文件，这个修复将会在后面同步到 github 上去。

改用這個版本後ROOT是保持住了，但是子節點更新成104.15%然後完成不了，重啟後依舊是104.15%得到相同的情況。
[image]

### 過程說明
我先將這兩片版子手動更新成這個版本編譯的(v59)，然後再把版號變成v60，在發一版OTA，ROOT更新後秒重啟，然後子節點開始更新但進度跑超過100停在104.15%然後就卡住了，手動RESET後他就重新跑下載依然是跑到104.15%
下面是子節點的OTA LOG 請核實
### 子節點 LOG
```
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> set IDF_PATH=C:/Users/alex/esp/esp-idf-v5.0.2/
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM17 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2\build\aIoTSys.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM17 instead...
--- idf_monitor on \\.\COM17 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (21) boot: compile time 14:37:59
I (21) boot: chip revision: v0.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (62) boot:  1 ota_0            OTA app          00 10 00010000 00100000
I (70) boot:  2 ota_1            OTA app          00 11 00110000 00100000
I (77) boot:  3 nvs              WiFi data        01 02 00210000 001f0000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=30114h (196884) map
I (137) esp_image: segment 1: paddr=0004013c vaddr=3ffccc30 size=038fch ( 14588) load
I (140) esp_image: segment 2: paddr=00043a40 vaddr=40024000 size=0c5d8h ( 50648) load
I (155) esp_image: segment 3: paddr=00050020 vaddr=40080020 size=a3b38h (670520) map
I (289) esp_image: segment 4: paddr=000f3b60 vaddr=400305d8 size=0c650h ( 50768) load
I (314) boot: Loaded app from partition at offset 0x10000
I (314) boot: Disabling RNG early entropy source...
I (326) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (326) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (332) esp_psram: Found 2MB PSRAM device
I (336) esp_psram: Speed: 40MHz
I (340) cpu_start: Pro cpu up.
I (754) esp_psram: SPI SRAM memory test OK
I (768) cpu_start: Pro cpu start user code
I (768) cpu_start: cpu freq: 160000000 Hz
I (768) cpu_start: Application information:
I (771) cpu_start: Project name:     aIoTSys
I (776) cpu_start: App version:      v0.1.0.59
I (781) cpu_start: Compile time:     Oct  8 2023 21:21:43
I (788) cpu_start: ELF file SHA256:  347deeff53931c88...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 0dedbb06af6faa1ae8185dfcc1504b63ac3061cc6420fd855f522c4d2a0bd4b6
I (794) cpu_start: ESP-IDF:          v5.0.2-dirty
I (799) cpu_start: Min chip rev:     v0.0
I (804) cpu_start: Max chip rev:     v1.99 
I (808) cpu_start: Chip rev:         v0.0
I (813) heap_init: Initializing. RAM available for dynamic allocation:
I (820) heap_init: At 3FFD7AB0 len 00024550 (145 KiB): DRAM
I (827) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (833) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (839) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (847) spi_flash: detected chip: generic
[... 178 lines cut ...]
I (18187) ESP_Mesh_Lite_OTA: Image bytes read: 308136
bytes_read: 308136, percentage: 31.33% 
I (19868) ESP_Mesh_Lite_OTA: Image bytes read: 410848
bytes_read: 410848, percentage: 41.77% 
I (21544) ESP_Mesh_Lite_OTA: Image bytes read: 512172
bytes_read: 512172, percentage: 52.08% 
I (23217) ESP_Mesh_Lite_OTA: Image bytes read: 614884
bytes_read: 614884, percentage: 62.52% 
I (24898) ESP_Mesh_Lite_OTA: Image bytes read: 717596
bytes_read: 717596, percentage: 72.96% 
I (26579) ESP_Mesh_Lite_OTA: Image bytes read: 820308
bytes_read: 820308, percentage: 83.41% 
I (28335) ESP_Mesh_Lite_OTA: Image bytes read: 921632
bytes_read: 921632, percentage: 93.71% 
I (29898) ESP_Mesh_Lite_OTA: Image bytes read: 1024344
bytes_read: 1024344, percentage: 104.15% 
I (30238) esp_image: segment 0: paddr=00110020 vaddr=3f000020 size=30114h (196884) map
I (30273) esp_image: segment 1: paddr=0014013c vaddr=3ffccc30 size=038fch ( 14588) 
I (30277) esp_image: segment 2: paddr=00143a40 vaddr=40024000 size=0c5d8h ( 50648) 
I (30288) esp_image: segment 3: paddr=00150020 vaddr=40080020 size=a3b38h (670520) map
I (30409) esp_image: segment 4: paddr=001f3b60 vaddr=400305d8 size=0c650h ( 50768) 
E (30418) ESP_Mesh_Lite_OTA: Error: esp_ota_write failed! err=0x104
E (30422) Mesh-Lite: LAN OTA Fail

```

tswen, 2023-10-09:
想请问下这部分是在做什么操作，bytes_read: 921632, percentage: 93.71%。

8888alex, 2023-10-09:
> 想请问下这部分是在做什么操作，bytes_read: 921632, percentage: 93.71%。

這個部份我只是想用MQTT來回報進度，取消這段後子傑點的OTA就正常了
但不知我要如何才能回報進度，下面是我回報進度的代碼。
這裡面應該只有攔截訊息被沒有干涉IMAGE的操作，但不知為何會影響OTA?

```
// 自定義日誌處理程序
int custom_log_handler(const char *format, va_list args)
{
    // 將日誌消息格式化到緩衝區
    char buffer[256];
    vsnprintf(buffer, sizeof(buffer), format, args);

    // 原樣輸出到日誌
    printf("%s", buffer);

    // 檢查是否包含 "ESP_Mesh_Lite_OTA"
    if (strstr(buffer, "ESP_Mesh_Lite_OTA") != NULL)
    {
        // 檢查是否包含 "Image bytes read:"
        char *image_bytes_read = strstr(buffer, "Image bytes read:");
        if (image_bytes_read != NULL)
        {
            // 提取 "Image bytes read:" 後面的數字
            int bytes_read = 0;
            sscanf(image_bytes_read, "Image bytes read: %d", &bytes_read);
            // printf("Image bytes read: %d, %d\n", bytes_read, filesize);
            OTAWriting();//閃爍LED
            float percentage = ((float)bytes_read / (float)iOtaFileSize) * 100.0;
            printf("bytes_read: %d, percentage: %.2f%% \n", bytes_read, percentage);
            char *msg;
            asprintf(&msg, "{\"ota\": %.2f}", percentage);
            publish_counter_msg("connected", msg, 0);
        }
    }

    return 0; // 返回值不重要，這裡設置為 0
}
```

8888alex, 2023-10-10:
> 任务看门狗造成的崩溃我不认为是 Mesh-Lite 造成的，您可以排查下您的应用层代码是否哪里有阻塞，是否跟 uart 相关，因为我看到跟 log 相关。
這兩個LOG的差異僅在於使用的不同版本的lib，不過我還是再爬爬看是否有那裡阻塞。

> 上面那段代码可能影响了内部 OTA 的流程，我们后面会增加一个时间进度的事件，LAN OTA 时会每 1/100 上报一次进度。
那我先取消這段等待您的OTA進度回報。

tswen, 2023-10-12:
您可以暂时使用 https://github.com/tswen/esp-mesh-lite/tree/feature/Report_LAN_ota_progress_events 这个分支，内部将通过 https://github.com/tswen/esp-mesh-lite/blob/feature/Report_LAN_ota_progress_events/components/mesh_lite/src/esp_mesh_lite.c#L200 ESP_MESH_LITE_EVENT_OTA_PROGRESS 该事件上报进度。

8888alex, 2023-10-12:
請問，
當我將uart0從ESP_CONSOLE_UART_DEFAULT變更成ESP_CONSOLE_USB_CDC後，
是否也將會導致無法進行OTA?

### 變更設定的內容如下
```
[SDK Configuration Editor]
{"version": 2, "set": { "ESP_CONSOLE_USB_CDC": true }}
[SDK Configuration Editor]
Set ESP_CONSOLE_USB_CDC
{"version": 2, "values": {"APPTRACE_DEST_UART0": false, "ESP_CONSOLE_UART_DEFAULT": false, "ESP_CONSOLE_USB_CDC": true, "ESP_CONSOLE_UART_NUM": -1, "ESP_CONSOLE_USB_CDC_RX_BUF_SIZE": 64, "ESP_CONSOLE_USB_CDC_SUPPORT_ETS_PRINTF": false}, "ranges": {"ESP_CONSOLE_USB_CDC_RX_BUF_SIZE": [4, 16384]}, "visible": {"APPTRACE_DEST_UART0": true, "ESP_CONSOLE_USB_CDC_RX_BUF_SIZE": true, "ESP_CONSOLE_USB_CDC_SUPPORT_ETS_PRINTF": true, "ESP32S2_KEEP_USB_ALIVE": false}}
[SDK Configuration Editor]
{"version":2,"save":"c:\\Users\\alex\\myProjects\\ESP32\\Coding\\aIoTSys_v5.0.2\\sdkconfig"}
[SDK Configuration Editor]
{"version": 2, "values": {}, "ranges": {}, "visible": {}}
```
### LOG如下
```
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> set IDF_PATH=C:/Users/alex/esp/esp-idf-v5.0.2/
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM18 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2\build\aIoTSys.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM18 instead...
--- idf_monitor on \\.\COM18 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (195916) wifi:state: run -> init (0)
I (195917) wifi:pm stop, total sleep time: lu us / lu us

I (195918) wifi:new:<5,0>, old:<5,1>, ap:<5,1>, sta:<5,1>, prof:1
E (195925) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 52
E (195926) mqtt_client: Poll read error: 119, aborting connection
I (195961) wifi:flush txq
I (195961) wifi:stop sw txq     
I (195963) wifi:lmac stop hw txq
I (34) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (35) boot: compile time 10:39:59
I (35) boot: chip revision: v0.0
I (36) boot.esp32s2: SPI Speed      : 80MHz
I (36) boot.esp32s2: SPI Mode       : DIO
I (36) boot.esp32s2: SPI Flash Size : 4MB
I (37) boot: Enabling RNG early entropy source...
I (38) boot: Partition Table:
I (38) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (40) boot:  1 ota_0            OTA app          00 10 00010000 00100000
I (41) boot:  2 ota_1            OTA app          00 11 00110000 00100000
I (42) boot:  3 nvs              WiFi data        01 02 00210000 001f0000
I (44) boot: End of partition table
I (44) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=30084h (196740) map
I (83) esp_image: segment 1: paddr=000400ac vaddr=3ffccbb0 size=038bch ( 14524) load
I (88) esp_image: segment 2: paddr=00043970 vaddr=40024000 size=0c6a8h ( 50856) load
I (102) esp_image: segment 3: paddr=00050020 vaddr=40080020 size=a2b04h (666372) map
I (236) esp_image: segment 4: paddr=000f2b2c vaddr=400306a8 size=0c504h ( 50436) load
I (261) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
I (693) spi_flash: detected chip: generic
I (693) spi_flash: flash io: dio
I (697) cpu_start: Starting scheduler on PRO CPU.
I (698) IoTSys: app_main: total: 2301663 used: 53564 free: 2248099 ... (remain 97.67%)
I (700) system_api: Base MAC address is not set
I (700) system_api: read default base MAC address from EFUSE
W (701) IoTSys: client_id: IoT_969660
I (701) NVS: nvs_init()
Count: UsedEntries = (183), FreeEntries = (62313), AllEntries = (62496), namespace_count = (6)
W (979) gpio_main: 初始化管腳及按鈕功能
I (979) gpio_main: 初始設定板載LED
I (980) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (991) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (992) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (993) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (994) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
[... 124 lines cut ...]
I (2600) wifi:connected with MikroTik-8A6FB4, aid = 3, channel 5, 40U, bssid = b8:69:f4:8a:6f:b4
I (2601) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (2621) wifi:pm start, type: 0

I (2679) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (3624) mesh_mqtt: mqtt_app_start()
mqtt://192.168.88.249:1883
W (3635) mqtt_client: Publish: Losing qos0 data when client not connected
W (3637) Mesh-Lite-Message: send_message_to_parent
I (3640) esp_netif_handlers: sta ip: 192.168.88.57, mask: 255.255.255.0, gw: 192.168.88.1
I (3642) bridge_wifi: Connected with IP Address:192.168.88.57
I (3643) vendor_ie: RTC store: ssid:MikroTik-8A6FB4; bssid:00:00:00:00:00:00 crc:3093236531
I (3641) mesh_mqtt: Other event id:7
I (3727) counter: publish_counter_value() topic:counter, msg:{"表一":45,"表二":50,"表三":185,"表四":31}
esp_mesh_lite_get_level: 1, getCurrentVersion: v0.1.0.50
I (3793) mesh_mqtt: 完成訂閱, msg_id=64751
I (3796) mesh_mqtt: 完成訂閱, msg_id=64063
I (3801) mesh_mqtt: 完成訂閱, msg_id=44809
I (3920) mesh_mqtt: topic:aIoTSys/IoT_969660/OTA/fileInfo, data:{"hash":"aIoTSys","version":"v0.1.0.51","fileSize":979040,"url":"http://192.168.88.249/uploads/aIoTSys.bin"}

收到OTA通知...StartCount:3
I (3925) mesh_lite_ota: ota_task_start version: v0.1.0.51, url:http://192.168.88.249/uploads/aIoTSys.bin
...
到這裡就再也不會動了，也不會進行OTA
```

tswen, 2023-10-12:
你这个是根节点无法正常从外部 url 获取固件吧，这个感觉跟 mesh-lite 和 LAN OTA 没有关系，你可以使用暂时去掉 mesh 功能进行排除下是不是哪里出现了问题。

tswen, 2023-10-17:
您好，该问题已再临时分支上进行了修复，请更新最新的组件：https://github.com/tswen/esp-mesh-lite/tree/feature/Report_LAN_ota_progress_events

8888alex, 2023-10-18:
請問造成char亂碼的可能原因，或者說我能怎麼排除這下面這個問題
ota_url: http://192.168.88.249/uploads/aIoTSys.bin
ota_url: o�?D{�?92.168.88.249/uploads/aIoTSys.bi\m�?
E (7087) HTTP_CLIENT: Error parse url o�?D{�?92.168.88.249/uploads/aIoTSys.bixn�?

### 說明
上面的打印來自於
```
 printf("ota_url: %s\n", ota_url);
 esp_mesh_lite_ota_start(iOtaFileSize, version, ota_task_start);
```
在更換組件後我進行了刪除build資料夾，然後進行編譯，
運行後得到HTTP_CLIENT: Error parse url 錯誤
經打印得到發現ota_url變成亂碼
進一步跟蹤發現
經由 esp_mesh_lite_ota_start 調用 ota_task_start 後所打印出來的URL會夾帶亂碼。

ota_task_start(void)的內容
```
esp_err_t ota_task_start(void)
{
    printf("ota_url: %s\n", ota_url);
    if (ota_handle == 0)
    {
        // printf("url: %s, size: %d", url, size);
        // asprintf(&ota_url, "%s", url);
        // iOtaFileSize = size;
        LAST_DOWNLOADED_OFFSET = 0;
    }
    if (ota_handle == 0 || isDISCONNECTED)
        reOtaTaskStart();

    return ESP_OK;
}
```
原以為是這版組件的問題，但我刪除mesh_lite組件資料夾重新複製，還是一樣的情況。
所以我恢復備份的組件但竟然連原本備份的組件都還是會夾帶亂碼。
請問我還能做刪除些甚麼?
其實，類似的情況之前也發生過一次，所以想說問問是否我哪裡做錯了。

### 前一次發生
在這之前也發生過一次URL夾帶亂碼。
原本我的config.url 是在http_resumeable_download_task才進行指定
在一次更新mesh_lite(master)組件後就突然出現夾帶亂碼，查無原因無法排除後來只好把
config移到 http_resumeable_download_task 外面。 代碼如下
```
 /* 不知為何 ota_url 到了 http_resumeable_download_task 裡面會出現亂碼 */
  esp_http_client_config_t config = {
      .url = ota_url,
      .event_handler = http_event_handler,
  };
  client = esp_http_client_init(&config);
  // printf("client:%p\n", client);
  xTaskCreate(&http_resumeable_download_task, "http_resumeable_download_task", 2048, NULL, 5, NULL); // 8192
```

### LOG
```
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> set IDF_PATH=C:/Users/alex/esp/esp-idf-v5.0.2/
PS C:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2> C:\Users\alex\esp\idf_tools\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\alex\esp\esp-idf-v5.0.2\tools\idf_monitor.py -p COM9 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 c:\Users\alex\myProjects\ESP32\Coding\aIoTSys_v5.0.2\build\aIoTSys.elf

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (34) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (35) boot: compile time 17:38:40
I (35) boot: chip revision: v0.0
I (36) boot.esp32s2: SPI Speed      : 80MHz      
I (36) boot.esp32s2: SPI Mode       : DIO        
[... 330 lines cut ...]
I (5209) ip select: IP Address:192.168.5.1
I (5209) ip select: GW Address:192.168.5.1
I (5209) ip select: NM Address:255.255.255.0
I (5212) bridge_common: ip reallocate new:192.168.5.1
W (5213) bridge_wifi: SoftAP IP network segment has changed, deauth all station
I (5214) vendor_ie: RTC store: ssid:ESP_Bridge_969661; bssid:7c:df:a1:96:96:61 crc:17420695
I (5208) mesh_mqtt: Other event id:7
W (5221) Mesh-Lite-Message: handle_message_process
W (5221) Mesh-Lite-Message: Received ROUTER_SSID: MikroTik-8A6FB4
W (5222) Mesh-Lite-Message: Received ROUTER_PASSWD: 13812829726
W (5223) Mesh-Lite-Message: Received mqttHost: 192.168.88.249
I (5263) counter: publish_counter_value() topic:counter, msg:{"表一":143,"表二":90,"表三":786,"表四":2,"表號":0}
esp_mesh_lite_get_level: 2, getCurrentVersion: v0.1.0.52
I (5307) mesh_mqtt: 完成訂閱, msg_id=9353
I (5310) mesh_mqtt: 完成訂閱, msg_id=3616
I (5314) mesh_mqtt: 完成訂閱, msg_id=62727
收到OTA通知...StartCount:50
I (5506) mesh_lite_ota: OTA Image version for Project: v0.1.0.52. Expected: v0.1.0.53
ota_url: http://192.168.88.249/uploads/aIoTSys.bin
ota_url: D{�?D{�?92.168.88.249/uploads/aIoTSys.bi$n�?f
E (8699) HTTP_CLIENT: Error parse url 
E (8699) HTTP_CLIENT: Failed to set URL
ota_url: 

```

tswen, 2023-10-19:
ota_url 是全局变量吗，可以增加 const 限定符试试。

8888alex, 2023-10-19:
> const

實測有效，感謝。

closed 2023-10-19 by 8888alex