espressif/esp-mesh-lite #34: esp_bridge_create_all_netif 功耗

8888alex, 2023-10-02:
### 說明
經測量在調用 esp_bridge_create_all_netif 之前，整體代碼功耗約28.7-29.1mA左右
調用 esp_bridge_create_all_netif 之後，整體代碼功耗約**87.9-88.2mA**左右(沒看錯,是暴增了約60mA)
完整運行所有代碼功耗約 88.6-89.2mA左右
請問是否有可以改善的方法?

### 環境
硬件: ESP32-S2-DevKitM-1 v1.0
IDF v5.0.2
idf_component.yml
dependencies:
    idf: ">=5.0"
    mesh_lite:
        version: "*"
    espressif/esp_modem:
        version: 0.*
    espressif/esp_modem_usb_dte:
        version: 1.*
        public: true
        rules:
        - if: "target in [esp32s2, esp32s3]"
    usb_device:
        path: components/usb/usb_device
        git: https://github.com/espressif/esp-iot-bridge.git
        rules:
        - if: "target in [esp32s2, esp32s3]"
    espressif/led_strip: "^2.0.0"

tswen, 2023-10-07:
对于 esp-mesh-lite 来说，esp_bridge_create_all_netif 里面主要调用了 esp_bridge_create_softap_netif 和 esp_bridge_create_station_netif，因为同时开启了 Wi-Fi Station 和 SoftAP，所以功耗会增高。有两个方式可以稍微降低些功耗
- 增大 beacon interval
- 将 SoftAP 的 ssid 隐藏

8888alex, 2023-10-10:
> * 增大 beacon interval
> * 将 SoftAP 的 ssid 隐藏

請問具體該怎設置，有範例代碼嗎? 謝謝

tswen, 2023-10-10:
```c
    wifi_init();

    wifi_config_t wifi_cfg;
    esp_wifi_get_config(WIFI_IF_AP, &wifi_cfg);
    wifi_cfg.ap.ssid_hidden = true;
    wifi_cfg.ap.beacon_interval = 200;
    esp_wifi_set_config(WIFI_IF_AP, &wifi_cfg);

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);
```

tswen, 2023-10-10:
[artifacts.zip](https://github.com/espressif/esp-mesh-lite/files/12855108/artifacts.zip)
如果想要使能  wifi_cfg.ap.ssid_hidden，需要使用这个 lib，目前 master 分支上，softap 隐藏 ssid 仍有一点问题，这个 lib 进行了修复，但是合入还需要一定时间。

8888alex, 2023-10-11:
改用這個版本後功號並沒有太大個改善，依舊維持在88mA上下。

xcguang, 2023-10-11:
启动 softap 后，就类似路由器一样，功耗会比较高

8888alex, 2023-10-11:
> 启动 softap 后，就类似路由器一样，功耗会比较高

所以是有計畫要改善?還是要直接關閉這個issue？

xcguang, 2023-10-12:
softap 的功耗改善不明显，这个只能关闭这个 issue 了

closed 2023-10-12 by 8888alex