espressif/esp-mesh-lite #35: Is 16 the absolute maximum for MESH_LITE_MAXIMUM_LEVEL_ALLOWED?

Teesmo, 2023-10-03:
I'm trying to create a mesh network with a depth level of about 20 - 25 hops. I went through the KConfig file in esp-mesh-lite/components/mesh_lite, and it defines
```
config MESH_LITE_MAXIMUM_LEVEL_ALLOWED
int "The maximum level allowed" 
default 10
range 1 16
```
Is there a particular reason why the maximum is 16?
Can I alter the range to something like

```
range 1 20

```

tswen, 2023-10-07:
Currently, it can only support up to 16 levels, but we have plans to support higher levels later.

Teesmo, 2025-03-17:
Hi @tswen, I just wanted to check progress on this

tswen, 2025-03-18:
Currently, we believe that a maximum level of 16 is sufficient to meet the needs of most scenarios. If the level is increased further, higher-level nodes may experience significant communication delays, potentially making the network less efficient than deploying multiple independent mesh groups. As such, there are no immediate plans to support higher levels.

Teesmo, 2025-03-18:
Alright, thank you.

closed 2025-03-18 by Teesmo