espressif/esp-mesh-lite #36: set-target esp32c3 No url found for submodule path 'components/json_generator/upstream'

xiao-orien, 2023-10-17:
When i set target esp32c3 always report  No url found for submodule path 'components/json_generator/upstream'.

 Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 111, in download_project_dependencies
      solution = solver.solve()
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 34, in solve
      self.solve_manifest(manifest)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 53, in solve_manifest
      self.solve_component(requirement)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 56, in solve_component
      cmp_with_versions = requirement.source.versions(
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 136, in versions
      commit_id = self._checkout_git_source(version, temp_dir, selected_paths=[self.component_path])
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 52, in _checkout_git_source
      return self._client.prepare_ref(
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 139, in prepare_ref
      self.run(
    File "D:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git --work-tree=.  -C
  C:\Users\ADMINI~1\AppData\Local\Temp\tmp2eshb6hw --git-dir
  C:\Users\Administrator\AppData\Local\Espressif\ComponentManager\Cache\b_git_5cbb1e8d
  submodule update --init --recursive' failed with exit code 128

  fatal: No url found for submodule path 'components/json_generator/upstream'
  in .gitmodules

tswen, 2023-10-18:
It's possible that there's an issue with your ESP-IDF environment. Have you updated the submodules using the "submodule update --init --recursive" command in ESP-IDF? Additionally, I recommend deleting the "build," "managed_components" directory, and the "dependencies.lock" file, and then recompiling.

xiao-orien, 2023-10-18:
I has already submodule updated and deleted the "build," "managed_components", "dependencies.lock" file.

xiao-orien, 2023-10-18:
D:\Espressif\frameworks>cd esp-idf-v4.4.4

D:\Espressif\frameworks\esp-idf-v4.4.4>git submodule update --init --recursive

D:\Espressif\frameworks\esp-idf-v4.4.4>cd ..

D:\Espressif\frameworks>cd mesh-lite\c3\esp-mesh-lite\examples\rainmaker\led_light

tswen, 2023-10-19:
You can try to upgrade the idf-component-manager.

`pip install --upgrade idf-component-manager`

xiao-orien, 2023-10-23:
It's ok know. Thanks!

closed 2023-10-23 by xiao-orien