espressif/esp-mesh-lite #37: 是否支持ESP32C3？

bpz-git, 2023-11-06:
esp-idf 版本是5.1.1，运行esp-mesh-lite\examples\mesh_local_control>例程

tswen, 2023-11-10:
暂时两个解决办法：1 不行再试 2
1. 更新下 pip，再将 idf-component-manager 更新到 1.4.1 试试
pip install --upgrade pip
pip install idf-component-manager==1.4.1
2. https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/idf_component.yml#L9 组件里面有个文件，那就尝试将这个改成 espressif/iot_bridge: "0.7.1"

bpz-git, 2023-11-10:
我按照1，2项都做过后还是提示失败，具体信息如下

神州六号
***@***.***

 
暂时两个解决办法：1 不行再试 2

tswen, 2023-11-10:
不好意思，Github 上无法看到您提到的“具体信息”。

bpz-git, 2023-11-10:
E:\esp_test\esp-mesh-lite\examples\mesh_local_control>idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'E:\esp_test\esp-mesh-lite\examples\mesh_local_control\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory E:\esp_test\esp-mesh-lite\examples\mesh_local_control\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=E:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 E:\esp_test\esp-mesh-lite\examples\mesh_local_control"...
-- Found Git: E:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: E:/Espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/Espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/Espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
Dependencies lock doesn't exist, solving dependencies.
Using component placed at E:\esp_test\esp-mesh-lite\components\mesh_lite for dependency mesh_lite(*), specified in E:/esp_test/esp-mesh-lite/examples/mesh_local_control/main\idf_component.yml
CMake Error at E:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/build.cmake:540 (message):
  HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for
  other targets: "esp32s3", "esp32s2".  Is your current target "esp32c3" set
  correctly?

  WARNING: Cannot find versions of "espressif/usb_host_cdc_acm" satisfying
  "2.*" for the current target "esp32c3".

  WARNING: Cannot find versions of "espressif/usb_host_cdc_acm" satisfying
  "2.*" for the current target "esp32c3".

  WARNING: Cannot find versions of "espressif/usb_host_cdc_acm" satisfying
  "1.*" for the current target "esp32c3".

  ERROR: 'git config --get remote.origin.url' failed with exit code 1

Call Stack (most recent call first):
  E:/Espressif/frameworks/esp-idf-v5.1.1/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:7 (project)

.............-- Configuring incomplete, errors occurred!
See also "E:/esp_test/esp-mesh-lite/examples/mesh_local_control/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the E:\esp_test\esp-mesh-lite\examples\mesh_local_control\build\log\idf_py_stderr_output_11524 and E:\esp_test\esp-mesh-lite\examples\mesh_local_control\build\log\idf_py_stdout_output_11524

E:\esp_test\esp-mesh-lite\examples\mesh_local_control>

tswen, 2023-11-14:
您可以更新新的 mesh-lite 仓库再试试

bpz-git, 2023-11-14:
已经更新到最新的版本了，问题提示也是一样的，no_router和led_light例程是可以把器件设置为esp32c3，并且可以正常编译，但是mesh_local_control例程无法设置为esp32c3，
所以我觉得应该还是mesh_local_control这个工程某些设置参数有问题
$ git log
commit cb8a837d52f444729398bdfa33542a2d3883d25c (HEAD -&gt; master)
Merge: 671702a 1b3131e
Author: Xu Chun Guang ***@***.***&gt;
Date:   Thu Oct 19 08:57:54 2023 +0800

    Merge branch 'feature/add_max_node_number_config' into 'master'

    feat: add max_node_number config

    Closes AEG-895

    See merge request ae_group/esp-mesh-lite!52

commit 1b3131e8a668af6fd3259d64c0f94c8317f0c04f
Author: Tian Sen Wen ***@***.***&gt;
Date:   Wed Sep 27 16:07:06 2023 +0800

    feat: Updated mesh-lite core version to a60bf56

    - feat: Deauth station when reached the maximum level
    - feat: Add max_node_number config
    - fix: Failed to connect to the original parent node when the device software restarts
:...skipping...
commit cb8a837d52f444729398bdfa33542a2d3883d25c (HEAD -&gt; master)
Merge: 671702a 1b3131e
Author: Xu Chun Guang ***@***.***&gt;
Date:   Thu Oct 19 08:57:54 2023 +0800

    Merge branch 'feature/add_max_node_number_config' into 'master'

    feat: add max_node_number config

    Closes AEG-895

    See merge request ae_group/esp-mesh-lite!52

神州六号
***@***.***

tswen, 2023-11-14:
请执行以下操作再试试看：
1. pip list | grep idf
2. pip install idf-component-manager==1.4.1
3. rm -rf build dependencies.lock managed_components
4. idf.py set-targrt esp32c3

bpz-git, 2023-11-15:
刚才按照这个操作流程确实无法解决，怀疑是工程文件本身问题，能发一个mesh_local_control例程编译通过esp-mesh-lite压缩包给我吗？我的邮件是 ***@***.***, 十分感谢！
e:\esp_test\esp-mesh-lite\examples\mesh_local_control&gt;pip list
Package               Version
--------------------- ---------
bitarray              2.8.1
bitstring             4.1.1
CacheControl          0.13.1
certifi               2023.7.22
cffi                  1.15.1
charset-normalizer    3.2.0
click                 8.0.4
colorama              0.4.6
construct             2.10.68
contextlib2           21.6.0
cryptography          39.0.2
ecdsa                 0.18.0
esp-coredump          1.6.0
esp-idf-kconfig       1.2.0
esp-idf-monitor       1.1.3
esp-idf-size          0.4.2
esptool               4.7.dev1
filelock              3.12.2
freertos-gdb          1.0.2
idf-component-manager 1.4.1
idna                  3.4
kconfiglib            14.1.0
msgpack               1.0.5
packaging             23.1
pip                   23.3.1
pycparser             2.21
pyelftools            0.29
pygdbmi               0.9.0.2
pyparsing             3.0.9
pyserial              3.5
PyYAML                6.0.1
reedsolo              1.7.0
requests              2.31.0
requests-file         1.5.1
requests-toolbelt     1.0.0
schema                0.7.5
setuptools            68.1.2
six                   1.16.0
tqdm                  4.66.1
urllib3               1.26.16
windows-curses        2.3.1

神州六号
***@***.***

 
请执行以下操作再试试看：
  
pip list | grep idf

tswen, 2023-11-15:
我有看了下之前你提供的 git log 日志，你目前的 esp-mesh-lite 并不是最新的，最新的 commit id 是 d444ce0

bpz-git, 2023-11-15:
mesh_local_control例程，使用了3个ESP32C3测试模块，放在了A B C个不同的地方, wifi 路由器是在A和B的中间位置，C在B的后面位置，上电顺序是A-&gt;B-&gt;C，然后在电脑端运行了一个TCP server，
从接收到的数据判断，A和B位置的无线模块都工作在第一层，C模块（B模块下面节点）工作在第二层，看文档介绍网络会自动选择一个模块作为主路由工作在第一层，其它模块以树状结构连接，感觉现在有2个个模块都工作在主路由，
这样的状态相对mesh-lite组网来讲是正常的吗？ A模块的IP是192.168.1.51， B模块的IP是192.168.1.52

[2023/11/15 15:51:47 037]  TCP服务器[192.168.1.51:65229]接收86字节: 
[2023/11/15 15:51:47 046]  {"src_addr": "54:32:04:44:aa:c0","data": "Hello TCP Server!","level": 1,"count": 62}

[2023/11/15 15:51:47 829]  TCP服务器[192.168.1.52:59778]接收86字节: 
[2023/11/15 15:51:47 837]  {"src_addr": "54:32:04:41:b8:bc","data": "Hello TCP Server!","level": 1,"count": 62}

[2023/11/15 15:51:49 793]  TCP服务器[192.168.1.52:49770]接收86字节: 
[2023/11/15 15:51:49 804]  {"src_addr": "54:32:04:44:b0:e8","data": "Hello TCP Server!","level": 2,"count": 63}

神州六号
***@***.***

bpz-git, 2023-11-17:
好的，感谢支持

closed 2023-11-17 by bpz-git