espressif/esp-mesh-lite #38: Unable to compile due to issue with modem ?

ThatBigPrint, 2023-11-06:
not compiling due to a lot of errors with esp modem ?

ThatBigPrint, 2023-11-11:
So I fixed the issue it was due to my mac m2 I had to call a build flag.. 

Aside from that I have had no end of issues trying to get this to work with an arduino library im trying to use a firebase library because idf dosn't have event listeners... I managed to get the old mdf working somewhat, but this one isn't letting the arduino lib continue I worked around this in the mdf by provisioning wifi on the arduino side then passing back to mdf... But on this we obviously need the out of Arduino mesh lite wifi init stuff

Do you know why this would be the case? Also the whole mesh thing is rather confusing you have three a parent mesh "systems" the old mdf, this esp mesh lite, and esp mesh that is in the main idf under mesh.... The one in the idf has an example with internal ip is that the same as this?

tswen, 2023-11-13:
Hello, **esp-mesh-lite** and **esp-wifi-mesh** are two completely different network implementation solutions, so they cannot be used together. The internal IP in IDF and MDF actually borrows interfaces from esp-wifi-mesh for implementation at the underlying level.

tswen, 2023-11-14:
If mdf can meet your development needs, you can continue to use mdf for development, but mdf will not invest more effort in maintenance. It is recommended to use the ESP-Mesh-Lite solution later.

closed 2023-12-21 by ThatBigPrint