espressif/esp-mesh-lite #39: ERROR using ESP32 target

temistoclesCWClamper, 2023-11-13:
Hi!

Example mesh_local_control build perfectly with target ESP32S3, but not with target ESP32 

I'm using Eclipse + IDF eclipse plugin 
ESP-IDF version = v4.4

Solving dependencies requirements
CMake Error at C:/dev/espressif/esp-idf-4.4.4/tools/cmake/build.cmake:452 (message):
................................................-- Configuring incomplete, errors occurred!
See also "C:/dev/espressif/esp32/mesh_lite/build/CMakeFiles/CMakeOutput.log".
  HINT: Component "espressif/usb_host_cdc_acm" has suitable versions for
  other targets: "esp32s3", "esp32s2".  Is your current target "esp32" set
  correctly?

  ERROR: Cannot find versions of "espressif/usb_host_cdc_acm" with version
  satisfying "2.*" for the current target "esp32"

Call Stack (most recent call first):
  C:/dev/espressif/esp-idf-4.4.4/tools/cmake/project.cmake:391 (idf_build_process)
  CMakeLists.txt:6 (project)

tswen, 2023-11-14:
Two temporary solutions: 1 or 2.

1. Update pip and then update idf-component-manager to version 1.4.1.
```
pip install --upgrade pip
pip install idf-component-manager==1.4.1
```

2. Delete the build and dependencies.lock directories in managed_components.Remove the following content from main/idf_component.yml:
```
# The "esp_modem_usb_dte" component is temporarily disabled in iot_bridge yml file,
# as idf-component-manager doesn't support uploading components with "rules" entries to the registry.
# So temporarily place the "esp_modem_usb_dte" "esp_modem" component under examples.
#
espressif/esp_modem:
  version: 0.*
espressif/esp_modem_usb_dte:
  version: 1.*
  public: true
  rules:
    - if: "target in [esp32s2, esp32s3]"
    - if: "idf_version >= 4.4"
```

temistoclesCWClamper, 2023-11-14:
It worked!

For everyone using Eclipse, remember to execute these commands in terminal inside eclipse.

C:\dev\espressif\esp32>pip install --upgrade pip
Requirement already satisfied: pip in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (23.0)
Collecting pip
  Using cached pip-23.3.1-py3-none-any.whl (2.1 MB)
ERROR: To modify pip, please run the following command:
C:\Users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\Scripts\python.exe -m pip install --upgrade pip

[notice] A new release of pip is available: 23.0 -> 23.3.1
[notice] To update, run: python.exe -m pip install --upgrade pip

C:\dev\espressif\esp32>python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (23.0)
Collecting pip
  Using cached pip-23.3.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0
    Uninstalling pip-23.0:
      Successfully uninstalled pip-23.0
Successfully installed pip-23.3.1

C:\dev\espressif\esp32>pip install idf-component-manager==1.4.1
Collecting idf-component-manager==1.4.1
  Using cached idf_component_manager-1.4.1-py2.py3-none-any.whl.metadata (11 kB)
Requirement already satisfied: packaging in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (23.0)
Requirement already satisfied: requests<3 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (2.28.2)
Requirement already satisfied: urllib3<2 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (1.26.14)
Requirement already satisfied: requests-file in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (1.5.1)
Requirement already satisfied: requests-toolbelt in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (0.10.1)
Requirement already satisfied: schema in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (0.7.5)
Requirement already satisfied: six in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (1.16.0)
Requirement already satisfied: tqdm<5 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (4.64.1)
Requirement already satisfied: pyparsing in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (2.3.1)
Requirement already satisfied: colorama in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (0.4.6)
Requirement already satisfied: pyyaml>5.2 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (6.0)
Requirement already satisfied: cachecontrol>0.12.6 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (0.12.11)
Requirement already satisfied: contextlib2>0.6.0 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (21.6.0)
Requirement already satisfied: click in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from idf-component-manager==1.4.1) (8.1.3)
Requirement already satisfied: msgpack>=0.5.2 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from cachecontrol>0.12.6->cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (1.0.4)
Requirement already satisfied: lockfile>=0.9 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from cachecontrol[filecache]>0.12.6; python_version >= "3.6"->idf-component-manager==1.4.1) (0.12.2)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (3.0.1)
Requirement already satisfied: idna<4,>=2.5 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from requests<3->idf-component-manager==1.4.1) (2022.12.7)
Requirement already satisfied: importlib-metadata in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from click->idf-component-manager==1.4.1) (6.0.0)
Requirement already satisfied: zipp>=0.5 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from importlib-metadata->click->idf-component-manager==1.4.1) (3.13.0)
Requirement already satisfied: typing-extensions>=3.6.4 in c:\users\temistocleswanderley\.espressif\python_env\idf4.4_py3.7_env\lib\site-packages (from importlib-metadata->click->idf-component-manager==1.4.1) (4.5.0)
Using cached idf_component_manager-1.4.1-py2.py3-none-any.whl (140 kB)
Installing collected packages: idf-component-manager
  Attempting uninstall: idf-component-manager
    Found existing installation: idf-component-manager 1.2.2
    Uninstalling idf-component-manager-1.2.2:
      Successfully uninstalled idf-component-manager-1.2.2
Successfully installed idf-component-manager-1.4.1

closed 2023-11-14 by temistoclesCWClamper