espressif/esp-mesh-lite #40: ESP32C3 use ESP BLE Prov APP tool provisioning return Failed to connect with BLE device check if POP is correct.

xiao-orien, 2023-11-20:
When I build project as ESP32C3 use ESP BLE Prov APP tool provisioning return Failed to connect with BLE device check if POP is correct.
But build project as ESP32S3 provisioning is ok.

I (36) cpu_start: Pro cpu up.
I (44) cpu_start: Pro cpu start user code
I (44) cpu_start: cpu freq: 160000000
I (44) cpu_start: Application information:
I (47) cpu_start: Project name:     gateway
I (52) cpu_start: App version:      v1.0.0-beta_8b13503-dirty
I (58) cpu_start: Compile time:     Nov 20 2023 08:58:54
I (64) cpu_start: ELF file SHA256:  df3e732299eca452...
I (70) cpu_start: ESP-IDF:          v4.4.4-dirty
I (75) heap_init: Initializing. RAM available for dynamic allocation:
I (82) heap_init: At 3FC99520 len 000431F0 (268 KiB): DRAM
I (89) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (95) heap_init: At 5000008C len 00001F64 (7 KiB): RTCRAM
I (102) spi_flash: detected chip: generic
I (106) spi_flash: flash io: dio
I (110) sleep: Configure to isolate all GPIO pins in sleep state
I (117) sleep: Enable automatic switching of GPIO sleep configuration
I (124) coexist: coexist rom version 9387209
I (129) cpu_start: Starting scheduler.
W (133) app_main: app_main 45 app_main const heap 274412
W (133) app_main: free heap 274236, minimum 274236
I (143) bridge_common: esp-iot-bridge version: 0.7.1
I (143) pp: pp rom version: 9387209
I (153) net80211: net80211 rom version: 9387209
I (163) wifi:wifi driver task: 3fca25cc, prio:23, stack:6144, core=0
I (163) system_api: Base MAC address is not set
I (163) system_api: read default base MAC address from EFUSE
I (173) wifi:wifi firmware version: 6567a16
I (173) wifi:wifi certification version: v7.0
I (183) wifi:config NVS flash: enabled
I (183) wifi:config nano formating: enabled
I (183) wifi:Init data frame dynamic rx buffer num: 32
I (193) wifi:Init management frame dynamic rx buffer num: 32
I (193) wifi:Init management short buffer num: 32
I (203) wifi:Init dynamic tx buffer num: 32
I (203) wifi:Init static tx FG buffer num: 2
I (213) wifi:Init static rx buffer size: 1600
I (213) wifi:Init static rx buffer num: 10
I (213) wifi:Init dynamic rx buffer num: 32
I (223) wifi_init: rx ba win: 6
I (223) wifi_init: tcpip mbox: 32
I (233) wifi_init: udp mbox: 6
I (233) wifi_init: tcp mbox: 6
I (233) wifi_init: tcp tx win: 5744
I (243) wifi_init: tcp rx win: 5744
I (243) wifi_init: tcp mss: 624
I (253) wifi_init: WiFi IRAM OP enabled
I (253) wifi_init: WiFi RX IRAM OP enabled
I (263) phy_init: phy_version 950,11a46e9,Oct 21 2022,08:56:12
I (303) wifi:mode : null
I (303) ip select: IP Address:192.168.4.1
I (303) ip select: GW Address:192.168.4.1
I (303) ip select: NM Address:255.255.255.0
I (303) bridge_wifi: IP Address:192.168.4.1
Add netif ap with 8b13503(commit id)
I (313) bridge_common: netif list add success
I (323) wifi:mode : softAP (10:91:a8:02:6b:f5)
I (323) wifi:Total power save buffer number: 16
I (333) wifi:Init max length of beacon: 752/752
I (333) wifi:Init max length of beacon: 752/752
Add netif sta with 8b13503(commit id)
I (343) bridge_common: netif list add success
I (343) wifi:mode : sta (10:91:a8:02:6b:f4) + softAP (10:91:a8:02:6b:f5)
I (353) wifi:enable tsf
I (353) wifi:Disabled PMF config for STA
W (353) vendor_ie: Error Get[4354]
W (363) vendor_ie: Error Get[4354]
I (953) wifi:Total power save buffer number: 16
I (953) bridge_wifi: softap ssid: ESP_Bridge_026bf5 password: 12345678
I (953) Mesh-Lite: esp-mesh-lite component version: 0.7.0
Mesh-Lite commit id: 34377a9
I (963) vendor_ie: Mesh ID: 77
W (963) vendor_ie: Error Get[4354]
W (973) vendor_ie: Error Get[4354]
I (973) ESP_Mesh_Lite_Comm: msg action add success
I (983) ESP_Mesh_Lite_Comm: Bind Socket 54, port 6364
I (983) ESP_Mesh_Lite_Comm: Bind Socket 55, port 6363
I (993) ESP_Mesh_Lite_Comm: Bind Socket 56, port 6366
I (1003) ESP_Mesh_Lite_Comm: Bind Socket 57, port 6365
I (1003) Mesh-Lite: Mesh-Lite connecting
I (1013) ESP_Mesh_Lite_Comm: msg action add success
I (1023) esp_rmaker_work_queue: Work Queue created.
I (1023) app_gateway: RainMaker Initialised.
I (1023) esp_rmaker_node: Node ID ----- 7eecb8e-0dcc-6cba-a27f-173e9bfc71e5
I (1033) esp_rmaker_node: Node attribute LiteMeshDevice created
I (1043) esp_rmaker_system_service: System service enabled.
I (1043) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (1053) esp_rmaker_core: Starting RainMaker Work Queue task
I (1063) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (1063) wifi:Haven't to connect to a suitable AP now!
E (1073) ESP_Mesh_Lite_Comm: Error Get!
W (1073) vendor_ie: Error Get[4354]
W (1083) vendor_ie: Error Get[4354]
W (1083) app_bridge: Error Get[4354]
I (1093) app_bridge: mesh service enabled
I (1093) ESPNOW: espnow [version: 1.0] init
I (1093) app_espnow: Start espnow task
I (1103) app_wifi: Starting provisioning
I (1103) app_wifi: reserve0 10 [Mesh-Lite Enable]:1 [root]:0 [child]:1
I (1113) BTDM_INIT: BT controller compile version [02bfb7f]
I (1123) BTDM_INIT: Bluetooth MAC: 10:91:a8:02:6b:f6

I (1123) protocomm_nimble: BLE Host Task Started
I (1133) NimBLE: GAP procedure initiated: stop advertising.

I (1143) NimBLE: GAP procedure initiated: advertise; 
I (1143) NimBLE: disc_mode=2
I (1143) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (1153) NimBLE: 

I (1163) wifi_prov_mgr: Provisioning started with service name : HOMII_0b026 
I (1163) app_wifi: Provisioning started
I (1173) app_wifi: Provisioning Started. Name : HOMII_0b026, POP : a8026bf4
I (1183) app_wifi: Provisioning will auto stop after 30 minute(s).
I (13303) NimBLE: GAP procedure initiated: advertise; 
I (13303) NimBLE: disc_mode=2
I (13303) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (13313) NimBLE: 

I (15073) NimBLE: GAP procedure initiated: advertise; 
I (15073) NimBLE: disc_mode=2
I (15073) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (15083) NimBLE:

I (17023) NimBLE: GAP procedure initiated: advertise;
I (17023) NimBLE: disc_mode=2
I (17023) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (17033) NimBLE:

I (18763) NimBLE: GAP procedure initiated: advertise;
I (18763) NimBLE: disc_mode=2
I (18763) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (18773) NimBLE:

I (25063) NimBLE: GAP procedure initiated: advertise;
I (25063) NimBLE: disc_mode=2
I (25063) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (25073) NimBLE:

I (27673) NimBLE: GAP procedure initiated: advertise;
I (27673) NimBLE: disc_mode=2
I (27673) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (27683) NimBLE:

I (31783) NimBLE: GAP procedure initiated: advertise;
I (31783) NimBLE: disc_mode=2
I (31783) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (31793) NimBLE:

xiao-orien, 2023-11-20:
[image]
[image]

tswen, 2023-11-20:
The examples in rainmaker require the use of **Nova Home** APP for network configuration. If you do not have access to the Rainmaker cloud and want to use the BLE Prov APP for network configuration, please manually port the [wifi_prov_mgr ](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr)example of idf to the mesh_local_control demo or your own project.

xiao-orien, 2023-11-20:
But build project as ESP32S3 use the same provisioning is ok.

xiao-orien, 2023-11-20:
I use Nova Home APP test again.
I build project as ESP32C3.  I use Nova Home APP for  for network configuration is also failed.
But build project as ESP32S3 is ok.

I (2174) protocomm_nimble: BLE Host Task Started
I (2174) NimBLE: GAP procedure initiated: stop advertising.

I (2184) NimBLE: GAP procedure initiated: advertise; 
I (2184) NimBLE: disc_mode=2
I (2194) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (2204) NimBLE: 

I (2204) wifi_prov_mgr: Provisioning started with service name : MESH_0b026b 
I (2214) esp_rmaker_local: Event 1
I (2214) app_wifi: Provisioning started
I (2214) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"MESH_0b026b","pop":"a8026bf4","transport":"ble"}
I (2234) app_wifi: Provisioning Started. Name : MESH_0b026b, POP : a8026bf4
I (2244) app_wifi: Provisioning will auto stop after 30 minute(s).
I (33484) NimBLE: GAP procedure initiated: advertise; 
I (33484) NimBLE: disc_mode=2
I (33484) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (33494) NimBLE: 

I (35254) NimBLE: GAP procedure initiated: advertise; 
I (35254) NimBLE: disc_mode=2
I (35254) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (35264) NimBLE: 

I (37024) NimBLE: GAP procedure initiated: advertise; 
I (37024) NimBLE: disc_mode=2
I (37024) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (37034) NimBLE: 

I (38794) NimBLE: GAP procedure initiated: advertise; 
I (38794) NimBLE: disc_mode=2
I (38794) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (38804) NimBLE: 

I (41404) NimBLE: GAP procedure initiated: advertise; 
I (41404) NimBLE: disc_mode=2
I (41404) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (41414) NimBLE: 

I (43984) NimBLE: GAP procedure initiated: advertise; 
I (43984) NimBLE: disc_mode=2
I (43984) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (43994) NimBLE: 

I (46894) NimBLE: GAP procedure initiated: advertise; 
I (46894) NimBLE: disc_mode=2
I (46894) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (46904) NimBLE: 

I (52474) NimBLE: GAP procedure initiated: advertise; 
I (52474) NimBLE: disc_mode=2
I (52474) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (52484) NimBLE: 

I (69604) NimBLE: GAP procedure initiated: advertise; 
I (69604) NimBLE: disc_mode=2
I (69604) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (69614) NimBLE: 

I (71374) NimBLE: GAP procedure initiated: advertise; 
I (71374) NimBLE: disc_mode=2
I (71374) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (71384) NimBLE: 

I (73144) NimBLE: GAP procedure initiated: advertise; 
I (73144) NimBLE: disc_mode=2
I (73144) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (73154) NimBLE: 

I (74914) NimBLE: GAP procedure initiated: advertise; 
I (74914) NimBLE: disc_mode=2
I (74914) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (74924) NimBLE: 

I (77554) NimBLE: GAP procedure initiated: advertise; 
I (77554) NimBLE: disc_mode=2
I (77554) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (77564) NimBLE: 

I (79834) NimBLE: GAP procedure initiated: advertise; 
I (79834) NimBLE: disc_mode=2
I (79834) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (79844) NimBLE: 

I (82444) NimBLE: GAP procedure initiated: advertise; 
I (82444) NimBLE: disc_mode=2
I (82444) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (82454) NimBLE: 

I (85054) NimBLE: GAP procedure initiated: advertise; 
I (85054) NimBLE: disc_mode=2
I (85054) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (85064) NimBLE:

I (87664) NimBLE: GAP procedure initiated: advertise;
I (87664) NimBLE: disc_mode=2
I (87664) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (87674) NimBLE:

I (89614) NimBLE: GAP procedure initiated: advertise;
I (89614) NimBLE: disc_mode=2
I (89614) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (89624) NimBLE:

I (91354) NimBLE: GAP procedure initiated: advertise;
I (91354) NimBLE: disc_mode=2
I (91354) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (91364) NimBLE:

I (93094) NimBLE: GAP procedure initiated: advertise; 
I (93094) NimBLE: disc_mode=2
I (93094) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (93104) NimBLE: 

I (94834) NimBLE: GAP procedure initiated: advertise; 
I (94834) NimBLE: disc_mode=2
I (94844) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (94844) NimBLE: 

I (97774) NimBLE: GAP procedure initiated: advertise; 
I (97774) NimBLE: disc_mode=2
I (97784) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (97784) NimBLE:

xiao-orien, 2023-11-21:
[image]

tswen, 2023-11-21:
Please check again whether the esp-mesh-lite repository you are using is the latest version. I haven't encountered the mentioned issue when using rainmaker/led_light from the latest esp-mesh-lite. Additionally, the version of Nova Home on my iPhone is 2.1.0.

If you decide not to use rainmaker and Nova Home in the end, I suggest testing with the **[wifi_prov_mgr](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr)** example under idf. Once the test is successful, you can then port it to your own project or mesh_local_control.
```
I (42) cpu_start: Unicore app
I (42) cpu_start: Pro cpu up.
I (50) cpu_start: Pro cpu start user code
I (50) cpu_start: cpu freq: 160000000 Hz
I (50) cpu_start: Application information:
I (53) cpu_start: Project name:     led_light
I (58) cpu_start: App version:      v1.0.0-beta_d444ce0
I (64) cpu_start: Compile time:     Nov 21 2023 10:40:50
I (70) cpu_start: ELF file SHA256:  9fbb8deefe5f2916...
I (76) cpu_start: ESP-IDF:          v5.0.4-dirty
I (81) cpu_start: Min chip rev:     v0.3
I (86) cpu_start: Max chip rev:     v0.99 
I (91) cpu_start: Chip rev:         v0.3
I (95) heap_init: Initializing. RAM available for dynamic allocation:
I (102) heap_init: At 3FC9E700 len 0003E010 (248 KiB): DRAM
I (109) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (115) heap_init: At 5000007C len 00001F5C (7 KiB): RTCRAM
I (122) spi_flash: detected chip: generic
I (126) spi_flash: flash io: dio
I (131) sleep: Configure to isolate all GPIO pins in sleep state
I (137) sleep: Enable automatic switching of GPIO sleep configuration
I (144) coexist: coexist rom version 9387209
I (149) app_start: Starting scheduler on CPU0
I (154) main_task: Started on CPU0
I (154) main_task: Calling app_main()
W (154) app_main: app_main 43 app_main const heap 254040
W (164) app_main: free heap 254040, minimum 254040
W (174) vfs: esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 3
I (174) bridge_common: esp-iot-bridge version: 0.9.0
I (184) pp: pp rom version: 9387209
I (184) net80211: net80211 rom version: 9387209
I (204) wifi:wifi driver task: 3fca73bc, prio:23, stack:6144, core=0
I (204) system_api: Base MAC address is not set
I (204) system_api: read default base MAC address from EFUSE
I (214) wifi:wifi firmware version: e2d0ad8
I (214) wifi:wifi certification version: v7.0
I (214) wifi:config NVS flash: enabled
I (224) wifi:config nano formating: enabled
I (224) wifi:Init data frame dynamic rx buffer num: 32
I (234) wifi:Init management frame dynamic rx buffer num: 32
I (234) wifi:Init management short buffer num: 32
I (244) wifi:Init dynamic tx buffer num: 32
I (244) wifi:Init static tx FG buffer num: 2
I (244) wifi:Init static rx buffer size: 1600
I (254) wifi:Init static rx buffer num: 10
I (254) wifi:Init dynamic rx buffer num: 32
I (264) wifi_init: rx ba win: 6
I (264) wifi_init: tcpip mbox: 32
I (264) wifi_init: udp mbox: 6
I (274) wifi_init: tcp mbox: 6
I (274) wifi_init: tcp tx win: 5744
I (284) wifi_init: tcp rx win: 5744
I (284) wifi_init: tcp mss: 624
I (284) wifi_init: WiFi IRAM OP enabled
I (294) wifi_init: WiFi RX IRAM OP enabled
I (294) phy_init: phy_version 1110,9c20f0a,Jul 27 2023,10:42:54
W (304) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (584) wifi:mode : null
I (584) ip select: IP Address:192.168.4.1
I (584) ip select: GW Address:192.168.4.1
I (584) ip select: NM Address:255.255.255.0
I (594) bridge_wifi: IP Address:192.168.4.1
Add netif ap with d444ce0(commit id)
I (604) bridge_common: netif list add success
I (604) wifi:mode : softAP (7c:df:a1:da:fd:05)
I (614) wifi:Total power save buffer number: 16
[... 175 lines cut ...]
I (36454) NimBLE: GAP procedure initiated: stop advertising.

I (36454) NimBLE: GAP procedure initiated: stop advertising.

I (36454) wifi_prov_mgr: Provisioning stopped
tlsf_add_pool: Memory size must be between 20 and 264 bytes.
I (36464) wifi_prov_scheme_ble: BLE memory released
I (36474) esp_rmaker_local: Event 6
I (36474) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 0
I (36494) esp_https_server: Starting server
I (36494) esp_https_server: Server listening on port 8080
I (36494) esp_rmaker_local: esp_local_ctrl service started with name : 7CDFA1DAFD04
W (36504) app_rainmaker: Unhandled RainMaker Event: 6
I (40204) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (43514) esp_rmaker_param: Received params: {"Time":{"TZ":"Asia\/Shanghai","TZ-POSIX":"CST-8"}}
I (43514) esp_rmaker_time_service: Received value = Asia\/Shanghai for Time - TZ
E (43524) esp_rmaker_param: Remote update to param Time - TZ failed
I (43524) esp_rmaker_time_service: Received value = CST-8 for Time - TZ-POSIX
W (43534) app_rainmaker: Unhandled RainMaker Common Event: 6
I (43544) esp_rmaker_time: The current time is: Tue Nov 21 10:43:45 2023 +0800[CST], DST: No.
I (43554) esp_rmaker_param: Reporting params: {"Time":{"TZ-POSIX":"CST-8"},"mesh":{"level":1,"self_ip":["192.168.5.1","192.168.4.100"]}}
I (61954) esp_rmaker_time: SNTP Synchronised.
I (61954) esp_rmaker_time: The current time is: Tue Nov 21 10:44:03 2023 +0800[CST], DST: No.
I (83224) esp_rmaker_param: Received params: {"fast_network_service":{"output":true,"provisionNow":true}}
```

xiao-orien, 2023-11-21:
I use the newest master branch for test and my NovaHome is 2.1.0.
For my project need use BLE Prov is better.