espressif/esp-mesh-lite #41: Communication FAQ;s related question

swaroopsuthar, 2023-11-20:
- How to send data to **specific Child** MAC ID through **Root** node without broadcast ?
- How to send **esp_mesh_lite_send_msg_to_root** with retransmissions ? 
- How to send raw data in **esp_mesh_lite_try_sending_msg** ?
- Could you please describe how to work **esp_mesh_lite_ota_start** because i want without internet Root node received data from UART and send data specific child device category?
- What is purpose of **esp_mesh_lite_ota_start**?

tswen, 2023-11-21:
CN
1. 使用传输层通信（TCP or UDP）无法做到这一点，不过您可以使用 [espnow](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html) 通信来实现点对点的发送
2. esp_mesh_lite_try_sending_msg 的第三个参数是在接收到第二个参数的 expect_msg 之前的最大重传次数
3. esp_mesh_lite_try_sending_msg 第四个参数是要发送的 json 字符串
4. LAN OTA 相关内容可以参考 [LAN OTA]( https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide.md#esp-mesh-lite-lan-ota) 这个文档，后续如果还有 LAN OTA 相关问题，可以新打开一个 issue
5. 如果用不到 “禁止别的设备加入 mesh 网路 `esp_mesh_lite_allow_others_to_join(false)`” 这个功能，就不需要关心 argot 的值，如果需要禁止别的设备加入 mesh 网络，那么同一个 mesh 网络内部的设备除了 mesh-id 要保持相同，argot 也要保持相同。

EN
1. Achieving this through transport layer communication (TCP or UDP) is difficult. However, you can use [espnow](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html) communication to achieve point-to-point transmission.
2. The third parameter of esp_mesh_lite_try_sending_msg is the maximum retry count before receiving the expected message (the second parameter).
3. The fourth parameter of esp_mesh_lite_try_sending_msg is the JSON string to be sent.
4. For LAN OTA-related content, you can refer to [LAN OTA](https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide.md#esp-mesh-lite-lan-ota). If you have further LAN OTA-related questions, feel free to open a new issue.
5. If you don't need the "Prohibit other devices from joining the mesh network esp_mesh_lite_allow_others_to_join(false)" feature, you don't need to concern yourself with the value of argot. However, if you want to prevent other devices from joining the mesh network, devices within the same mesh network need to have the same argot value in addition to maintaining the same mesh-id.

tswen, 2023-11-21:
[esp_mesh_lite.zip](https://github.com/espressif/esp-mesh-lite/files/13421457/esp_mesh_lite.zip)
This code file attempts to write the usage of all currently open internal communication APIs.

swaroopsuthar, 2023-11-21:
Thanks for sharing the knowledge.

Now we will manage internal table of children for each parent and send data by TCP protocol. Can you tell how much data is passed over TCP or UDP protocol?

Because we want Root node to receive OTA from UART and send OTA binary data through TCP protocol.

tswen, 2023-11-21:
I don’t think you need to do this process again, because the mesh lite component implements a LAN OTA solution internally, which can update the OTA firmware layer by layer through the parent node. For specific LAN OTA introduction, interface description and migration guide, please refer to [User_Guide](https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide.md#esp-mesh-lite-lan-ota).

swaroopsuthar, 2023-11-21:
Because Root node has no internet and Device type as Gateway. and other child node Device type as Switches with diff. category.

tswen, 2023-11-23:
Yes, if the root node and child nodes are of different categories, it is indeed necessary to establish a link between the root node and the second-level node to transmit the firmware transmitted from uart.

closed 2023-11-23 by swaroopsuthar

iwonako, 2024-12-11:
> 2. esp_mesh_lite_try_sending_msg 的第三个参数是在接收到第二个参数的 expect_msg 之前的最大重传次数
> 3. esp_mesh_lite_try_sending_msg 第四个参数是要发送的 json 字符串

这个方法的第一第二个参数是什么意思？怎么决定的？另外问问这个库有 api 文档吗