espressif/esp-mesh-lite #43: The second layer node can,t connect to the root node with version 0.6.0 and higher

WonderfulMoment, 2023-11-24:
Our platform is esp32s3. We use the 0.4.0 befor. it could built a mesh net work. Now we want to use newer version to replace the old one. when I replaced the lib with 0.5.0, it could worked. But wen i replace the lib with 0.6.0, there was a issue. when I started the first esp32s3, It could connected to the router and the layer is 1. Then I started the second esp32s3. At first the layer is 1 and it connected to the router. After a while, the layer of one of the esp32s3 became 2 and disconnected from the router. Few minutes later,  the layter 2 esp32s3 reconnect to the router but not the root node and it could not connet to the internet. I tried it with version 0.9.0 and met the  same issue.

tswen, 2023-11-25:
Hello, updating mesh_lite requires updating the entire mesh_lite component instead of just updating lib. In addition, later versions added the esp_mesh_lite_start API, which requires esp_mesh_lite_start to be executed after esp_mesh_lite_init is executed at the application layer.

WonderfulMoment, 2023-11-25:
> Hello, updating mesh_lite requires updating the entire mesh_lite component instead of just updating lib. In addition, later versions added the esp_mesh_lite_start API, which requires esp_mesh_lite_start to be executed after esp_mesh_lite_init is executed at the application layer.

写了一堆中式英文才发觉可以用中文交流😂，我替换了库也替换了component的代码的，0.9.0版本我在调用esp_mesh_lite_init之后调用了esp_mesh_lite_start接口的，但是结果还是一样，0.6.0版本还没有增加这个接口，所以没有调用，现象就是感觉它有在尝试组网但是二层的节点连不上根节点，这个问题在0.4.0和0.5.0上没有出现，所以不知道0.6.0以上的版本有什么特殊的地方？

tswen, 2023-11-27:
idf.py monitor 就会输出日志。

WonderfulMoment, 2023-11-27:
> idf.py monitor 就会输出日志。

我们专门做了串口监控日志输出的，但是组网的时候没看到mesh lite有什么特殊的打印，我的意思是，需要设置哪个模块到某个特殊的日志等级再抓给你么？还是直接给你启动时候的日志信息？wifi相关模块的日志等级被我们设置到了告警级别，这些需要重新设置一下么？

WonderfulMoment, 2023-11-27:
> 就正常 info 等级的日志就好了。

设哪几个模块到info？

WonderfulMoment, 2023-11-27:
> > 就正常 info 等级的日志就好了。
> 
> 设哪几个模块到info？

能给下具体日志模块名称么？我对咋们的系统还不是特别熟

WonderfulMoment, 2023-11-28:
> idf.py monitor 就会输出日志。
我把以下模块设置成了info抓了两块卡的log
esp_log_level_set("wifi", ESP_LOG_INFO);
esp_log_level_set("wifi_init", ESP_LOG_INFO);
esp_log_level_set("net80211", ESP_LOG_INFO);
esp_log_level_set("ESP_Mesh_Lite_Comm", ESP_LOG_INFO);
1970/01/01 00:00:31.446 [ls bridge/I] System information, channel: 1, layer: 1, self mac: 48:27:e2:2d:7f:f1, parent bssid: 48:27:e2:2d:8f:b1, parent rssi: -27, free heap: 206168 write_mesh_id<156>  read_mesh_id<156>这个是我们在定时器里加的打印，打印连接的基本信息，这个每10s打印一次
root.txt是第一块上电的卡
layer2_node.txt是第二块上电的卡

麻烦抽空看一下呢，谢谢
[layer2_node.txt](https://github.com/espressif/esp-mesh-lite/files/13482630/layer2_node.txt)
[root.txt](https://github.com/espressif/esp-mesh-lite/files/13482632/root.txt)

tswen, 2023-11-28:
你可以先跑一下 mesh_local_control 示例，然后对比下缺失了些什么。

WonderfulMoment, 2023-11-28:
1，我直接到mesh_local_control示例里编译，没编译过，提示如下错误信息，我对咱们的工具链不是很熟，不知道这个yml文件是干嘛用的。
2，我自己的工程里我也是把库里面idf_component.yml里的dependencies:下内容删掉才编译通过的，我是新接手的项目，之前的同事的这个文件里的这部分内容也是删了的，但是能正常工作，测试过0.4.0，0.5.0两个版本都是能正常工作的
3，我自己也看了下mesh_local_control的代码，除了ssid和password是我们自己定义的以外，没看到应用代码上由啥区别欸，调用的接口和顺序基本都是一致的，也没漏什么接口，所以这个问题有点不知道怎么往下排查
4，另外想请问以下，log信息里看到什么可疑的信息么？

以下是直接编译示例代码的编译错误信息，我们用的编译环境idf版本是5.0,我自己的工程没有动yml文件里的dependencies:情况下也是同样的错误
CMake Error at /home/lhq/software/esp-idf/tools/cmake/build.cmake:519 (message):
  ERROR: Solver failed processing dependency "espressif/cmake_utilities" from
  the manifest file
  "/home/lhq/prj_esp32/esp-mesh-lite-master-0.9.0/examples/mesh_local_control/main/idf_component.yml".

  HTTPSConnectionPool(host='components-file.espressif.com', port=443): Max
  retries exceeded with url: /components/espressif/cmake_utilities.json
  (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object
  at 0x7fc463ed7c70>: Failed to establish a new connection: [Errno -3]
  Temporary failure in name resolution'))

Call Stack (most recent call first):
  /home/lhq/software/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:7 (project)

WonderfulMoment, 2023-11-28:
刚看了一下，好像跟mac尾缀有关系，0.6.0删了后mac尾缀这个变量，请问一下，现在根节点的layer层级改成0了么？之前的根节点layer是1

tswen, 2023-11-29:
看样子像是网络问题导致包管理器拉去依赖的组件失败。

WonderfulMoment, 2023-11-29:
> 看样子像是网络问题导致包管理器拉去依赖的组件失败。
把dependencies:去掉规避这个编译问题没有问题吧？
另外，我把编译选项里的SoftAP SSID ends with the MAC关掉还是不行；
用组件里面的示例代码mesh_local_control放到我自己的工程里编译，发觉还是有问题，第二层节点里的打印的父节点的mac信息和根节点打印的自己的mac地址信息不一致；

tswen, 2023-12-04:
第二层节点里的打印的父节点的mac信息和根节点打印的自己的mac地址信息不一致 是因为节点打印自己的是 station mac，打印父节点是父节点的 softap mac。

WonderfulMoment, 2023-12-05:
但是网络是不通的，而且我们0.4.0和0.5.0我们测试看到的子节点上显示的父节点的mac信息和父节点自己显示自己的mac信息是一致的，0.9.0上改了获取的父节点的mac的显示逻辑么？
另外请教以下，我们节点作为sta连接父节点可以设置连接的信号wifi强度阈值么？比如，扫描到-80db以上的信号强度我才连接，低于-80db的信号强度的节点，我就忽略，不作为连接的候选父节点；

tswen, 2023-12-05:
[image]
一直都是这样。你应该是看错了。分别是父节点的 station mac 和 softap mac。

这个是 mesh-lite 内部的选择选择机制，暂时没有接口开放设置，后续可能会考虑。

WonderfulMoment, 2023-12-05:
今天用0.5.0版本，发现个比较奇怪的问题
设备要不就不连路由器，要不就是连了，打印的layer 为 0，大部分时间是连接不上（打印是10s打印一次）
环境是由两个esp32s3，作为mesh节点（初始化为ap+sta,并初始化mesh lite），另外一个esp32s3只初始化ap（当路由器用）
以下是两个初始化为ap+sta的两个设备的打印
[image]
[image]

WonderfulMoment, 2023-12-06:
> 今天用0.5.0版本，发现个比较奇怪的问题 设备要不就不连路由器，要不就是连了，打印的layer 为 0，大部分时间是连接不上（打印是10s打印一次） 环境是由两个esp32s3，作为mesh节点（初始化为ap+sta,并初始化mesh lite），另外一个esp32s3只初始化ap（当路由器用） 以下是两个初始化为ap+sta的两个设备的打印 [...]

环境还是和昨天的环境一样，今天的现象又不一样，
1，我重启了其中一个esp32s3 （sta+ap）卡1，起来之后，它去连接另外一个esp32s3（ap+sta）模式的 卡2，而没有连接ap
2，卡2此时也没有连接ap
3，此外显示的父节点的mac地址确实是父节点显示的自己的mac地址
这是卡1的log信息
[image]
这是卡2的log信息
[image]

现在有点懵，搞不清楚啥逻辑，感觉现象有点乱，没规律

tswen, 2023-12-11:
1. 打印 layer 0 是因为虽然连上了 wifi，但是信号强度太差了 -90db，一直获取不到 ip，所以并没有更新 level。
2. 对于 mesh 内的设备，应该要尽量不要去做硬重启，否则根据上电后连接逻辑，其发现了可连接的父节点（有可能是其原来的子节点），连接上去之后可能形成环形，不过新的版本中我们已经增加了对于这种环形连接的一个检测和破除逻辑，但是需要一定的事件恢复成正常的网络，所以还是希望尽量避免硬重启的出现。

WonderfulMoment, 2023-12-26:
谢谢你的回复，现在我把库更新到最新的版本，发现层级layer在根节点重启的时候还是会出现乱的情况
路由器mac：48:27:e2:2d:8f:b1
两个esp32s3 卡作为mesh节点测试，一张mac=48:27:e2:2d:7f:f1，另一张mac=48:27:e2:2e:a5:81
1，路由器正常工作
2，升级卡48:27:e2:2d:7f:f1并启动，结果：连上路由器，此时看到自身层级layer=1
3，升级卡48:27:e2:2e:a5:81并启动，连接上路由器，自身层级为2，但不去连接根节点
4，重启48:27:e2:2d:7f:f1，并同时重启48:27:e2:2e:a5:81
5，48:27:e2:2d:7f:f1连接上了路由器，但自身层级layer变成了9
6，48:27:e2:2e:a5:81连接上48:27:e2:2d:7f:f1，自身层级layer变成了10

后面又测试了几次，发现只要根节点重启断开，原来的子节点去连接路由器，不会改变自身的层级layer值，导致重复操作重启根节点，layer会一直往上加

[image]

[image]

tswen, 2023-12-28:
CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED 这个配置项最高层级目前默认应该是 5，不建议设置很高的层级。

WonderfulMoment, 2024-01-08:
> CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED 这个配置项最高层级目前默认应该是 5，不建议设置很高的层级。

我的理解，这里不管CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED 配置为多少，mesh中只有两个节点的情况下，不管怎么重启两个卡，应该都是一个是1一个是2吧？不应该出现更高的值才对哇？目前测试的情况是它这个值会变化，而且在根节点重启的情况下，后续的layer值是不确定的

tswen, 2024-01-08:
如果根节点硬件重启，其再次上电之后有可能连到其原来的子节点上去，从而造成一个环形连接，层级信息会受到影响，不断更新，某个设备达到最高层级之后会主动断开连接破除这种环形连接状态，当然这也只是其中一种破除环形连接的方式。这种后续会自恢复为正常拓扑。

WonderfulMoment, 2024-01-08:
> 如果根节点硬件重启，其再次上电之后有可能连到其原来的子节点上去，从而造成一个环形连接，层级信息会受到影响，不断更新，某个设备达到最高层级之后会主动断开连接破除这种环形连接状态，当然这也只是其中一种破除环形连接的方式。这种后续会自恢复为正常拓扑。

谢谢回复！
根节点重启之后，原来的二层节点主动去连接路由器了，成为了根节点，但是这个时候自身的layer值没有改为1，可能还保持原来的layer值（这个是正常逻辑么？还是bug？），感觉这个layer值不停的变跟这个有很大关系。

tswen, 2024-01-09:
你好，最新的 master 分支已经解决了层级显示错误的问题，请将 esp-mesh-lite 更新到最新版本并进行测试。

WonderfulMoment, 2024-01-10:
> 你好，最新的 master 分支已经解决了层级显示错误的问题，请将 esp-mesh-lite 更新到最新版本并进行测试。

我就是用的最新的库测试的哦，changlog里已经是0.10.0

tswen, 2024-01-10:
最新的版本是昨天下午更新的，使用最新的还会出现问题吗？如果还有问题，请再描述下你的复现步骤和所有日志文件。

WonderfulMoment, 2024-01-10:
> 最新的版本是昨天下午更新的，使用最新的还会出现问题吗？如果还有问题，请再描述下你的复现步骤和所有日志文件。
我用的12月26号下载的代码和库

WonderfulMoment, 2024-01-12:
> 最新的版本是昨天下午更新的，使用最新的还会出现问题吗？如果还有问题，请再描述下你的复现步骤和所有日志文件。

昨天用最新的库试了一下，layer变动的问题目前测试没用问题了，另外想请教几个问题：
1，esp_mesh_lite_set_rssi_threshold  这个接口是修改连接父节点的信号阈值的么？比如我想低于-80db的节点才成为候选父节点，参数该怎么给呢？
2，新版本支持ESP_MESH_LITE_ROUTER 和 ESP_MESH_LITE_MESH，想请教一下这两种模式的主要区别是啥，有相关的资料可以查看么？
谢谢!