espressif/esp-mesh-lite #44: Hidden network still not working

ThatBigPrint, 2023-11-26:
Hi guys i have attempted to get hidden network working i can get it hiding fine but once its hidden the nodes no longer connect i notced @tswen had a brach with it working but why is it not incorporated into the main? i have also tried the manual country declaration for wifi but it's still not connecting...

also is there a better way to determine if the node has access not only to ip but the internet? if a root drops out i want the nodes to temporarily pause http requests ?

tswen, 2023-11-27:
Yes, currently, the branch https://github.com/tswen/esp-mesh-lite/tree/feature/support_ssid_hidden supports hidden SSID. However, this branch is just a temporary fix and is not fully compatible with scenarios where users customize both the softap prefix and suffix while hiding the SSID. Therefore, it will take some time to merge this into the master branch.

You can confirm a successful internet connection by sending ICMP packets.

tswen, 2023-11-27:
You can set it like this
```c
    wifi_init();

    wifi_config_t wifi_cfg;
    esp_wifi_get_config(WIFI_IF_AP, &wifi_cfg);
    wifi_cfg.ap.ssid_hidden = true;
    wifi_cfg.ap.beacon_interval = 200;
    esp_wifi_set_config(WIFI_IF_AP, &wifi_cfg);

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);
```

ThatBigPrint, 2023-11-27:
@tswen when will this be released i think the utility of hiding the network is far more of a priority than having names surely spamming people's network lists in a real aplication is not desirable at all... ?   

Also when will there be an ota example ?

and on top of that for my use ie lights im using a  custom firebase database on each light... how do i synchronise time between them like the mdf did ? 

is the idea for ota for the root node to distribute or the nodes to download it themselves from the external source?

ruslanslobozhenuk, 2023-11-27:
Working!!!!
In my code in function esp_bridge_wifi_set(...) need add : 

 wifi_cfg.ap.authmode = strlen((char*)wifi_cfg.ap.password) < 8 ? WIFI_AUTH_OPEN : WIFI_AUTH_WPA2_PSK;
 //Add hiden ssid
 wifi_cfg.ap.ssid_hidden = true;
 wifi_cfg.ap.beacon_interval = 200;
 ...
 
 Thank you!!!

tswen, 2023-11-27:
In fact, there is no need to make changes in the iot_bridge component. As I said above, you can first get wifi softap config and then customize the configuration, and then esp_wifi_set_config.

Currently mesh-lite has implemented LAN OTA, please refer to the [documentation](https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide.md#esp-mesh-lite-lan-ota) for details.

ThatBigPrint, 2023-11-27:
are there other changes i would need to do to the main branch to make this work ?

tswen, 2023-11-28:
This is not necessary, it is already supported on the master branch.

ThatBigPrint, 2023-11-28:
@tswen I was referring to the hidden... With the changes you suggested it was unable to connect. When hidden they couldn't find each other... Are there any other prerequisites for the main branch?

tswen, 2023-11-28:
https://github.com/tswen/esp-mesh-lite/tree/feature/support_ssid_hidden supports hidden SSID. However, this branch is just a temporary fix and is not fully compatible with scenarios where users customize both the softap prefix and suffix while hiding the SSID. Therefore, it will take some time to merge this into the master branch.

ThatBigPrint, 2023-11-28:
What I'm asking is what is the rest of the changes? Is it in the .a file?

tswen, 2023-11-28:
Yes

ruslanslobozhenuk, 2023-11-28:
I have 3 ESP32C3 boards. My ssid is hidden and the boards are merged into a mesh. In my scenario, there will be about 100 devices in the room and broadcasting ssids is not allowed! After a day of testing everything works! I would like to ask, if possible, to make it possible to manage logs in the library (I have a timestamp in the logs without color highlighting), the library adds its own logs with highlighting and ticks. I use ESP-INSIGHTS and this leads to disruption of the monitoring system. Overall I like ESP-MESH-LITE. MDF did not handle many scenarios very well. Good work @tswen!

tswen, 2023-11-29:
Do you mean you want to change the log level of mesh_lite's internal output through an API? Do the current logs, etc. have any impact on your application?

ruslanslobozhenuk, 2023-11-29:
I wrote a log message parser. By default, my firmware is set to date and log colors are disabled. When your library reports information to me via log, I have to process it further. This is not exactly what I would like to see. This is absolutely not critical! It just seems to me that this needs to be taken into account when developing. If ESP-IDF provides such an opportunity, then it would be nice to implement it in the library. I don’t quite understand the espressif policy, why close pieces of code in SDK? This quite often brings a lot of inconvenience! ESP-MESH-LITE promises to be a good product, which is why we want to integrate it into our ecosystem.

tswen, 2023-12-04:
> I wrote a log message parser. By default, my firmware is set to date and log colors are disabled. When your library reports information to me via log, I have to process it further. This is not [...]

Okay, we are going to implement an external implementation to encapsulate a log interface for the internal lib to call, so that the outside can control the color of the internal output log.

tswen, 2023-12-04:
> @tswen do you have a local control version of the ota example, please ? im struggling to get it working from docs just need a basic example if you can with a declared url ?

You need to first implement the OTA process for a single device yourself, and then add LAN OTA implementation according to the documentation.

tswen, 2023-12-05:
> I wrote a log message parser. By default, my firmware is set to date and log colors are disabled. When your library reports information to me via log, I have to process it further. This is not [...]

https://github.com/tswen/esp-mesh-lite/tree/feature/add_esp_mesh_lite_log
You can use this branch for testing

ThatBigPrint, 2023-12-09:
> You need to first implement the OTA process for a single device yourself, and then add LAN OTA implementation according to the documentation.

so basically i do the https ota example on a single device then use the mesh ota to distribute?  can you do me a favor and show me a quick demo example ? 

can I ask why are the folk at espressif so ambiguous if it's not related to Rainmaker? i can understand the motive to push Rainmaker but if the pricing for it was more transparent and approachable I would jump straight on it... i know it is not particularly relevant to you directly but, just make the pricing more attractive and not hide the price behind an email enquiry the businesses i represent would be use it straight away....

xcguang, 2023-12-10:
> can I ask why are the folk at espressif so ambiguous if it's not related to Rainmaker? i can understand the motive to push Rainmaker but if the pricing for it was more transparent and approachable I [...]

For Rainmaker price, could you please send leeshaoyu@espressif.com? He can give you a full price about all you want to know, and how to deploy the server.

ThatBigPrint, 2023-12-21:
When is the hidden network option going to be released in the main ?

tswen, 2023-12-25:
The feature of hiding SSID has some additional handling for special forms of SoftAP SSID that needs to be addressed. We are currently working on this, aiming to complete it by January and update it to the master branch.
If you don't have any specific settings for the SoftAP SSID, you can use the temporary branch at https://github.com/tswen/esp-mesh-lite/tree/feature/support_ssid_hidden in the meantime.

tswen, 2024-01-09:
Hello, everyone. The support for the hidden SoftAP SSID network feature has been merged into the master branch. Simultaneously, the example has been updated for easier configuration of SoftAP settings. Below is a code example for setting the SoftAP hidden SSID:
```c
    snprintf((char *)wifi_config.ap.ssid, sizeof(wifi_config.ap.ssid), "%s", CONFIG_BRIDGE_SOFTAP_SSID);
    strlcpy((char *)wifi_config.ap.password, CONFIG_BRIDGE_SOFTAP_PASSWORD, sizeof(wifi_config.ap.password));
    wifi_config.ap.ssid_hidden = true;
    esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_config);
```
[Code Source](https://github.com/espressif/esp-mesh-lite/blob/master/examples/mesh_local_control/main/local_control.c#L185C1-L187C58)

If there are no issues found during testing, you can close this issue.

Nepenthes, 2024-01-15:
你好，我更新了master后 commit 为 82e72f15b1235807de5ed089165f88b0880f843e，我在设置wifi_config.ap.ssid_hidden = true;后，配置网络，发现所有设备都变成了root，无法组网.

tswen, 2024-01-17:
你好，请问是不是有哪些地方配置错误了，mesh id 确定是一致的吗？您可以先使用 mesh_local_control 示例试一下，示例测试下来是没问题的。

closed 2024-03-04 by xcguang