espressif/esp-mesh-lite #49: 透過mesh-lite無法解析主機名稱

8888alex, 2023-12-26:
### 問題
透過mesh連接mqtt服務器時，僅能使用IP位址，若使用主機名稱則子節點無法解析。

### 說明
打算使用TLS對MQTT進行保護，以自簽的憑證部署mosquitto，使用mosquitto_pub及mosquitto_sub驗證可正常收發。
主機位於內部網路且於WIFI的路由器上設定靜態IP位址。
root節點可透過WIFI 的靜態DNS正常解析mqtt主機名稱並完成連接，但子節點卻連接失敗。
由下圖可以看到電腦連接softAP可以取得IP及DNS，但無法解析。
[image]
有辦法可以轉發DNS的查詢嗎?

### 環境
esp32-s2-devkitm-1-v1
IDF v5.0.2

idf_component.yml
dependencies:
idf: ">=5.0"
mesh_lite:
version: ""
espressif/esp_modem:
version: 0.
espressif/esp_modem_usb_dte:
version: 1.*
public: true
rules:

if: "target in [esp32s2, esp32s3]"
usb_device:
path: components/usb/usb_device
git: https://github.com/espressif/esp-iot-bridge.git
rules:
if: "target in [esp32s2, esp32s3]"
espressif/led_strip: "^2.0.0"
lib
version
libesp_mesh_lite_esp32.a :473e10d
libesp_mesh_lite_esp32c2.a :473e10d
libesp_mesh_lite_esp32c3.a :473e10d
libesp_mesh_lite_esp32c6.a :473e10d
libesp_mesh_lite_esp32s2.a :473e10d
libesp_mesh_lite_esp32s3.a :473e10d

tswen, 2023-12-28:
可以输入 ipconfig /all 看一下，dns 服务器地址是多少，目前接入 esp-mesh-lite 设备获取到的 dns 地址都是 114.114.114.114。

8888alex, 2023-12-29:
當我把電腦當成node連上root
確實得到DNS為 114.114.114.114 如下圖
[image]

但是我的MQTT主機是以靜態IP地址的方式寫在ROUTER上的DNS 如下圖
[image]
請問我可以怎樣讓NODE可以得到這個解析IP

tswen, 2023-12-29:
managed_components/espressif__iot_bridge 组件里面 src/bridge_wifi.c 这个文件，里面有个 wifi_event_ap_start_handler API，这个回调函数里面有设置 DNS 地址，你可以将这个 DNS 地址改成你所需要的。

后面我们会更新 iot_bridge 组件，让 softap 能直接分配路由器给 station 的 DNS 地址。

8888alex, 2023-12-29:
> managed_components/espressif__iot_bridge 组件里面 src/bridge_wifi.c 这个文件，里面有个 wifi_event_ap_start_handler API，这个回调函数里面有设置 DNS 地址，你可以将这个 DNS 地址改成你所需要的。

修改有效! 
感謝

closed 2023-12-29 by 8888alex