espressif/esp-mesh-lite #50: In the example ro_router in mesh lite v0.9.0, using esp_mesh_lite_init, ports 6363, 6364, 6365, and 6366 are bound.

sjj1507, 2023-12-27:
idf : ESP-IDF v5.1.2-498-gc57b352725-dirty
chip: esp32c3
esp-mesh-lite component version: 0.9.0

I was wondering what ports 6363, 6364, 6365, and 6366 are used for and if there is a way to disable them.

[image]

tswen, 2023-12-28:
"Mesh-lite" internally also utilizes socket communication, thus occupying these four sockets and ports. May I ask for what reason it is necessary to disable these four ports?

sjj1507, 2024-01-02:
> "Mesh-lite" internally also utilizes socket communication, thus occupying these four sockets and ports. May I ask for what reason it is necessary to disable these four ports?

I asked using 4 random ports.
I assume you are using it for internal communication of mesh lite.

closed 2024-01-02 by sjj1507