espressif/esp-mesh-lite #51: 連接mqtts時會發生憑證失敗

8888alex, 2023-12-30:
### 說明
使用mqtt範例ssl(內建example_connect())建立網路連結可以正常驗證並收發訊息。  如下圖
[image]
[LOG_ssl.txt](https://github.com/espressif/esp-mesh-lite/files/13798126/LOG_ssl.txt)

使用mesh範例local_control將mqtt範例ssl的代碼置入後以mesh-lite建立網路連結，結果無法驗證憑證回報憑證失敗。  如下圖
[LOG_local_control.txt](https://github.com/espressif/esp-mesh-lite/files/13798133/LOG_local_control.txt)
[image]

除了MQTT外使用 pre_encrypted_ota 範例時 esp_https_ota_begin() 也無法驗證憑證。
以上還請播冗核實，謝謝。

### 環境
esp32-s2-devkitm-1-v1
IDF v5.0.2

idf_component.yml
dependencies:
idf: ">=5.0"
mesh_lite:
version: ""
espressif/esp_modem:
version: 0.
espressif/esp_modem_usb_dte:
version: 1.*
public: true
rules:

if: "target in [esp32s2, esp32s3]"
usb_device:
path: components/usb/usb_device
git: https://github.com/espressif/esp-iot-bridge.git
rules:
if: "target in [esp32s2, esp32s3]"
espressif/led_strip: "^2.0.0"
lib
version
libesp_mesh_lite_esp32.a :473e10d
libesp_mesh_lite_esp32c2.a :473e10d
libesp_mesh_lite_esp32c3.a :473e10d
libesp_mesh_lite_esp32c6.a :473e10d
libesp_mesh_lite_esp32s2.a :473e10d
libesp_mesh_lite_esp32s3.a :473e10d

tswen, 2024-01-02:
证书什么的对的吗，请再 check 下 CMake 和所需要的一些配置，确保您真正移植成功了。

8888alex, 2024-01-02:
> 证书什么的对的吗，请再 check 下 CMake 和所需要的一些配置，确保您真正移植成功了。

為此，我反過來將 mesh-lite範例(源自於mast的local_control)加入ssl範例
在確定ssl範例使用example_connect()連線可以正常驗證的前提下
改用local_control_main()去建立連線
第一個錯誤便是無法解析我的自訂主機名稱mqtt.aiotsys.com(但這個錯誤跟之前的不一樣)
這個主機的靜態IP我寫在路由器的DNS設定裡面，後去尋找將114.114.114.114的位置改成我的路由器但無效依舊無法解析
隨附 [LOG.txt](https://github.com/espressif/esp-mesh-lite/files/13812525/LOG.txt)
及我的範例專案[ssl.zip](https://github.com/espressif/esp-mesh-lite/files/13812526/ssl.zip)

tswen, 2024-01-03:
[ssl.tar.gz](https://github.com/espressif/esp-mesh-lite/files/13815573/ssl.tar.gz)
[image]

我将 mesh 移植到 ssl 示例里，测试没有问题。

8888alex, 2024-01-05:
您是對的，您給的範例確實可以正常運行。
但我將我的專案全部註解僅放入您的範例卻依舊發生錯誤。
我不理解VSCODE中是否有隱藏甚麼導致兩者的結果不同。
若您方便可否給個查找問題方向?
謝謝您的時間。

8888alex, 2024-01-05:
經過反覆的交叉測試，最後鎖定在sdkconfig的差異。
用您範例附帶的sdkconfig編譯可正常執行，但若將sdkconfig reset則編譯後運行便會報錯。
```
I (47672) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x8017
I (47678) MQTTS_EXAMPLE: Last tls stack error number: 0x7f00
I (47684) MQTTS_EXAMPLE: Last captured errno : 0 (Success)
I (47691) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (57696) MQTTS_EXAMPLE: Other event id:7
E (57713) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (57713) esp-tls: create_ssl_handle failed
E (57713) esp-tls: Failed to open new connection
E (57719) transport_base: Failed to open a new connection
E (57727) mqtt_client: Error transport connect
I (57732) MQTTS_EXAMPLE: MQTT_EVENT_ERROR
I (57734) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x8017
I (57741) MQTTS_EXAMPLE: Last tls stack error number: 0x7f00
I (57748) MQTTS_EXAMPLE: Last captured errno : 0 (Success)
I (57754) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
```

比對兩者的差異
[image]
但無法設定CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER
不知為何您的sdkconfig有DYNAMIC可以選RESET後就沒有了
[image]
[image]

也不知這是否是必要的設定，
請問您在reset後需要進行那些必要設定才能正常運行。

8888alex, 2024-01-05:
我覺得我找到原因了。

Type of WiFi TX buffers 必須要是 DYNAMIC mqtts才能正常驗證。
[image]

而  PSRAM 必須要 unchecked，Type of WiFi TX buffers才有 DYNAMIC 可以選。
[image]

而 RESET sdkconfig 後 PSRAM 默認是 checked的，導致Type of WiFi TX buffers為static。
所以用RESET後的默認 sdkconfig 驗證過程便會報錯。

現在的問題變成，
我要checked PSRAM才能有額外的 2MB可以用，
但我又需要Type of WiFi TX buffers 為 DYNAMIC 才能正常驗證，
請問我能怎麼做?
又或者是我搞錯了?

8888alex, 2024-01-06:
最後我開啟了psram 並且
SPI RAM access method 選取 Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM)
同時 核取 Try to allocate memories of WiFi and LWIP in SPIRAM firstly. If failed, allocate internal memory
及 Allow .bss segment placed in external memory
不知道我這樣設定對不對，但內存似乎得到緩解，
```
I (863) IoTSys: app_main: total: 2266144 used: 19948 free: 2246196 ... (remain 99.12%)
I (1154) IoTSys: gpio_app_main: total: 2266144 used: 59788 free: 2206356 ... (remain 97.36%)
I (1228) IoTSys: sysStart: total: 2266144 used: 65012 free: 2201132 ... (remain 97.13%)
I (1250) IoTSys: nvs: total: 2266144 used: 65028 free: 2201116 ... (remain 97.13%)
I (1258) IoTSys: nvs_flash_init: total: 2266144 used: 65028 free: 2201116 ... (remain 97.13%)
I (1268) IoTSys: esp_netif_init: total: 2266144 used: 69364 free: 2196780 ... (remain 96.94%)
I (1277) IoTSys: esp_event_loop_create_default: total: 2266144 used: 72760 free: 2193384 ... (remain 96.79%)
I (1546) IoTSys: esp_bridge_create_all_netif: total: 2266144 used: 180860 free: 2085284 ... (remain 92.02%)
I (2924) IoTSys: wifi_init: total: 2266144 used: 181076 free: 2085068 ... (remain 92.01%)
I (2983) IoTSys: esp_mesh_lite_start: total: 2266144 used: 195076 free: 2071068 ... (remain 91.39%)
```
最重要的是 mqtts 不報錯，可以正常運行了。
我先關閉這個Issues了。
如果我哪裡錯了還請提醒，感謝您的時間。

closed 2024-01-06 by 8888alex