espressif/esp-mesh-lite #52: 新年快乐，请教应该采取什么方案

StarCompute, 2023-12-31:
其实我的目的就是想多个esp32设备组网做语音对讲机，基于esp-now的方案网上都有，但是它的通信距离太短。
基于esp wifi mesh 的通讯我已经试过了，但是它需要设置路由器，没有路由器的方案没有折腾通，折腾了好多天，求示例，谢谢。

也看过mwifi的介绍。
后来看你们对于lite的介绍好像lite可以无路由自由通信，但是从api上通信又是父广播给子或者子发送给父，始终觉得别扭：想要无parent，或者完全投票选举parent的方案：

 一堆esp（不超过20个） 开机就自己组网，不需要什么路由，任何一个esp发信息都可以广播式都能收到，所以到底应该选择什么方案？
第三方的painless的方案我是知道的，但是它不能发二进制。
谢谢，待复。

tswen, 2024-01-02:
如果说是无路由器，任何一个esp发信息都可以广播式都能收到的话，更推荐使用 espnow，espnow 可以通过开启 LR 模式 增加通信距离。https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L57

StarCompute, 2024-01-02:
> 如果说是无路由器，任何一个esp发信息都可以广播式都能收到的话，更推荐使用 espnow，espnow 可以通过开启 LR 模式 增加通信距离。https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L57

espnow的通讯距离好像很短，我是想用wifi mesh 方案，无遮挡可以200米。

tswen, 2024-01-02:
espnow 也是用的用的 wifi，一样的。

StarCompute, 2024-01-02:
> espnow 也是用的用的 wifi，一样的，只不过是无连接的一个 wifi 管理帧，你们也不需要更复杂的 mesh 管理，只需要广播通信就行了吧。

谢谢，那我就先试试esp now的通信距离。

StarCompute, 2024-01-04:
原谅我继续来请教一下：esp now需要获取其他设备的mac地址，除了提前获取有没有其他方法；你看wifi mesh 除了要求router外并没有这个麻烦的需求。 
虽然好像可以通过wifi scan，但是依然觉得非常不方便；所以目前那种乐鑫的方案可以： 自由组网，无路由器，不需要提前知道mac地址，节点可以随时加入，实现这样的有示例吗？

tswen, 2024-01-04:
mesh-lite 有无路由器示例，你可以试下。espnow 只有单播需要添加对方 mac 地址至 peer，广播的话只需要添加广播地址就好了。

closed 2024-03-04 by xcguang

iwonako, 2024-12-23:
> 虽然好像可以通过wifi scan，但是依然觉得非常不方便；所以目前那种乐鑫的方案可以： 自由组网，无路由器，不需要提前知道mac地址，节点可以随时加入，实现这样的有示例吗？

你好，请问最后有找到合适的实现方案吗

iwonako, 2024-12-24:
> mesh-lite 有无路由器示例，你可以试下。espnow 只有单播需要添加对方 mac 地址至 peer，广播的话只需要添加广播地址就好了。

esp now 中未配对的节点能起中继作用吗，比如点对点的通信能否让未配对的节点作为中继？

tswen, 2024-12-24:
> > mesh-lite 有无路由器示例，你可以试下。espnow 只有单播需要添加对方 mac 地址至 peer，广播的话只需要添加广播地址就好了。
> 
> esp now 中未配对的节点能起中继作用吗，比如点对点的通信能否让未配对的节点作为中继？

如果使用的是 mesh lite 方案，espnow 通信的中继和转发需要自己应用层实现，如果仅仅使用 espnow 通信不用 mesh lite 组件，可以参考 https://github.com/espressif/esp-now