espressif/esp-mesh-lite #53: New functional (pause or stop mesh)

ruslanslobozhenuk, 2024-01-04:
Hi!
I use an EnOcean switch without a battery, which sends a BLE message when pressed/released. 
Theoretically, I can skip the click (the user will repeat the click), but if I catch the click, then I need to stop WIFI so that I can scan BLE constantly. 
Is it possible to implement esp_mesh_lite_pause(...) or esp_mesh_lite_stop(...)?

tswen, 2024-01-05:
Hello, after mesh-lite devices form a network, they will only maintain some states through beacon or internal communication. I'm not quite clear on what you mean by "stop Mesh." Are you referring to making devices that have already formed a network revert to being idle nodes, and then deinitiating the WiFi?

ruslanslobozhenuk, 2024-01-05:
Here is part of my code that handles the button click in the BLE GAP notification processing function:
...
if (status & BIT(0))//EnOcean button push, coex configure all time to BLE, WiFi stop
{
	esp_coex_status_bit_set(ESP_COEX_ST_TYPE_BLE, ESP_COEX_BLE_ST_MESH_CONFIG);
	esp_wifi_stop();		
}
else //EnOcean button release, coex configure to BLE scan, WiFi start
{
	esp_coex_status_bit_clear(ESP_COEX_ST_TYPE_BLE, ESP_COEX_BLE_ST_MESH_CONFIG);
	esp_wifi_start();
}		
...
Since the EnOcean switch is powered by a piezo element, it will only send 3 notifications every 20 milliseconds, and I need to receive at least one of them. Therefore, it is necessary to stop WiFi so that all the time is given to BLE.

This code works! But I suspect there will be a lot of overhead when reconnecting to the mesh lite.

tswen, 2024-01-09:
Hello, this approach is acceptable. However, we recommend using the latest version from the master branch ([latest version](https://github.com/espressif/esp-mesh-lite), commit id: 82e72f15).
In the latest version, issues related to the impact of application-layer calls to `esp_wifi_stop` on internal Wi-Fi reconnection have been addressed.

ruslanslobozhenuk, 2024-01-09:
After stop WiFi at last commit master branch, i have a problem:

I (27461) wifi:state: run -> init (0)
I (27461) wifi:pm stop, total sleep time: 10144404 us / 25938444 us

I (27471) wifi:<ba-del>idx:0, tid:0
I (27471) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (27671) wifi:flush txq
I (27671) wifi:stop sw txq
I (27671) wifi:lmac stop hw txq
W (31121) [vendor_ie]: esp_mesh_lite_wifi_connect[246] return ESP_FAIL:0x3002
I (31121) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP

***ERROR*** A stack overflow in task Tmr Svc has been detected.
Core  0 register dump:
MEPC    : 0x40380796  RA      : 0x4038935c  SP      : 0x3fc98f40  GP      : 0x3fc95800
TP      : 0x3fc6a4c4  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000002
S0/FP   : 0x00000003  S1      : 0x80000004  A0      : 0x3fc98f58  A1      : 0x3c11431c
A2      : 0x00000003  A3      : 0x3fc98f85  A4      : 0x00000001  A5      : 0x3fc9d000
A6      : 0x3fc99624  A7      : 0x0000000a  S2      : 0x00001881  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x40380206  S11     : 0x00001881
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000021  T6      : 0x676e6974
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc98f40: 0x00001881 0x3fce0000 0x3fce0000 0x4038a6e6 0xfffbffff 0x00081e89 0x452a2a2a 0x524f5252
3fc98f60: 0x202a2a2a 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x726d5420
3fc98f80: 0x63765320 0x73616820 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000
3fc98fa0: 0x00000000 0x3c1142e0 0x3fca5b24 0x3c114308 0x00001881 0x80000004 0x00000000 0x4038bf48
3fc98fc0: 0x00000000 0x00000000 0x00000000 0x4038b524 0x60023000 0x40388e24 0x40388e30 0x00000000
3fc98fe0: 0x3fc98fdc 0x00000000 0x00000000 0x00000000 0x3fc98ff4 0xffffffff 0x3fc98ff4 0x3fc98ff4
3fc99000: 0x00000000 0x3fc99008 0xffffffff 0x3fc99008 0x3fc99008 0x00000001 0x00000001 0x00000000
3fc99020: 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc99030 0x00000000 0x00000000
3fc99040: 0x00000000 0x3fc99048 0xffffffff 0x3fc99048 0x3fc99048 0x00000000 0x3fc9905c 0xffffffff
3fc99060: 0x3fc9905c 0x3fc9905c 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0x00000000
3fc99080: 0x00000000 0x60023000 0x40388e24 0x40388e30 0x3fca1e38 0x3fca1efc 0x3fca1fc0 0x3fca6414
3fc990a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc990c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc990e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99100: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99120: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99140: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99160: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99180: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc991a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc991c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc991e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99200: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99220: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99240: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99260: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc99280: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc992a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00058100 0x00000000 0x00000000 0x00000000
3fc992c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc992e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x33505345
3fc99300: 0x00000032 0x636f6e65 0x006e6165 0x3b305b1b 0x496d3233 0x6c252820 0x25202975 0x00000047
3fc99320: 0x00000057 0x0032004d 0x5f505345 0x64697242 0x00006567 0x34333231 0x38373635 0x00000000

tswen, 2024-01-10:
What version of IDF and chip are you using? You can try increasing the `FREERTOS_TIMER_TASK_STACK_DEPTH` configuration item.

ruslanslobozhenuk, 2024-01-10:
I use ESP32C3 for testing. After increasing FREERTOS_TIMER_TASK_STACK_DEPTH to 4096, no error is observed.

closed 2024-01-12 by ruslanslobozhenuk