espressif/esp-mesh-lite #54: release版本

heyangxinmei, 2024-01-08:
请问什么时间会出release版本呢？现在这个master分支可以用在大批量产品上吗？谢谢

xcguang, 2024-01-08:
现在我们采用组件管理、小版本快速迭代进行版本的升级，可参见每个版本的修改记录 https://components.espressif.com/components/espressif/mesh_lite，也是推荐客户采用最新的小版本进行开发。

[image]

heyangxinmei, 2024-01-08:
[image]
那我可以直接用master进行开发吗？

xcguang, 2024-01-08:
可以，有问题可以给我们直接反馈，我们会进行处理

heyangxinmei, 2024-01-09:
好的

heyangxinmei, 2024-01-09:
mesh-lite组到同一个网络中的设备，登录路由器的账号和密码是不是要相同啊？还是通过其他方式来区分不同的设备进入不同的网络呢？

tswen, 2024-01-09:
同一个 Mesh-Lite 组网的设备 路由器信息、Mesh ID 都是要一样的。
如果已经有一个配过网的设备，后面加入的子节点设备如果没有配过网，只要 Mesh ID 也是能加入这个根节点的。

有关 Mesh-Lite 的具体信息可以看下这个指导文档：https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide_CN.md

heyangxinmei, 2024-01-10:
[image]
三个设备组网测试中，出现这个报错，是什么原因呢？

tswen, 2024-01-10:
存在于根节点意外跟路由器断开连接或者硬件重启，其在重新寻找新的父节点时，为了避免连接到它子节点形成环形的一些提示，不需要关心这部分日志，另外昨天更新的 0.10.0 版本已将这部分日志改成了 Debug 等级，可以更新下 mesh-lite 组件。

heyangxinmei, 2024-01-11:
[image]
这个错误是什么问题呢？

tswen, 2024-01-11:
因为内部在扫描的时候为了节省一点时间，做了一些扫描参数的配置，但如果打开蓝牙的话，Wi-Fi 就会直接使用默认扫描配置，不会有什么影响，只是会打印这个提示。

tswen, 2024-01-11:
VID0、VID1、MeshID（对应2）和 路由器信息（对应3）要保持一直才能在一个组网中。

softap ssid（对应1）可以不一致。softap password 要保持一致。

heyangxinmei, 2024-01-13:
[image]
有一个子节点隔一段时间就会断开，重连。是什么原因呢？

heyangxinmei, 2024-01-15:
节点是第二层的，一共4台设备，idf的版本是esp-idf-v4.4.5。
[Uploading [Com COM8] (2024-01-13_171631).log…]()
没有太多的日志了，有问题的日志就这么多。
[image]

heyangxinmei, 2024-01-15:
[image]
[image]
应该是这个原因引起的与ap的断连。该节点是在第二层，一共有4台设备。

heyangxinmei, 2024-01-19:
mesh-lite默认的组网的信号强度是多少呢？信号强度怎么配置呢？

heyangxinmei, 2024-01-19:
mesh-lite默认组网的信号强度是多少呢？信号强度怎么配置呢？

tswen, 2024-01-29:
可以通过 esp_mesh_lite_rssi_threshold_list_t 来设置。
待组网设备在寻找最佳父节点时，会根据周围已经组网成功节点的信号强度（相对于当前设备）以及层级进行判断，待组网设备正常来说会优先选择层级数较低（即越靠近根节点）的节点作为最佳父节点，除非有一个高层级的节点信号强度高于低层级节点信号强度某个阈值（阈值默认为 15dB）

这个接口可以阶梯型设置不同信号强度范围内的一个阈值大小

{0,  -35, 20} 代表设备在寻找最佳父节点时，如果周围已经组网的节点信号强度（相对于当前设备）为 0 到 -35 之间，阈值是 20dB
{-55,  -75, 10} 代表设备在寻找最佳父节点时，如果周围已经组网的节点信号强度（相对于当前设备）为 -55 到 -75 之间，阈值是 10dB

tswen, 2024-01-29:
后面我们也会提供一个接口来设置组网时的一个最小阈值，低于这个阈值的设备的信号强度将不会被节点选择为候选父节点。

heyangxinmei, 2024-02-26:
[image]
组网的设备能够进入网络，但是长时间不能获取IP是什么问题啊？出现这种情况我该怎么处理呢？

closed 2024-03-01 by heyangxinmei