espressif/esp-mesh-lite #56: Multiple esp32S3, using mesh-lite questions

yel-best, 2024-01-22:
Hi, Espressif
I am doing an esp mesh evaluation, so I have some questions. I hope you can answer them. thanks.

* How many nodes can we currently support the most?
* If I have 10 nodes, after using this, I can independently connect to **mqtt or http** to send msg directly to the server, and do they also support OTA upgrades?
* What is the network latency or connection read like?

It would be best if there are examples of mesh-lite & mqtt connect that you can refer to, thanks 😃

xcguang, 2024-01-22:
> How many nodes can we currently support the most?

We tested about one hundred nodes, and we recommend build a mesh network less than 50 nodes
 
> If I have 10 nodes, after using this, I can independently connect to mqtt or http to send msg directly to the server, and do they also support OTA upgrades?

Yes

yel-best, 2024-01-22:
> We tested about one hundred nodes, and we recommend build a mesh network less than 50 nodes
>
>
> Yes

the recommendation around a single root node is 50 nodes? so if I add multiple root nodes in the same environment, does it mean that I can use 100, 200, or more nodes in the same environment?

thank you very much for taking time out of your busy work to answer my questions.

xcguang, 2024-01-22:
the nodes number is limited to channel interference, all mesh nodes under one router are in the same channel, so the mesh nodes are not always be added successfully.

yel-best, 2024-01-22:
> the nodes number is limited to channel interference, all mesh nodes under one router are in the same channel, so the mesh nodes are not always be added successfully.

ok, If I want to use 200 nodes, is there a recommended solution? Or how should I implement it?

xcguang, 2024-01-22:
If the devices are not too close, channel interference is not severe, maybe 200 nodes work well. Otherwise, I think add one router which use the different channel is the best choice.

yel-best, 2024-01-23:
> If the devices are not too close, channel interference is not severe, maybe 200 nodes work well. Otherwise, I think add one router which use the different channel is the best choice.

ok，**System information, channel: 1, layer: 2**

I saw a layer log in the local_control example. Regarding whether the layer is the same as the description of wifi-mesh in esp-idf,

**Maximum Downstream Connections to Accept: 6
Maximum Permissible Layers: 6**

Will mesh-lite also be subject to this limitation?

[image]

yel-best, 2024-01-24:
Hi @xcguang 

I currently see that if I have multiple esp32 using mesh-lite, their respective WIFI SSIDs will be created according to esp_xxxxxx after the final run. Can this SSID be hidden? Let mesh-lite be networked and managed in the hidden SSID, or I can let all mesh-lite nodes use the SSID with the same name. Is that okay? Otherwise if I have 200 esp using mesh-lite, the WIFI list will have 200 different SSIDs

[image]

thx

tswen, 2024-01-29:
[image]

yel-best, 2024-01-31:
> ![Screenshot from 2024-01-29 17-56-05](https://private-user-images.githubusercontent.com/55544524/300411558-8a2544c7-b1c5-497e-b64b-2994758142fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY2NzExODksIm5iZiI6MTcwNjY3MDg4OSwicGF0aCI6Ii81NTU0NDUyNC8zMDA0MTE1NTgtOGEyNTQ0YzctYjFjNS00OTdlLWI2NGItMjk5NDc1ODE0MmZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMxVDAzMTQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3YWE2NGI3NTUzOWI0MjY4NWQ4NjllN2VlYjdjZDU0OGNhZmQ3MTA5OGM1ODY0NmE5ZGYwMjE2YjAzNTQ0NmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xu00J6oBJ9M3W6ZqGDoqc4k_YoGzfTjiz2osUYMuAvM)

yes , thank you for your guidance. This method is effective. Thank you very much. 😄

closed 2024-02-27 by yel-best

danielewyss, 2024-09-02:
> the nodes number is limited to channel interference, all mesh nodes under one router are in the same channel, so the mesh nodes are not always be added successfully.

Hi,
I'm looking to have some tool to analyse the wifi to have some kind of tool to get a feeling of how hostile is the location where I want to install the devices communicating over the mesh.
I found out that the assigned MESH channel corresponds to the DSSS but the communication looks like going over multiple channels, this makes more difficult to make the analysis....
Does somebody knows if what I saw is really correct? and maybe how to know the mechanism used to determinate the used channels ?
Thanks

tswen, 2024-09-05:
There are many apps on Google Play and other app stores that can detect the hotspot signals and number of each WiFi channel. You can download them and try them out.