espressif/esp-mesh-lite #58: mesh_lite_local_control  &  use esp-modbus a compilation error occurred

yel-best, 2024-01-30:
When I add esp-modbus to the mesh_lite_local_control example of mesh-lite, the following error occurs during compilation. Is it because of incompatibility?

[image]

```C
In file included from E:/MyProject/IoT/esp/esp-idf/components/esp_common/include/esp_check.h:9,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_common.h:18,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_master.h:13,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/mbcontroller.h:20,
                 from E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c:10:
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c: In function 'master_init':
E:/MyProject/IoT/esp/esp-idf/components/log/include/esp_log.h:265:27: error: format '%x' expects argument of type 'unsigned int', but argument 8 has type 'long unsigned int' [-Werror=format=]
  265 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

E:/MyProject/IoT/esp/esp-idf/components/log/include/esp_log.h:340:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  340 | #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
E:/MyProject/IoT/esp/esp-idf/components/esp_common/include/esp_check.h:264:13: note: in expansion of macro 'ESP_LOGE'
  264 |             ESP_LOGE(log_tag, "%s(%d): " format, __FUNCTION__, __LINE__, ##__VA_ARGS__);        \
      |             ^~~~~~~~
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/managed_components/espressif__esp-modbus/freemodbus/common/include/esp_modbus_common.h:21:59: note: in expansion of macro 'ESP_RETURN_ON_FALSE'       
   21 | #define MB_RETURN_ON_FALSE(a, err_code, tag, format, ...) ESP_RETURN_ON_FALSE(a, err_code, tag, format __VA_OPT__(,) __VA_ARGS__)
      |                                                           ^~~~~~~~~~~~~~~~~~~
E:/MyProject/IoT/PlatformIO-ESP/esp-idf-mesh-lite/mesh-lite-test/main/app_power_modbus.c:208:5: note: in expansion of macro 'MB_RETURN_ON_FALSE'
  208 |     MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
      |     ^~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

```

yel-best, 2024-01-30:
But when I comment out the code about **MB_RETURN_ON_FALSE**, it compiles normally. Is it because of compatibility issues?

// MB_RETURN_ON_FALSE((master_handler != NULL), ESP_ERR_INVALID_STATE, TAG,
//                    "mb controller initialization fail.");
// MB_RETURN_ON_FALSE((err == ESP_OK), ESP_ERR_INVALID_STATE, TAG,
//                    "mb controller initialization fail, returns(0x%x).",
 //    (uint32_t)err);

[image]

tswen, 2024-02-03:
Printed type does not match actual type

error: format '%x' expects argument of type 'unsigned int', but argument 8 has type 'long unsigned int' [-Werror=format=]

yel-best, 2024-02-04:
> Printed type does not match actual type
> 
> error: format '%x' expects argument of type 'unsigned int', but argument 8 has type 'long unsigned int' [-Werror=format=]

yes, this question has been answered by the modbus components team. Thank you very much for your concern, thanks

closed 2024-02-04 by yel-best