espressif/esp-mesh-lite #59: How should I proceed with OTA

yel-best, 2024-02-01:
I have completed the mesh-lite application, but the OTA part is still missing. I found some information, but I can't run it normally. Please give me some opinions, thk

I looked through the previous issue and found a link like this, which shows that OTA will be performed through a method called `mesh_lite_LAN_OTA`. My current scenario is to receive the URL of an OTA Bin file by monitoring the topic, and then I want to receive this The node of msg can download the bin package and upgrade it according to this URL, and then you can see the status and stage of the upgrade.

https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota

I saw this in another repository and branch, so does the latest version of mesh-lite also support this approach? Can you give me some reference examples or give me some guidance? Thanks

yel-best, 2024-02-04:
oh...can anyone help me, thanks 🥲

tswen, 2024-02-18:
The latest Mesh-Lite supports mesh-lite-lan-ota, but you need to turn on ESP_MESH_LITE_OTA_ENABLE through menuconfig, and then combine some operations of mesh lite lan ota with your own OTA process according to the [documentation](https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota).