espressif/esp-mesh-lite #61: Mesh Lite An exception occurred after OTA was added. Procedure

yel-best, 2024-02-06:
After referring to this example to add the OTA function based on MESH Lite, only the master node can connect to wifi & mqtt, but although the child node can enter the mesh, it cannot connect to mqtt through the network of the master node, and the following error occurs.

My child nodes will have tasks after init mesh lite, connect to mqtt and send msg periodically. The master node can connect to wifi & mqtt normally.

https://github.com/espressif/esp-mesh-lite/tree/master/examples/rainmaker/led_light

```C
I (35988) app_main_mesh: System information, channel: 1, layer: 2, self mac: 34:85:18:6a:a2:00, parent bssid: 34:85:18:6a:53:85, parent rssi: -22, free heap: 241368
E (45667) esp-tls: [sock=58] select() timeout
E (45668) transport_base: Failed to open a new connection: 32774
E (45668) mqtt_client: Error transport connect
I (45673) app_mqtt: MQTT_EVENT_ERROR
E (45677) app_mqtt: Last error reported from esp-tls: 0x8006
I (45683) app_mqtt: Last errno string (Success)
I (45689) app_mqtt: MQTT_EVENT_DISCONNECTED
W (45693) app_bee: Unhandled Bee Common Event: 4
```

app_main function code 
```C
ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

    esp_mesh_lite_start();
 
    esp_bee_mqtt_init(mqtt_conn_params);

    /* Enable OTA */
    esp_bee_ota_config_t ota_config = {
        .server_cert = "123456",
        .ota_cb      = esp_bee_mesh_lite_ota_cb,
    };
    esp_demo_ota_enable(&ota_config, OTA_USING_TOPICS);

    esp_demo_mesh_lite_ota_init(&ota_config);

// Where esp_demo_ota_enable(&ota_config, OTA_USING_TOPICS);
// esp_demo_mesh_lite_ota_init(&ota_config); The contents of the two functions are consistent with the examples

```

[image]

https://github.com/espressif/esp-mesh-lite/blob/master/examples/rainmaker/led_light/components/app_rainmaker/app_rainmaker_ota.h

tswen, 2024-02-18:
```
# LWIP
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

CONFIG_MESH_LITE_ENABLE=y
```

Please make sure that these configuration items are all turned on, whether it is the root node or the child node.

yel-best, 2024-02-27:
> ```
> # LWIP
> CONFIG_LWIP_IP_FORWARD=y
> CONFIG_LWIP_IPV4_NAPT=y
> 
> CONFIG_MESH_LITE_ENABLE=y
> ```
> 
> Please make sure that these configuration items are all turned on, whether it is the root node or the child node.

Oh, it ended up being a problem with my network that prevented the https bin file from downloading. thanks

closed 2024-02-27 by yel-best