espressif/esp-mesh-lite #63: An explanation of how to implement and docs on new functions

ThatBigPrint, 2024-02-22:
Hi guys,

I think this may be the opinion of manny but i think it would be a good idea to explain in basic terms as to what some new functions do vs ambiguous terms, for instance fusion, i presume this is some kind of time synchronisation method to trigger events in a time critical manor ? 

also the ota stuff just telling people to figure it out from the rainmaker example is not ideal... most of us are buying your chips not cloud services 

I do love using espressif stuff, but im finding it harder and harder to justify onboarding new clients....

tswen, 2024-02-23:
"Fusion" refers to a mechanism within the mesh network where multiple root nodes are merged into one. Do you have any other aspects that you're not quite clear about?

Additionally, regarding OTA methods, there are indeed various approaches. In the mesh-lite repository, we happen to utilize the OTA functionality from Rainmaker. For most customers, their projects typically require further development based on mesh-lite, so they cannot solely rely on the example code in the repository for mass production.

If you have any further questions, feel free to ask, and we'll do our best to address them.

tswen, 2024-03-05:
Regarding HTTPS examples, there are many in esp-idf/examples that you can use as a reference. The main purpose of mesh-lite is to allow devices to access network resources through routers without directly connecting to them. However, the operations at the application layer remain the same.

As for the time synchronization method you mentioned, these are all application-layer matters, and there are many ways to achieve this application. For example, devices can send a broadcast command via ESP-NOW to notify all devices to execute corresponding operations, or synchronize via network time synchronization.