espressif/esp-mesh-lite #65: Mesh-lite WIFI internet event cb

yel-best, 2024-02-27:
I consulted some documents. I need an event that can monitor changes in the mesh lite network. 
For example, 
when mesh connected, a callback event is performed, and mesh disconnected triggers a callback event. This makes it easier for me to monitor the network status. For example, I can know what Sometimes network fluctuations are triggered within the mesh, or there are some events that I need to perform after mesh connect, such as ntp sync.

please help me,thank you

tswen, 2024-03-03:
Mesh-Lite is entirely based on Wi-Fi's SoftAP+Station mode, so you can fully utilize Wi-Fi-related events such as WIFI_EVENT_STA_CONNECTED, WIFI_EVENT_STA_DISCONNECTED, WIFI_EVENT_AP_STACONNECTED, WIFI_EVENT_AP_STADISCONNECTED.

yel-best, 2024-03-14:
You're right, it works for me, thank you for your help, thank you 😄

closed 2024-03-14 by yel-best