espressif/esp-mesh-lite #66: Example usage for MQTT

jorgie0, 2024-02-28:
No details are available for how to utilise MQTT while running ESP MESH LITE.
The user guide mentions that MQTT can be used.
Would be an excellent improvement if an example was created that showed MQTT usage.

xcguang, 2024-02-29:
You can use MQTT like a general Wi-Fi device. MQTT connects to the MQTT broker directly.

jorgie0, 2024-02-29:
> You can use MQTT like a general Wi-Fi device. MQTT connects to the MQTT broker directly.

Thank you for a very unhelpful answer. The request remains extant:
"how to utilise MQTT while running ESP MESH LITE?"
Please feel free to provide a link to how to implement a solution

ThatBigPrint, 2024-02-29:
@jorgie0 just to help out a tad here, basicaly treat the esp mesh just like wifi connect it should be a drop in replacement this is an example from the migration guide ```/*Event` handler for catching system events */
static void event_handler(void* arg, esp_event_base_t event_base,
                          int32_t event_id, void* event_data)
{
    if (event_base == WIFI_PROV_EVENT) {
        switch (event_id) {
            case WIFI_PROV_START:
                ESP_LOGI(TAG, "Provisioning started");
                break;
            case WIFI_PROV_CRED_RECV: {
                wifi_sta_config_t *wifi_sta_cfg = (wifi_sta_config_t *)event_data;
                ESP_LOGI(TAG, "Received Wi-Fi credentials"
                         "\n\tSSID     : %s\n\tPassword : %s",
                         (const char *) wifi_sta_cfg->ssid,
                         (const char *) wifi_sta_cfg->password);
#if CONFIG_MESH_LITE_ENABLE
                mesh_lite_sta_config_t config;
                memcpy((char*)config.ssid, (char*)wifi_sta_cfg->ssid, sizeof(config.ssid));
                memcpy((char*)config.password, (char*)wifi_sta_cfg->password, sizeof(config.password));
                config.bssid_set = wifi_sta_cfg->bssid_set;
                if (config.bssid_set) {
                    memcpy((char*)config.bssid, (char*)wifi_sta_cfg->bssid, sizeof(config.bssid));
                }
                esp_mesh_lite_set_router_config(&config);
                esp_mesh_lite_connect();
#else
                esp_wifi_set_storage(WIFI_STORAGE_FLASH);
                esp_wifi_set_config(ESP_IF_WIFI_STA, (wifi_config_t*)wifi_sta_cfg);
                esp_wifi_set_storage(WIFI_STORAGE_RAM);

                esp_wifi_disconnect();
                esp_wifi_connect();
#endif /* CONFIG_MESH_LITE_ENABLE */
                break;
            }
        }
    }
}```

all you need to do is use the mesh lite to get the wifi going instead of the normal wifi, and the exising in this case mqtt example will just run as before, just through other nodes vs each one connecting to router

jorgie0, 2024-03-01:
@ThatBigPrint 
Thanks, that does provide an indication of the direction I need to go.
While I am a software engineer I'm not a C coder. I can follow established patterns to get the functionality that I want, I'm next to useless with generating code when trying to expand on snippets of code. 
It would still be nice if Espressif were able to provide some more advanced examples.

jorgie0, 2024-03-01:
@ThatBigPrint 
After a few hours of playing around I have managed to get a working example of MQTT over ESP MESH LITE, there is an initial hiccup at the start:
`I (6976) bridge_wifi: SoftAP IP network segment has changed, deauth all station
I (6984) [vendor_ie]: RTC store: ssid:ESP_Bridge_0c39ad; bssid:e8:9f:6d:0c:39:ad crc:2216604720 
E (8363) esp-tls: couldn't get hostname for :mqtt.eclipseprojects.io: getaddrinfo() returns 202, addrinfo=0x0
E (8364) transport_base: Failed to open a new connection: 32769
E (8370) mqtt_client: Error transport connect
I (8375) local_control: MQTT_EVENT_ERROR
E (8379) local_control: Last error reported from esp-tls: 0x8001
I (8386) local_control: Last errno string (Success)`

After that it correctly connects to the broker and sends and received messages as expected

Full code is:

`
#include <inttypes.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"

#include "esp_wifi.h"
#include "nvs_flash.h"
#include <sys/socket.h>
#include "protocol_examples_common.h"
#include "freertos/semphr.h"
#include "freertos/queue.h"

#include "lwip/sockets.h"
#include "lwip/dns.h"
#include "lwip/netdb.h"

#include "esp_log.h"
#include "mqtt_client.h"

#define CONFIG_BROKER_URL "mqtt://mqtt.eclipseprojects.io"

#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 4, 0)
#include "esp_mac.h"
#endif

#include "esp_bridge.h"
#include "esp_mesh_lite.h"

#define PAYLOAD_LEN       (1456) /**< Max payload size(in bytes) */

static int g_sockfd    = -1;
static const char *TAG = "local_control";

static void log_error_if_nonzero(const char *message, int error_code)
{
    if (error_code != 0) {
        ESP_LOGE(TAG, "Last error %s: 0x%x", message, error_code);
    }
}

/*
 * @brief Event handler registered to receive MQTT events
 *
 *  This function is called by the MQTT client event loop.
 *
 * @param handler_args user data registered to the event.
 * @param base Event base for the handler(always MQTT Base in this example).
 * @param event_id The id for the received event.
 * @param event_data The data for the event, esp_mqtt_event_handle_t.
 */
static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%" PRIi32 "", base, event_id);
[... 174 lines cut ...]

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

    esp_mesh_lite_start();

    /**
     * @breif Create handler
     */
    mqtt_app_start();

    TimerHandle_t timer = xTimerCreate("print_system_info", 10000 / portTICK_PERIOD_MS,
                                       true, NULL, print_system_info_timercb);
    xTimerStart(timer, 0);
}
`