espressif/esp-mesh-lite #67: 测试组网设备10台，设备进入网络，但是个别设备很长时间获取不到IP地址

heyangxinmei, 2024-03-01:
[image]

组网的设备能够进入网络，但是长时间不能获取IP是什么问题啊？出现这种情况我该怎么处理呢？

tswen, 2024-03-03:
你用的 idf 是哪个版本呢，这种情况容易复现吗？是否可以提供下空口包呢

heyangxinmei, 2024-03-04:
[image]
idf版本是不是这样看啊？应该是V4.4.5。也经常复现。空口包指的是什么意思啊？

tswen, 2024-03-04:
可以搜索一下如果抓取 Wi-Fi 空口包，如果没有条件的话，可以更新 idf 到 5.0 再试试，目前使用 idf5.0 的客户都没有遇到获取不到 IP 的问题

heyangxinmei, 2024-03-07:
[image]
esp-mesh-lite组网的softap 中的password值不能带字符，只能是数字啊？我带字符组网就会失败。

tswen, 2024-03-07:
> esp-mesh-lite 基于 idf5.2 版本能用吗？

可以，不过目前使用的最多的还是 release/v5.1 版本

tswen, 2024-03-07:
> ![75413868129f7ea0cad3ff48bc8ecab](https://private-user-images.githubusercontent.com/29226337/310782546-323e2c29-25fe-4860-b70a-53058df679b6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk4MTE5MDQsIm5iZiI6MTcwOTgxMTYwNCwicGF0aCI6Ii8yOTIyNjMzNy8zMTA3ODI1NDYtMzIzZTJjMjktMjVmZS00ODYwLWI3MGEtNTMwNThkZjY3OWI2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA3VDExNDAwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczNTZhMzQ5OTI3MjIxNzYyZTQwYzU2MWYzOWMxZmM1MDE0NmUxZjFlNzI5OTc4NGVkM2FlMjNkMmJjNGM5MjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.exgkG9H7KPNHTr8OXzLhCnsZp_HZjlo4eCIXhKLrdMY) esp-mesh-lite组网的softap 中的password值不能带字符，只能是数字啊？我带字符组网就会失败。

可以带字符，请确保所有设备的的 password 保持一致

heyangxinmei, 2024-03-08:
[image]
[image]
通过strlcpy((char *)wifi_config.ap.password, "123456", sizeof(wifi_config.ap.password));esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_config)；设置softap password后。设置的password与menuconfig中的不一致，就不能组网了？请问password值不能设置吗？谢谢

tswen, 2024-03-11:
设置 ssid 和 password 最好都在 menuconfig 里面设置，因为并不是一个地方使用到了 BRIDGE_SOFTAP_PASSWORD，这同样也是传入 mesh 的一个参数，不一致的话是组不了网的。如果不想设置密码，可以直接在 menuconfig 中将密码设置为空。

heyangxinmei, 2024-03-11:
根节点怎么获取整个网络内的子节点的数量和MAC？谢谢

tswen, 2024-03-11:
根节点只能直接获取到与其直接连接的 Station 子节点的 mac（通过 esp_wifi_ap_get_sta_list） 剩下的子节点只能应用层上报给根节点消息的方式来让根节点获取到。

heyangxinmei, 2024-03-14:
如果是根节点，使能CONFIG_MESH_LITE_NODE_INFO_REPORT配置，esp_mesh_lite_get_child_node_number（）这个函数的返回值是与根节点直接相连接的第二层的子节点数量呢，还是整个网络的子节点数量呢？谢谢

heyangxinmei, 2024-03-14:
子设备连接的时候根节点打印如下日志，这正常吗？
[image]

tswen, 2024-03-27:
正常的，这个提示只是说明有设备尝试连接了根节点，但是此时根节点 SoftAP 已经被连满了