espressif/esp-mesh-lite #69: Compatibility of ESP-MESH-LITE with PlatformIO

matthieup240, 2024-03-04:
Hello,

I am currently exploring the ESP-MESH-LITE project for a Wi-Fi networking application and am interested in integrating it with PlatformIO for an ESP32 project. I have encountered some challenges in compiling the ESP-MESH-LITE project using PlatformIO and was wondering if there's a recommended approach or if compatibility adjustments are needed.

**Steps to Reproduce:**
1. Created a new PlatformIO project and set the framework to ESP-IDF.
2. Added ESP-MESH-LITE to the `lib` directory of the project.
3. Attempted to compile the project.

**Issues Encountered:**
- Received errors indicating implicit declaration of functions (`esp_bridge_create_all_netif`, `esp_mesh_lite_init`, `esp_mesh_lite_start`) and unknown types (`esp_mesh_lite_config_t`).
- It seems that the ESP-MESH-LITE header files are not being correctly included or recognized by the compiler.

**Questions:**
1. Is there a specific configuration or setup required to build ESP-MESH-LITE projects with PlatformIO?
2. Are there plans to ensure compatibility between ESP-MESH-LITE and PlatformIO, especially considering the ESP-IDF version dependencies?

Any guidance or recommendations on integrating ESP-MESH-LITE with PlatformIO would be greatly appreciated. Thank you for your support and for developing such an interesting project.

Thanks.

tswen, 2024-03-07:
Hello, we currently have no plans to port it to PlatformIO.

kristofkatzenberger, 2024-04-11:
Hi @matthieup240 ! Have you had any luck integrating ESP-MESH-LITE with PlatformIO? Thanks!