espressif/esp-mesh-lite #70: Modifying lite mesh ESP_Bridge SSID & pwsd is invalid

yel-best, 2024-03-18:
* idf (5.1.3)
* espressif/mesh_lite (0.10.0)

I modified the default MESH wifi SSID & pwd to a custom `SSID: ESP_WIFI_MESH & PWD: abcd.123`
After I completed the modification and compiled it and wrote it to ESP32S3, 
I found that the WIFI MESH in the LOG was still finding `ESP_Bridge`

```
I (7261425) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (7264253) [vendor_ie]: Mesh-Lite Scan done
I (7264253) [vendor_ie]: Find ESP_Bridge
I (7264254) [vendor_ie]: RTC store: ssid:ESP_Bridge; bssid:74:4d:bd:8a:1b:75 crc:3293051760
I (7264261) [vendor_ie]: wifi_cfg ESP_Bridge router_config mesh-app-iot
I (7264269) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK

```

`mesh-app-iot` **is my router WIFI, the master node should connect to this routing SSID**

what is the problem?

I have enabled the option of WIFI using PSRAM space in the configuration. Is there any impact?

[image]

**Are there any rules for MESH LITE's SSID &PWD? For example, SSID cannot have special symbols, and PWD can only use letters or numbers?**

thanks

yel-best, 2024-03-20:
can you help me? 🥲 

thanks.

yel-best, 2024-03-20:
Hi

I used a simple example, when I ran it for the first time using the default system configuration` SSID: ESP_Bridge, PWD: 123456` I found that it can run normally

But when I try to modify the `SSID: WIFI-MESH-IOT, PWD:123#123#123 `will begin to fail, unable to network normally, and will always be stuck in the LOG.

```bash
I (59294) [vendor_ie]: Mesh-Lite Scan done
I (59295) [vendor_ie]: Find ESP_Bridge
I (59296) [vendor_ie]: RTC store: ssid:ESP_Bridge; bssid:68:b6:b3:4f:da:d1 crc:2117311556
I (59303) [vendor_ie]: wifi_cfg ESP_Bridge router_config esq-apps-iot
I (59310) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK

```

I just modified the parameters in the configuration file: `CONFIG_BRIDGE_SOFTAP_SSID & CONFIG_BRIDGE_SOFTAP_PASSWORD`

These are the codes of my `app_main`function

```c

static esp_err_t esp_storage_init(void)
{
    esp_err_t ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    return ret;
}

static void wifi_init(void)
{
    // Station + ap
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = CONFIG_ROUTER_SSID,
            .password = CONFIG_ROUTER_PASSWORD,
        },
    };
    esp_bridge_wifi_set_config(WIFI_IF_STA, &wifi_config);

    // Softap
    snprintf((char *)wifi_config.ap.ssid, sizeof(wifi_config.ap.ssid), "%s", CONFIG_BRIDGE_SOFTAP_SSID);
    strlcpy((char *)wifi_config.ap.password, CONFIG_BRIDGE_SOFTAP_PASSWORD, sizeof(wifi_config.ap.password));
    esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_config);
}

void app_wifi_set_softap_info(void)
{
    char softap_ssid[32];
    uint8_t softap_mac[6];
    esp_wifi_get_mac(WIFI_IF_AP, softap_mac);
    memset(softap_ssid, 0x0, sizeof(softap_ssid));

#ifdef CONFIG_BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC
    snprintf(softap_ssid, sizeof(softap_ssid), "%.25s_%02x%02x%02x", CONFIG_BRIDGE_SOFTAP_SSID, softap_mac[3], softap_mac[4], softap_mac[5]);
#else
    snprintf(softap_ssid, sizeof(softap_ssid), "%.32s", CONFIG_BRIDGE_SOFTAP_SSID);
#endif
    esp_mesh_lite_set_softap_ssid_to_nvs(softap_ssid);
    esp_mesh_lite_set_softap_psw_to_nvs(CONFIG_BRIDGE_SOFTAP_PASSWORD);
    esp_mesh_lite_set_softap_info(softap_ssid, CONFIG_BRIDGE_SOFTAP_PASSWORD);
}

void app_main()
{
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %" PRIu32 " bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    /**
     * @brief Set the log level for serial port printing.
     */
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("mqtt_client", ESP_LOG_VERBOSE);
    esp_log_level_set("MQTT_EXAMPLE", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT_BASE", ESP_LOG_VERBOSE);
    esp_log_level_set("esp-tls", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT", ESP_LOG_VERBOSE);
    esp_log_level_set("outbox", ESP_LOG_VERBOSE);

    esp_storage_init();
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

    esp_mesh_lite_start();
}

```

tswen, 2024-03-27:
请问您是改动的 CONFIG_BRIDGE_SOFTAP_SSID、CONFIG_BRIDGE_SOFTAP_PASSWORD 配置项吗，是否所有节点都进行了更改。