espressif/esp-mesh-lite #71: mesh lite内部通信是否存在随机延时发送机制

kkkintama, 2024-03-21:
基于无路由器的案例做了mesh lite组网，两块模块（一个根节点，一个子节点）调用esp_mesh_lite_try_sending_msg进行内部通信时发现一个现象：
根节点设置了200ms间隔定时广播一条信息，且只广播一次无重发，通过系统tick计算得到实际广播间隔为203±1ms，符合预期；而子节点这边通过tick计算发现接收间隔竟然是0~400ms之间的随机数，并不是预期的200ms左右。根节点向下单向通信时，从子节点看接收间隔大部分时间稳定在200±10ms，少部分时间是0~400ms之间的随机数，这个现象在两个模块双向通信时特别明显。
由于手里的项目对时序要求比较高，所以想确认一下，mesh lite内部通信为了规避撞包，是否有做随机延时发送的机制在里面？如果存在这个机制，那么最大发送延时是多久？

tswen, 2024-03-27:
如果此时子节点在进行别的通信或者 Wi-Fi 扫描动作，是会出现一定的通信延迟。

kkkintama, 2024-03-27:
> 不存在您说的随机延时发送，但如果此时子节点在进行别的通信或者 Wi-Fi 扫描动作，是会出现一定的通信延迟

子节点应该不至于时时刻刻扫描吧，测试时我每一帧的接收间隔都在跳动，还挺离谱的。
后来我在论坛上扫了一圈，有人提出以下几种方案去优化wifi接收延时的：
1.调用esp_wifi_set_ps(WIFI_PS_NONE)关闭低功耗
2.关闭wifi AMPDU TX和 AMPDU RX
3.设置 socket option TCP_NODELAY（这个我不知道哪里设置）

实测下来第一种关闭低功耗的配置确实有效，通信延时基本可以做到150ms~250ms；第二种配置实测无效，第三种不知道怎么去配置。

请问有相关文档说明怎么去优化wifi通信延时的么，或者您知道有上述3种配置之外的优化方法么，我想都尝试一下，看能不能把接收延时做到与发送延时相差±10ms左右。感觉mesh通信这块有好多隐藏配置，不知道有什么用和怎么去用，需要文档学习一下。

closed 2024-04-11 by kkkintama

iwonako, 2024-12-11:
请问有找到一些参考文档吗