espressif/esp-mesh-lite #73: 根节点内存泄漏

heyangxinmei, 2024-03-26:
[image]
组网测试12个设备，根节点的内存会慢慢泄漏。子节点不会。子节点和根节点的程序是一样的。

xcguang, 2024-03-27:
请先排查下是否是应用层内存泄漏，当前几个用户出现同样问题，经排查都是用户代码导致内存泄漏

heyangxinmei, 2024-03-28:
我用的mesh_local_control这个例程测试的，测试设备12个，根节点还是会出现内存泄漏的问题。

xcguang, 2024-03-28:
是会一直减少吗？每连上一个节点，父节点都会消耗点内存的

heyangxinmei, 2024-03-28:
我是这样测试的，一共12个设备，根节点一直在线，让11个子节点半个小时断电一次，再上电。这样测试了一天，内存大概泄漏了10220字节。

xcguang, 2024-03-28:
10220字节 是相对什么时候的？有没有可能是 11 个子节点连上来本身的消耗？你可以继续在测试一下，看看还会不会减少。我们内部也测试下

heyangxinmei, 2024-03-29:
不是消耗到11个子节点连接时占用的内存上了。根节点的内存只要有断线连接会一直减少（子节点数量越多越容易发现这个问题），现在内存泄漏到17180了。 我保存的有日志。
[image]
[image]

heyangxinmei, 2024-03-29:
你好，这是保存的日志，从上面可以看到内存泄漏的变化过程。谢谢  

| |
何阳阳
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年3月29日 09:23 |
| 收件人 | ***@***.***> |
| 主题 | 回复： [espressif/esp-mesh-lite] 根节点内存泄漏 (AEGHB-601) (Issue #73) |

你好，这是保存的日志，从上面可以看到内存泄漏的变化过程。谢谢

| |
何阳阳
|
|
***@***.***
|
---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年3月28日 15:42 |
| 收件人 | ***@***.***> |
| 抄送人 | ***@***.***>,
***@***.***> |
| 主题 | Re: [espressif/esp-mesh-lite] 根节点内存泄漏 (AEGHB-601) (Issue #73) |

10220字节 是相对什么时候的？有没有可能是 11 个子节点连上来本身的消耗？你可以继续在测试一下，看看还会不会减少。我们内部也测试下

tswen, 2024-04-24:
@heyangxinmei 你好，最新的 master 上已针对该问题做了修复，可以更新到最新的进行测试

heyangxinmei, 2024-04-25:
你好，更新为最新的master版本后，编译后提示这个问题。请问怎么解决啊？
[image]

tswen, 2024-04-25:
删除掉工程里面的 build 和 managed_components 目录再试试呢？另外能通过 pip show idf-component-manager 看到 idf-component-manager 包管理器的版本是多少吗？

heyangxinmei, 2024-04-25:
我的还是编译不通过，请看下是什么原因？谢谢
[image]

heyangxinmei, 2024-04-26:
删除掉工程里面的 build 和 managed_components 目录还是不行。在哪里运行pip show idf-component-manager指令啊？谢谢

heyangxinmei, 2024-04-26:
删除掉工程里面的 build 和 managed_components 目录还是不行。在哪里运行pip show idf-component-manager指令啊？谢谢
[image]

heyangxinmei, 2024-04-27:
我的同事现在也编译不了了，都提示的这个问题，是不是mesh lite 版本匹配的问题啊？
[image]

heyangxinmei, 2024-04-27:
[image]
idf_component_manager版本是v1.5.2

heyangxinmei, 2024-04-27:
[image]
把这句屏蔽掉就能编译了，这句话是什么作用啊？

tswen, 2024-04-28:
> ![image](https://private-user-images.githubusercontent.com/29226337/326165293-a840c348-5692-471c-b9ec-9a839e0ba60c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

如果这个相对地址有 mesh_lite 组件，就使用本地的 mesh_lite 组件。可能 idf-component-manager 更新之后用法变了，你使用拉取组件的方式，可以先把这句注释掉。

closed 2024-05-27 by xcguang