espressif/esp-mesh-lite #74: Self-organising crashes. 2 Roots forces reset loop

djswirl, 2024-03-31:
I have been struggling for a number of weeks to get mesh light to work. 
Due to the limited documentation error tracing is extremely difficult. Automatic root selection should happen if more than one node connects to the router.

The issue I am facing if I power my first node  it will connect to the router without issue, however when I power the second node which also connects to the router something is Happening with the automatic selection which then causes both nodes getting stuck in a loop of
stack canary watchpoint triggered (tmr svc), then resetting. 
I have increased the stack sizes across the board to try a trace this. I don't think it has something to do with the stack size but more to do with how it is handling messages.

Can somebody give me further information on what is happening in the background in order to determine the automatic root selection so that I can attempt to trace where this error is coming from.

Is there an uncompiled version of the libraries of available to enable us to debug code more easily?

Thanks 

Brian

tswen, 2024-04-01:
Hello, would it be possible to provide the complete device logs for further diagnosis?

djswirl, 2024-04-01:
here are 2 logs. the first node seems to run with issus until i start the second node, then is crashes, witch makes the 2 node crash, and when it starts makes the 1st crash, and it just goes round in a loop

Node 1
`[    13][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
5     4     3     2     1     
[  3288][I][main.cpp:175] setup(): [Mesh_Test] IDF Version : v4.4.6-dirty
[  3289][I][main.cpp:176] setup(): [Mesh_Test] Flash size  : 4 MB (4194304)

I (3293) bridge_common: esp-iot-bridge version: 0.0.0
I (3381) bridge_common: IP Address:192.168.4.1
I (3382) bridge_common: GW Address:192.168.4.1
I (3382) bridge_common: NM Address:255.255.255.0
I (3386) bridge_wifi: IP Address:192.168.4.1
Add netif ap with MyMesh(commit id)
I (3394) bridge_common: netif list add success
Add netif sta with MyMesh(commit id)
I (3403) bridge_common: netif list add success
I (7135) bridge_common: [WIFI_AP_DEF ]Name Server1: 114.114.114.114
I (7147) bridge_wifi: [esp_bridge_wifi_set_config] sta ssid: xxxxxx password: xxxxxx
[  3435][I][main.cpp:96] wifi_init(): [Mesh_Test] esp_bridge_wifi_set_config(STA) : ESP_OK
I (7162) bridge_wifi: [esp_bridge_wifi_set_config] softap ssid: xxxxx_035745 password: xxxxxx
[  4007][I][main.cpp:102] wifi_init(): [Mesh_Test] esp_bridge_wifi_set_config(AP) : ESP_OK
I (7727) Mesh-Lite: esp-mesh-lite component version: 0.10.2
I (7732) bridge_common: [WIFI_AP_DEF ]Name Server1: 114.114.114.114
I (7738) ESPNOW: espnow [version: 1.0] init
I (7741) [vendor_ie]: Mesh-Lite commit id: d919851

I (7747) [vendor_ie]: Mesh ID: 77
D (7752) [vendor_ie]: ota_data_len is 1388 ota_wnd 8328
I (7755) [ESP_Mesh_Lite_Comm]: msg action add success
I (7760) [ESP_Mesh_Lite_Comm]: msg action add success 
I (7762) [ESP_Mesh_Lite_Comm]: Bind Socket 48, port 6364
I (7770) [ESP_Mesh_Lite_Comm]: Bind Socket 49, port 6363
I (7775) [ESP_Mesh_Lite_Comm]: Bind Socket 50, port 6366
I (7780) [ESP_Mesh_Lite_Comm]: Bind Socket 51, port 6365
D (7785) [ESP_Mesh_Lite_Comm]: free heap: 134056
I (7790) [mesh-lite-espnow]: Start espnow task
I (7795) Mesh-Lite: Mesh-Lite connecting
I (7800) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
[  5076][I][main.cpp:50] printWifiMode(): [Mesh_Test] WiFi mode: Station + Access Point
[  5077][I][main.cpp:148] mesh_setup(): [Mesh_Test] Mesh Networking Mode : ROUTER
[  5082][I][main.cpp:150] mesh_setup(): [Mesh_Test] Node Level : 0
I (10644) [vendor_ie]: Mesh-Lite Scan done 
I (10645) [vendor_ie]: RTC store: ssid:xxxxx; bssid:00:00:00:00:00:00 crc:4109069645
I (10647) [vendor_ie]: wifi_cfg gatts router_config xxxxx
I (10652) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK
I (11621) bridge_wifi: Connected with IP Address:192.168.1.102
I (11622) bridge_common: [WIFI_AP_DEF ]Name Server1: 192.168.1.1
I (11624) [vendor_ie]: RTC store: ssid:xxxxx; bssid:00:00:00:00:00:00 crc:3408422391
[ 15089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 128476
[ 15120][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 0
[ 25089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 128476
[ 25093][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 0
[ 35089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 128476
[ 35093][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 0
[ 45089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 128476
[ 45093][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 0
[ 55089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 128476
[ 55093][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 0
I (64776) bridge_wifi: STA Connecting to the AP again... <<<<<<<<<<<<<<<< start second node
D (64878) [ESP_Mesh_Lite_Comm]: free heap: 125660 
D (66088) [ESP_Mesh_Lite_Comm]: free heap: 125124 
D (66091) [ESP_Mesh_Lite_Comm]: free heap: 125528
D (67082) [ESP_Mesh_Lite_Comm]: free heap: 125528 
D (68080) [ESP_Mesh_Lite_Comm]: free heap: 125528 
[ 65089][I][esp_mesh_init.cpp:55] print_system_info_timercb(): [local_control] Sys info, channel: 2, layer: 1, self mac: 3c:71:bf:03:57:44, free heap: 125400
[ 65093][I][esp_mesh_init.cpp:65] print_system_info_timercb(): [local_control] Num of child nodes  : 1
[ 65102][I][esp_mesh_init.cpp:68] print_system_info_timercb(): [local_control] Child mac: 08:d1:f9:25:e8:ac
Guru Meditation Error: Core  0 panic'ed (Double exception). 

[... 119 lines cut ...]
  #2  0x4008d164:0x3ffbcfe0 in xQueueTakeMutexRecursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:731
  #3  0x40084dd1:0x3ffbd000 in lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:153
  #4  0x40084edd:0x3ffbd030 in _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/locks.c:167
[... cut ...]
  #21 0x400d85f6:0x3ffbd6a0 in poll_timer_cb at /builds/ae_group/esp-mesh-lite-priv/esp-mesh-lite/components/mesh_lite_priv/priv_src/esp_mesh_lite_comm.c:2152
  #22 0x4008f27b:0x3ffbd6c0 in prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:565
      (inlined by) prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:634
      (inlined by) prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/timers.c:597
`

djswirl, 2024-04-14:
Any news on this ?

xcguang, 2024-04-15:
Sorry, the developer is on leave, we will debug it this week.

tswen, 2024-04-17:
What version of IDF are you using, down to which commit? Also, have there been updates to IDF's submodules? You can update them with git submodule update --init --recursive.

YannickIngecom, 2024-04-29:
Hi, I get the same issue here:

SDK 5.1.3
Last version of submodules checkout today.

```
I (13) boot: ESP-IDF v5.1.3-dirty 2nd stage bootloader
I (13) boot: compile time Apr 29 2024 16:50:08
I (13) boot: Multicore bootloader
I (17) boot: chip revision: v3.0
I (21) boot.esp32: SPI Speed      : 40MHz
I (25) boot.esp32: SPI Mode       : DIO
I (30) boot.esp32: SPI Flash Size : 8MB
I (34) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 factory          factory app      00 00 00010000 00150000
I (58) boot:  1 ota_0            OTA app          00 10 00160000 00150000
I (66) boot:  2 ota_1            OTA app          00 11 002b0000 00150000
I (73) boot:  3 otadata          OTA data         01 00 00400000 00002000
I (81) boot:  4 phy_init         RF data          01 01 00402000 0000c000
I (88) boot:  5 storage          Unknown data     01 81 0040e000 00384000
I (96) boot:  6 nvs              WiFi data        01 02 00792000 0004b000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3c92ch (248108) map
I (210) esp_image: segment 1: paddr=0004c954 vaddr=3ffb0000 size=036c4h ( 14020) load
I (216) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=e1f70h (925552) map
I (550) esp_image: segment 3: paddr=00131f98 vaddr=3ffb36c4 size=00234h (   564) load
I (551) esp_image: segment 4: paddr=001321d4 vaddr=40080000 size=153bch ( 86972) load
I (603) boot: Loaded app from partition at offset 0x10000
I (603) boot: Disabling RNG early entropy source...
I (614) cpu_start: Multicore app
I (615) quad_psram: This chip is ESP32-D0WD
I (615) esp_psram: Found 8MB PSRAM device
I (617) esp_psram: Speed: 40MHz
I (621) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (628) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (637) cpu_start: Pro cpu up.
I (641) cpu_start: Starting app cpu, entry point is 0x40080f10
0x40080f10: call_start_cpu1 at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/esp_system/port/cpu_start.c:159

I (641) cpu_start: App cpu up.
I (1526) esp_psram: SPI SRAM memory test OK
I (1534) cpu_start: Pro cpu start user code
I (1534) cpu_start: cpu freq: 240000000 Hz
I (1534) cpu_start: Application information:
I (1538) cpu_start: Project name:     ap2201
I (1543) cpu_start: App version:      0.3.0
I (1547) cpu_start: Compile time:     Apr 29 2024 17:23:32
I (1554) cpu_start: ELF file SHA256:  ac6b182545cc2014...
I (1560) cpu_start: ESP-IDF:          v5.1.3-dirty
I (1565) cpu_start: Min chip rev:     v3.0
I (1570) cpu_start: Max chip rev:     v3.99 
I (1575) cpu_start: Chip rev:         v3.0
I (1580) heap_init: Initializing. RAM available for dynamic allocation:
I (1587) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1593) heap_init: At 3FFB9730 len 000268D0 (154 KiB): DRAM
I (1599) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1606) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1612) heap_init: At 400953BC len 0000AC44 (43 KiB): IRAM
I (1619) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1627) spi_flash: detected chip: generic
I (1631) spi_flash: flash io: dio
I (1636) app_start: Starting scheduler on CPU0
I (1640) app_start: Starting scheduler on CPU1
I (1640) main_task: Started on CPU0
I (1650) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1655) main_task: Calling app_main()
I (1660) ap2201: INGECOM Access Point 2201
[... 115 lines cut ...]
I (7620) ota: Init
I (7620) tftp: Init
I (12180) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<0,0>, prof:11
I (12180) wifi:station: b0:b2:1c:b4:5c:08 join, AID=1, bgn, 40D
I (12200) bridge_wifi: STA Connecting to the AP again...
I (12230) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
E (36260) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (41260) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (46260) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (51260) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (56260) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 

***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40081632:0x3ffbc5e0 0x4008ab39:0x3ffbc600 0x4008b1ea:0x3ffbc620 0x4008b8da:0x3ffbc6a0 0x4008ac70:0x3ffbc6d0 0x4008ac22:0xa5a5a5a5 |<-CORRUPTED
0x40081632: panic_abort at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/esp_system/panic.c:472
0x4008ab39: esp_system_abort at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/esp_system/port/esp_system_chip.c:84
0x4008b1ea: vApplicationStackOverflowHook at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:581
0x4008b8da: vTaskSwitchContext at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/freertos/FreeRTOS-Kernel/tasks.c:3840
0x4008ac70: _frxt_dispatch at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450
0x4008ac22: _frxt_int_exit at /Users/yannick/Documents/Esp-Idf/esp-idf_5.1.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245

ELF file SHA256: ac6b182545cc2014

```

tswen, 2024-05-07:
You can increase the ESP_TIMER_TASK_STACK_SIZE configuration in menuconfig.

djswirl, 2024-05-16:
The stack is already set high. I've given up trying to get this to work. Maybe I'll re visit in the future when it's more stable. 
Do you have plans to make it Platformio compatible?

tswen, 2024-05-27:
There are currently no plans to be compatible with Platformio.

xcguang, 2024-05-27:
> The stack is already set high. I've given up trying to get this to work. Maybe I'll re visit in the future when it's more stable. Do you have plans to make it Platformio compatible?

1. After failed to build, mv `managed_components/espressif__iot_bridge` to `components/espressif__iot_bridge`
2. Modify components/espressif__iot_bridge/CMakeLists.txt as following
```
set(FALLBACK_DNS_SERVER "8.8.8.8")
target_compile_options(__idf_lwip PRIVATE
                       "-DIP_NAPT_PORTMAP=1"
                       "-DIP_PORTMAP_MAX=255"
                       "-DFALLBACK_DNS_SERVER_ADDRESS(p)=ipaddr_aton(\"${FALLBACK_DNS_SERVER}\",(p))")
```

to 

```
target_compile_options(__idf_lwip PRIVATE
                       "-DIP_NAPT_PORTMAP=1"
                       "-DIP_PORTMAP_MAX=255")
```
3.  This is my sdkconfig, it can run

[sdkconfig.esp32s3camlcd.zip](https://github.com/espressif/esp-mesh-lite/files/15455410/sdkconfig.esp32s3camlcd.zip)

```
␛[0;31mE (86342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (86342) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK ␛[0m
␛[0;32mI (89172) [vendor_ie]: Mesh-Lite Scan done ␛[0m
␛[0;32mI (89182) [vendor_ie]: RTC store: ssid:; bssid:00:00:00:00:00:00 crc:4043245259 ␛[0m
␛[0;31mE (91342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (91392) no_router: System information, channel: 11, layer: 0, self mac: 34:85:18:90:e2:98, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 249912␛[0m
␛[0;31mE (96342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (96342) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK ␛[0m
␛[0;32mI (99172) [vendor_ie]: Mesh-Lite Scan done ␛[0m
␛[0;32mI (99172) [vendor_ie]: RTC store: ssid:; bssid:00:00:00:00:00:00 crc:4043245259 ␛[0m
␛[0;31mE (101342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (101392) no_router: System information, channel: 11, layer: 0, self mac: 34:85:18:90:e2:98, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 249900␛[0m
␛[0;31mE (106342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (106342) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK ␛[0m
␛[0;32mI (109172) [vendor_ie]: Mesh-Lite Scan done ␛[0m
␛[0;32mI (109172) [vendor_ie]: RTC store: ssid:; bssid:00:00:00:00:00:00 crc:4043245259 ␛[0m
␛[0;31mE (111342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (111392) no_router: System information, channel: 11, layer: 0, self mac: 34:85:18:90:e2:98, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 249912␛[0m
␛[0;31mE (116342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (116342) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK ␛[0m
␛[0;32mI (119172) [vendor_ie]: Mesh-Lite Scan done ␛[0m
␛[0;32mI (119182) [vendor_ie]: RTC store: ssid:; bssid:00:00:00:00:00:00 crc:4043245259 ␛[0m
␛[0;31mE (121342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
␛[0;32mI (121392) no_router: System information, channel: 11, layer: 0, self mac: 34:85:18:90:e2:98, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 249900␛[0m
␛[0;31mE (126342) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm ␛[0m
```