espressif/esp-mesh-lite #75: The component usb_device could not be found

Mair, 2024-04-09:
Hi there,

I'm trying the following
1. in vs code created an empty project targeting the ESP32-S3
2. ran `idf.py add-dependency "espressif/mesh_lite^0.10.2"`
3. tried to build with `idf.py build'

I get the following error
> HINT: The component usb_device could not be found....

I looked for other issues related to this and found #20 
if I add the following to my idf_component.yml
```yaml
  usb_device:
    path: components/usb/usb_device
    git: https://github.com/espressif/esp-iot-bridge.git
    rules:
      - if: "target in [esp32s2, esp32s3]"
```
then I get this errer
/Users/mair/projects/spike/mesh-lite-spike/managed_components/espressif__iot_bridge/src/bridge_common.c:487:13: error: implicit declaration of function 'ip_napt_enable' [-Werror=implicit-function-declaration]
  487 |             ip_napt_enable(ip_info->ip.addr, 1);
      |             ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors

Could you please advise how to get the library compiling

I understand the library is new however, might I humbly request a getting started guide while the library is having teething issues.

tswen, 2024-04-09:
You can check your configuration to confirm whether the following two configuration items are enabled:

CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y

Mair, 2024-04-10:
thanks @tswen 
updating the sdkconfig with 
CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y
and adding the following to my idf_component.yml
```yaml
  usb_device:
    path: components/usb/usb_device
    git: https://github.com/espressif/esp-iot-bridge.git
    rules:
      - if: "target in [esp32s2, esp32s3]"
```

Has enabled me to compile

closed 2024-04-10 by Mair