espressif/esp-mesh-lite #77: 为什么ESP32S3运行NO_ROUTER时会周期性重启

HuayongLiang, 2024-04-13:
未对程序做任何改动，按照例程进行设定，出现周期性重启，重启处的信息如下：
I (1064) [vendor_ie]: Mesh ID: 77
I (1070) [ESP_Mesh_Lite_Comm]: msg action add success 
I (1075) [ESP_Mesh_Lite_Comm]: Bind Socket 54, port 6364
I (1080) [ESP_Mesh_Lite_Comm]: Bind Socket 55, port 6363
I (1087) [ESP_Mesh_Lite_Comm]: Bind Socket 56, port 6366 
I (1093) [ESP_Mesh_Lite_Comm]: Bind Socket 57, port 6365
I (1099) [mesh-lite-espnow]: Start espnow task
I (1104) [ESP_Mesh_Lite_Comm]: msg action add success 
I (1110) no_router: Root node
I (1113) Mesh-Lite: Mesh-Lite connecting
I (1118) main_task: Returned from app_main()
I (11118) no_router: System information, channel: 11, layer: 1, self mac: f4:12:fa:81:4b:f4, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 262528
I (21118) no_router: System information, channel: 11, layer: 1, self mac: f4:12:fa:81:4b:f4, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 262528
E (31069) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (31069) Mesh-Lite: Mesh-Lite Core is in abnormal state, preparing to reboot the device.

assert failed: esp_mesh_lite_event_ip_changed_handler esp_mesh_lite.c:222 (0)

Backtrace: 0x403756f2:0x3fca5c30 0x4037b88d:0x3fca5c50 0x40382c11:0x3fca5c70 0x4200a48c:0x3fca5d90 0x42093a59:0x3fca5db0 0x42093582:0x3fca5de0 0x4209365d:0x3fca5e20 0x4037c6e5:0x3fca5e40
0x403756f2: panic_abort at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_system/panic.c:466
0x4037b88d: esp_system_abort at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x40382c11: __assert_func at E:/Program_file/esp_idf/v5.2/esp-idf/components/newlib/assert.c:81
0x4200a48c: esp_mesh_lite_event_ip_changed_handler at D:/uwb/my_project/esp32/test3/esp-mesh-lite/examples/no_router/managed_components/espressif__mesh_lite/src/esp_mesh_lite.c:222
0x42093a59: handler_execute at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_event/esp_event.c:137
0x42093582: esp_event_loop_run at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_event/esp_event.c:593
0x4209365d: esp_event_loop_run_task at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_event/esp_event.c:107
0x4037c6e5: vPortTaskWrapper at E:/Program_file/esp_idf/v5.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: fc54d6bbd

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403756dc
0x403756dc: esp_restart_noos at E:/Program_file/esp_idf/v5.2/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

walker2048, 2024-04-13:
我也是，使用NO_ROUTER工程，未修改任何代码，遇到了相同的错误。使用两个ESP32C6，Root节点会周期性重启，另一个节点能连上Root节点，获取到IP地址，然后发送了消息之后，Root节点就重启了。错误代码如下
`I (1618) no_router: Root node
I (1620) Mesh-Lite: Mesh-Lite connecting
I (1625) main_task: Returned from app_main()
I (11625) no_router: System information, channel: 11, layer: 1, self mac: 40:4c:ca:4c:ae:1c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 327524
I (19603) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<0,0>, prof:11
I (19603) wifi:(trc)phytype:CBW20-LGI, snr:72, maxRate:0, highestRateIdx:1
I (19605) wifi:(trc)rate(L-MCS7, schedIdx:1), ampdu(rate:L-MCS7, schedIdx(1, stop:8)), snr:72, ampduState:wait operational
I (19615) wifi:ifidx:1, rssi:-21, nf:-93, phytype(0x2, CBW20-LGI), phymode(0x3, 11bgn), max_rate:0, he:0
I (19625) wifi:max ampdu length exponent:0(8191 bytes), mmss:7(16 us)
I (19631) wifi:station: 54:32:04:0b:4a:b0 join, AID=1, bgn, 20
I (19744) bridge_wifi: STA Connecting to the AP again...
I (19966) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (21625) no_router: System information, channel: 11, layer: 1, self mac: 40:4c:ca:4c:ae:1c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 324748
I (21629) no_router: Child mac: 54:32:04:0b:4a:b0
E (31576) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (31576) Mesh-Lite: Mesh-Lite Core is in abnormal state, preparing to reboot the device.

assert failed: esp_mesh_lite_event_ip_changed_handler esp_mesh_lite.c:222 (0)
Stack dump detected
Core  0 register dump:
MEPC    : 0x4080062e  RA      : 0x40806810  SP      : 0x4087f260  GP      : 0x40812800  
0x4080062e: panic_abort at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_system/panic.c:452

0x40806810: __ubsan_include at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x4085956c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000007a  S1      : 0x00000001  A0      : 0x4087f29c  A1      : 0x408144e9  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4081a000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4087f3bd  
S4      : 0x408144e8  S5      : 0x00000000  S6      : 0xffffffff  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x4087f29c "assert failed: esp_mesh_lite_event_ip_changed_handler esp_mesh_lite.c:222 (0)") at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_system/panic.c:452
452         *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x4087f29c "assert failed: esp_mesh_lite_event_ip_changed_handler esp_mesh_lite.c:222 (0)") at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_system/panic.c:452
#1  0x40806810 in esp_system_abort (details=details@entry=0x4087f29c "assert failed: esp_mesh_lite_event_ip_changed_handler esp_mesh_lite.c:222 (0)") at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4080baae in __assert_func (file=file@entry=0x420a429c "", line=line@entry=222, func=<optimized out>, func@entry=0x420a442c <__func__.0> "", expr=expr@entry=0x420a26d0 "") at /mnt/MyNAS/walker/.bin/esp-idf/components/newlib/assert.c:81
#3  0x4200a124 in esp_mesh_lite_event_ip_changed_handler (arg=<optimized out>, event_base=<optimized out>, event_id=<optimized out>, event_data=0x0) at /mnt/MyNAS/walker/esp32/esp-mesh-lite/components/mesh_lite/src/esp_mesh_lite.c:222
#4  0x4209dcfe in handler_execute (loop=loop@entry=0x4087e668, handler=<optimized out>, post=...) at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_event/esp_event.c:137
#5  0x4209e442 in esp_event_loop_run (event_loop=event_loop@entry=0x4087e668, ticks_to_run=ticks_to_run@entry=4294967295) at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_event/esp_event.c:593
#6  0x4209e574 in esp_event_loop_run_task (args=0x4087e668, args@entry=<error reading variable: value has been optimized out>) at /mnt/MyNAS/walker/.bin/esp-idf/components/esp_event/esp_event.c:107
#7  0x40809092 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /mnt/MyNAS/walker/.bin/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:202
ELF file SHA256: a9053b3a0f313768

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a: software_reset_cpu in ROM`

xcguang, 2024-04-15:
https://github.com/espressif/esp-mesh-lite/blob/961f40219d70ede721e6864e34400121813320cb/components/mesh_lite/src/esp_mesh_lite.c#L222  这一行需要先注释掉

xcguang, 2024-05-27:
请拉取最新代码

closed 2024-05-27 by xcguang